ROBOTIS-GIT/turtlebot3 #626: obstacle detection does not work

kumaxxp, 2020-07-28:
ISSUE TEMPLATE ver. 0.3.0

1. Which TurtleBot3 you have?

    - [x] Burger
    - [ ] Waffle
    - [ ] Waffle Pi

2. Which ROS is working with TurtleBot3?

    - [ ] ROS 1 Kinetic Kame
    - [ ] ROS 1 Melodic Morenia
    - [x] ROS 2 Dashing Diademata
    - [ ] ROS 2 Eloquent Elusor
    - [ ] etc (PLEASE, WRITE DOWN YOUR ROS VERSION HERE)

3. Which SBC(Single Board Computer) is working on TurtleBot3?

    - [x] Raspberry Pi 3
    - [ ] Intel Joule 570x
    - [ ] etc (PLEASE, WRITE DOWN YOUR SBC HERE)

4. Which OS you installed in SBC?

    - [ ] Raspbian
    - [ ] Ubuntu MATE 16.04 or later
    - [x] Ubuntu Server 18.04 or later
    - [ ] etc (PLEASE, WRITE DOWN YOUR OS)

5. Which OS you installed in Remote PC?

    - [ ] Ubuntu 16.04 LTS (Xenial Xerus)
    - [x] Ubuntu 18.04 LTS (Bionic Beaver)
    - [ ] Linux Mint 18.x
    - [ ] Linux Mint 19.x
    - [ ] etc (PLEASE, WRITE DOWN YOUR OS)

6. Write down software version and firmware version

    - Software version: [x.x.x]
    - Firmware version: [x.x.x]

7. Write down the commands you used in order

    - ros2 launch turtlebot3_example turtlebot3_obstacle_detection

8. Copy and Paste your error message on terminal

    - file 'turtlebot3_obstacle_detection' was not found in the share directory of package 'turtlebot3_example' which is at '/home/kuma/turtlebot3_ws/install/turtlebot3_example/share/turtlebot3_example'

9. Please, describe detailedly what difficulty you are in

(TurtleBot3 18.3.2 Obstacle Detection)

I have ROS2 installed on a remote PC.
Most features will work. I can run SLAM, etc., but the obstacle_detection doesn't work.
I would appreciate your advice.

ashekim, 2020-08-03:
@kumaxxp 
Hello @kumaxxp. I'm sorry for the late reply.

I'll check the issue and get back to you.
Thank you.

ashekim, 2020-08-04:
@kumaxxp 

Hello @kumaxxp.
Please command $ros2 run turtlebot3_example turtlebot3_obstacle_detection 
instead of  $ros2 launch turtlebot3_obstacle_detection

Thank you.

kumaxxp, 2020-08-05:
@ashekim 

Thank you very much. It seems to have been executed successfully.

$ ros2 run turtlebot3_example turtlebot3_obstacle_detection
[INFO] [turtlebot3_obstacle_detection]: Turtlebot3 obstacle detection node has been initialised.

Later, I will move TurtleBot3 and enjoy.

ashekim, 2020-08-10:
@kumaxxp 

Hello @kumaxxp . I would greatly appreciate it.
Please enjoy with your turtlebot3 :) 

Thanks.

closed 2020-08-10 by ashekim