ROBOTIS-GIT/turtlebot3 #627: /odom/twist/linear value from turtlebot3 is abnormally small (at the order of 5e-12)

wilsonthongwk, 2020-08-09:
ISSUE TEMPLATE ver. 0.3.0

1. Which TurtleBot3 you have?

    - [X] Burger
    - [ ] Waffle
    - [ ] Waffle Pi

2. Which ROS is working with TurtleBot3?

    - [ ] ROS 1 Kinetic Kame
    - [ ] ROS 1 Melodic Morenia
    - [X] ROS 2 Dashing Diademata
    - [ ] ROS 2 Eloquent Elusor
    - [ ] etc (PLEASE, WRITE DOWN YOUR ROS VERSION HERE)

3. Which SBC(Single Board Computer) is working on TurtleBot3?

    - [x] Raspberry Pi 3
    - [ ] Intel Joule 570x
    - [ ] etc (PLEASE, WRITE DOWN YOUR SBC HERE)

4. Which OS you installed in SBC?

    - [ ] Raspbian
    - [ ] Ubuntu MATE 16.04 or later
    - [ ] Ubuntu Server 18.04 or later
    - [X] etc http://old-releases.ubuntu.com/releases/18.04.3/ubuntu-18.04.3-preinstalled-server-arm64+raspi3.img.xz (suggested by https://github.com/ROBOTIS-GIT/turtlebot3/issues/546#issuecomment-610725769)

5. Which OS you installed in Remote PC?

    - [ ] Ubuntu 16.04 LTS (Xenial Xerus)
    - [ ] Ubuntu 18.04 LTS (Bionic Beaver)
    - [ ] Linux Mint 18.x
    - [ ] Linux Mint 19.x
    - [X] etc (docker image `osrf/ros:dashing-desktop`)

6. Write down software version and firmware version

    - Software version: [x.x.x]
    - Firmware version: [x.x.x]

7. Write down the commands you used in order

8. Copy and Paste your error message on terminal

9. Please, describe detailedly what difficulty you are in

My turtlebot3 is unable to navigation around my map, so I am following the [guide](http://kaiyuzheng.me/documents/navguide.pdf) and attempt to tune the navigation2 stack.

however when trying to tune the maximum velocity according to /odom output, I found the /odom/twist/linear value are very abnormal. 

### actual turtlebot3

at the full speed forward, the /odom/twist/linear/x is only -5e-12.

[image]

As i accelerate turtlebot3 from 0 to max speed foward, the change of /odom/twist/linear/x value is shown below.

[image]

The velocity value from odom looks very wrong.

### gazebo turtlebot3

I then run the gazebo and check the /odom/twist/linear/x value in simulation.

the /odom/twist/linear/x value in simulation looks very normal. At full speed foward, the /odom/twist/linear/x  is 0.22

[image]

as I acceulate the turtlebot3 in gazebo, the /odom/twist/linear/x changes as follow. It looks very normal also.

[image]

Is there any bugs in computing the /odom/twist/linear/x in the actual turtlebot3? 

thanks,
Wilson

ashekim, 2020-08-18:
Hello @wilsonthongwk 

Thanks for your interest in turtlebot3.
We'll take a look at it and get back to you.

Thank you for your patience.

ROBOTIS-Will, 2020-12-01:
The odom does not represent the linear / angular velocity.
In order to properly read them, you should read cmd_vel topic instead.

ROBOTIS-Will, 2020-12-01:
This issue thread will be closed as there is no recent activity. Please feel free to reopen if needed.

closed 2020-12-01 by ROBOTIS-Will