ROBOTIS-GIT/turtlebot3 #628: put on larger wheels and make the robot four times bigger

stevensu1838, 2020-08-11:
ISSUE TEMPLATE ver. 0.3.0

1. Which TurtleBot3 you have?
    - [ ] Waffle Pi

2. Which ROS is working with TurtleBot3?
    - [ ] ROS 1 Kinetic Kame
3. Which SBC(Single Board Computer) is working on TurtleBot3?
    - [ ] Raspberry Pi 3
4. Which OS you installed in SBC?
    - [ ] Ubuntu MATE 16.04 or later
5. Which OS you installed in Remote PC?
    - [ ] Ubuntu 16.04 LTS (Xenial Xerus)
9. Please, describe detailedly what difficulty you are in

    Hi, I've got a Waffle Pi and extra waffle boards but this robot is too small for my application. I am trying to put on larger wheels and make the robot twice as long and twice as wide. So the robot will be actually larger that all these turtlebot3 friends you provided and I can't use them directly. Since I am not the only user who is having this problem. Could you please tell me what do we need to change in your ROS turtblebot package after customizing the turtlebot waffle to a large one? I still want to do LSD sensing and SLAM. But the position of the LDS will be higher and the distance between wheels will be bigger. How does this influence the use of your turtblebot package? Thank you so much in advance.

cwillia109, 2020-08-15:
I did not write the driver nodes, but I suspect that the big changes are:

- faster linear speeds with larger wheels

- different turning speeds (this depends of the distance between the tires and the new linear speed)

- a new caster mounting is required

Teleop control might not require many changes, but automated control will require this new parameters. Changing the URDF will also be needed to get correct TF handling. This should fix your LDS issues when using the robot_state_publisher.

stevensu1838, 2020-08-18:
Thanks a lot for your reply. 

I am using a Waffle Pi.

1. For simulation, I'll need to modify the URDF for my new relative poses of things and potentially wheel meshes to look like your real version. The simulated version will also need updated wheel radius / separations. My question is in which file shall change the wheel radius / separations parameters for my own customized robot? 

2. For navigation, I'd need to change the footprint appropriately. The TB itself has an embedded board that will need to be aware of the changes. For the burger it looks like this file has the [constants](https://github.com/ROBOTIS-GIT/OpenCR/blob/8546128ad32797ea3f5d267b85b64abc49f45839/arduino/opencr_arduino/opencr/libraries/turtlebot3/examples/turtlebot3_burger/turtlebot3_core/turtlebot3_burger.h). you'd need to update and then reflash. Where shall I change the contacts for the waffle pi?

cwillia109, 2020-08-18:
1. It looks like the URDF is located here: https://github.com/ROBOTIS-GIT/turtlebot3/tree/master/turtlebot3_description/urdf

2. Good find. I haven't seen that file before. There are also several files here that need to be edited: https://github.com/ROBOTIS-GIT/turtlebot3/tree/master/turtlebot3_navigation/param

That's all I can think of at the moment for what changes need to be made.

ROBOTIS-Will, 2020-08-21:
The odom is calculated from the OpenCR firmware so you should also revise the source code to reflect the wheel change.
You'll need to modify [various definitions](https://github.com/ROBOTIS-GIT/OpenCR/blob/master/arduino/opencr_arduino/opencr/libraries/turtlebot3/examples/turtlebot3_waffle/turtlebot3_core/turtlebot3_waffle.h).
Since we are investigating on some custom firmware issues with the latest OpenCR board manager, please try with v1.4.5 or lower for now.
Thank you.

ROBOTIS-Will, 2020-12-01:
This issue thread will be closed as there is no recent activity. Please feel free to reopen if needed.

closed 2020-12-01 by ROBOTIS-Will

apusky, 2023-06-07:
@ROBOTIS-Will Hello and thank you for this info. I helped me to get started.
The issue that I have is when downloading the code with Arduino IDE.
The download is successful(I do not hear the tone tho)
The issue arises when I try to bringup my turtlebot I get the following error:
`[turtlebot3_ros-3] [ERROR] [1685856599.759004375] [DynamixelSDKWrapper]: Failed to open the port(/dev/ttyACM0)!`
Please let me know if there are specific steps to follow during the download process.
Thank you!

shiba-8ro, 2023-06-12:
@apusky 
Just to make sure, do you have OpenCR1.0 in recovery mode?
Plz check 3.3.7 in OpenCR setup in e-manual.
https://emanual.robotis.com/docs/en/platform/turtlebot3/opencr_setup/#opencr-setup
[image]

ROBOTIS-Will, 2023-06-13:
@apusky 
Your OpenCR may not be connected to the RPi through the `ttyACM0` port.
If there are other USB devices connected to the OpenCR or the ACM0 port is occupied, you may get the port error.
Please check which port is currently in use with the following command.
```
$ ls dev/ttyACM*
```

apusky, 2023-06-14:
> @apusky 
> 
> Your OpenCR may not be connected to the RPi through the `ttyACM0` port.
> 
> If there are other USB devices connected to the OpenCR or the ACM0 port is occupied, you may get the port error.
> 
> Please check which port is currently in use with the following command.
> 
> ```
> 
> $ ls dev/ttyACM*
> 
> ```
I made sure to test only with the openCR board connected.
I got a question about the board when programed with Arduino IDE. Should the board make the usual sound when booting? Or is the beeping sound gone after loading a modified program.
Thank you!

ROBOTIS-Will, 2023-07-06:
@apusky

In case of OpenCR ROS2 firmware, do-re-mi melody will be played after all initialization is completed.

```c++
void TurtleBot3Core::begin(const char* model_name)
{
...
...

  //To indicate that the initialization is complete.
  sensors.makeMelody(1); 

  DEBUG_PRINTLN("Begin End...");
}
```