ROBOTIS-GIT/turtlebot3 #629: revert nav2_bringup RVIZ config file location in ROS-Dashing

tlpss, 2020-08-11:
I believe 9e8b874 should be reverted for ROS2 dashing

The .rviz file for nav2_bringup is still in the launch directory in this dashing-devel branch  [see here](https://github.com/ros-planning/navigation2/tree/dashing-devel/nav2_bringup) , so if you install `ros-dashing-nav-bringup` using apt on Ubuntu 18.04 and then try to follow the tutorial on [navigation](https://emanual.robotis.com/docs/en/platform/turtlebot3/ros2_simulation/#ros-2-simulation#virtual-navigation) , the RVIZ configuration is not loaded

changing the folder from `rviz` to `launch` however,  seems to resolve the issue

ashekim, 2020-08-18:
@tlpss 

I'm sorry for the late reply.
we'll check the issue and get back to you.

Thank you.

ashekim, 2020-08-19:
Hi @tlpss 

Thank you for fixing the file location :) 
I'll apply this change to Nav2 related dashing-devel branches.

closed 2020-08-19 by ashekim

tlpss, 2020-08-19:
you're welcome!