ROBOTIS-GIT/turtlebot3 #631: obstacle_distance  alway get 0.0 value

ncnynl, 2020-08-19:
ISSUE TEMPLATE ver. 0.3.0

1. Which TurtleBot3 you have?

    - [ ] Burger

2. Which ROS is working with TurtleBot3?

    - [ ] ROS 2 foxy

  

3. Which SBC(Single Board Computer) is working on TurtleBot3?

    - [ ] Raspberry Pi 4

4. Which OS you installed in SBC?

    - [ ] Ubuntu mate 20.04  

5. Which OS you installed in Remote PC?

    - [ ] Ubuntu 20.04 LTS 

7.  issue

$ ros2 run turtlebot3_example turtlebot3_obstacle_detection

on turtlebot3_obstacle_detection.py 

obstacle_distance = min(self.scan_ranges)

the value alway is 0.0   

i find the self.scan_ranges have non value .   i fixed it as 

import numpy as np

a = self.scan_ranges
obstacle_distance = min(np.array(a)[np.flatnonzero(a)])

ROBOTIS-Will, 2020-08-21:
Thank you for reporting the issue.
We'll take a look at it and update the code.

whiteocean016, 2020-09-18:
I had the same problem but on waffle_pi, Ubuntu 18.04, Raspberry Pi 3b+.
The fix from @ncnynl didn't exactly work for me. I had to do it like this:
```
a = np.array(self.scan_ranges)
obstacle_distance = min(a[np.flatnonzero(a)])
```

GyuH13, 2024-10-31:
We had completed updating the code.
I wish you had solved this problem.

closed 2024-11-08 by sunghowoo