ROBOTIS-GIT/turtlebot3 #632: Can I Install Ubuntu MATE on Raspberry Pi 3 B + or not?

stevensu1838, 2020-08-20:
Hi,
In the official e-manual of turtlebot3, it states Ubuntu Mate does not correspond to Raspberry Pi 3 B + at present.
Install Linux (Ubuntu MATE)
1) Install Ubuntu MATE on TurtleBot PC

**WARNING: Ubuntu Mate does not correspond to Raspberry Pi 3 B + at present. If you have it, please install Raspbian instead.**

However, the ubuntu mate site show it is compatable with Raspberry Pi 3 B + now as shown below: 
[Ubuntu mate for Rasbbery Pi](https://ubuntu-mate.org/ports/raspberry-pi/)
[image]

My question is can I Install Ubuntu MATE on Raspberry Pi 3 B + or not? If so, you may need to update the e-manual. Cheers

ROBOTIS-Will, 2020-08-21:
Thank you for letting us know.
We'll run the test and see if there's any available version of Ubuntu Mate with RPi 3B+.

ROBOTIS-Will, 2020-08-25:
@andrestoga Thank you for sharing the test result.
We'll revise the eManual after the internal test to provide more detail instructions.
Thank you all!

ROBOTIS-Will, 2020-09-25:
According to our recent test, Ubuntu Mate 18.04 and 20.04 64bit tends to freeze on RPi 3B+ from time to time.
We're currently testing with updated Ubuntu 20.04 that fixes this issue.

andrestoga, 2020-09-25:
@ROBOTIS-Will What's the updated Ubuntu 20.04 version that fixes that issue? I thought it freezes because the RPi 3B+ only has 1GB of RAM....

ashekim, 2020-09-25:
@andrestoga 
Hi. 
Even if the RAM was set to 2G, a freeze occurred.
Currently we are testing with the Ubuntu 20.04 server. [link](https://github.com/ROBOTIS-GIT/turtlebot3/issues/593#issuecomment-697243005)

Thank you :)

ROBOTIS-Will, 2020-12-01:
This issue thread will be closed as there is no recent activity. Please feel free to reopen if needed.

closed 2020-12-01 by ROBOTIS-Will

digixmax, 2021-03-17:
Ubuntu-Mate 20.04.1 32-bit failed to complete boot on the Pi 3B.

ROBOTIS-Will, 2021-03-18:
@digixmax 
We have once tested with Ubuntu Mate 20.04.1 64bit on RPi 3B+ up to SLAM and Navigation, but the thorough test had not been performed.
You might need to allocate 2GB swap space.

digixmax, 2021-03-18:
I tried and was able to install ubuntu-20.04.2-preinstalled-server-armhf+raspi.img and then added LightDM/Mate desktop to it.

The server build by itself performed decently, but with Mate desktop added became very sluggish to the point of un-usable.