ROBOTIS-GIT/turtlebot3 #634: TurtleBot3 LDS Lidar laser scan msg.time_increment is 20 x too small ?

MikeHallettUK, 2020-08-31:
ISSUE TEMPLATE ver. 0.3.0

1. Which TurtleBot3 you have?

    - [ x] Burger
    - [ ] Waffle
    - [ ] Waffle Pi

2. Which ROS is working with TurtleBot3?

    - [ ] ROS 1 Kinetic Kame
    - [ x] ROS 1 Melodic Morenia
    - [ ] ROS 2 Dashing Diademata
    - [ ] ROS 2 Eloquent Elusor
    - [ ] etc (PLEASE, WRITE DOWN YOUR ROS VERSION HERE)

3. Which SBC(Single Board Computer) is working on TurtleBot3?

    - [x ] Raspberry Pi 3
    - [ ] Intel Joule 570x
    - [ ] etc (PLEASE, WRITE DOWN YOUR SBC HERE)

4. Which OS you installed in SBC?

    - [x ] Raspbian
    - [ ] Ubuntu MATE 16.04 or later
    - [ ] Ubuntu Server 18.04 or later
    - [ ] etc (PLEASE, WRITE DOWN YOUR OS)

5. Which OS you installed in Remote PC?

    - [ ] Ubuntu 16.04 LTS (Xenial Xerus)
    - [ ] Ubuntu 18.04 LTS (Bionic Beaver)
    - [ ] Linux Mint 18.x
    - [ ] Linux Mint 19.x
    - [x ] etc (Windows 10 pc)

6. Write down software version and firmware version

    - Software version: [x.x.x]
    - Firmware version: [x.x.x]

7. Write down the commands you used in order

    - rostopic echo /scan -n1
    - and wrote python to tf scans to PointCloud2 while TurtleBot3 LDS Lidar rotates clockwise:  Since laser_geometry::LaserProjection  transformLaserScanToPointCloud in C++ on Rasberry PI does not seem to cater for clockwise lidar spin.
    - also used cartographer SLAM, which works OK, but local maps can swing wildly; suggesting cartographer  is also not happy with the published time_increment, and / or does not cater for clockwise lidar spin.

8. Copy and Paste your error message on terminal

    - Output from "rostopic echo /scan -n1" =

time_increment: 2.95299996651e-05
scan_time: 0.0

   - and 360 x 2.95299996651e-05 = 0.0106 seconds   .... while the scanner topic is running at ~ 5hz : e.g. ~ 0.2 seconds scan_time.
   - while my python to tf scans to PointCloud2 program only correctly provides a reasonably stable PointCloud2 with a moving TurtleBot3, once the scan_time had been set to 0.2 seconds, and therefore time_increment = ~ 0.0005556

9. Please, describe detailedly what difficulty you are in

    - you (i.e. the Robotis cpp code for TurtleBot3) seem to calculate time_increment from the motor rpms... 
    - So, is this wrong ?  if so can you correct it in hlds_laser_publisher.cpp ?
    - Or am I misunderstanding all this ?
    - any suggestions for fixing this so cartographer SLAM copes better with the published scan time_increment, and for clockwise lidar spin ?

Thanks

Michael Hallett

ROBOTIS-Will, 2020-09-04:
Hi,
Please allow us some time to review this issue regarding the LDS-01.
Thank you.

ROBOTIS-Will, 2020-09-08:
I've fixed the incorrect time_increment value. 
Could you try with the updated source code in the `hotfix-laserscan` branch of `hls_lfcd_lds_driver` repo?
Thank you.

MikeHallettUK, 2020-09-10:
OK... your new code works = but does not solve the clockwork issue....

- rostopic echo /scan -n1
- …
- time_increment: 0.000559284118935
- scan_time: 0.201342284679

eg using "transformLaserScanToPointCloud" it looks too far into future ( + 0.2 seconds) ...

- [ WARN] [1599729403.605459568]: Lookup would require extrapolation into the future.  Requested time 1599729403.755665139 but the latest data is at time 1599729403.599492954, when looking up transform from frame [base_scan] to frame [odom]

eg with this code = 

`#include <ros/ros.h>
#include <sensor_msgs/PointCloud2.h>
#include <tf2_ros/transform_listener.h>
#include <sensor_msgs/LaserScan.h>
#include <laser_geometry/laser_geometry.h>

ros::Publisher pub;
laser_geometry::LaserProjection projector_;
tf2_ros::Buffer tfBuffer;

void scanCallback (const sensor_msgs::LaserScan::ConstPtr& scan_in)
{
  sensor_msgs::PointCloud2 cloud;
  ros::Duration(0.05).sleep();   // wait for first odom @ 30 Hz after scan ... = fixes extrapolation into the future error.
				  // @ 0.04 seconds ... still getting some warnings ....
    // inscan->time_increment   // needs to be negative to account for clockwise spin of lidar
  try{
	projector_.transformLaserScanToPointCloud("odom",*scan_in,cloud,tfBuffer);
	}
	catch (tf2::TransformException &ex) {
		  ROS_WARN("%s",ex.what());
	}
  
  // Publish the data.
  pub.publish (cloud);
}

int main (int argc, char** argv)
{
  // Initialize ROS
  ros::init (argc, argv, "tf_laser");
  ros::NodeHandle nh;
  tf2_ros::TransformListener tfListener(tfBuffer);

  // Create a ROS publisher for the output point cloud
  pub = nh.advertise<sensor_msgs::PointCloud2> ("tf_pc", 1);

  ros::Subscriber sub = nh.subscribe ("scan", 1000, scanCallback);
  ros::spin ();
}
`

So I added this line into "hls_lfcd_lds_driver" to give negative increment... 

`      scan->time_increment = -scan->time_increment;    // make negative to account for clockwise spin of lidar`

which gives ...

- rostopic echo /scan -n1
- …
- time_increment: -0.000559284118935
- scan_time: 0.201342284679

Then "transformLaserScanToPointCloud" works quite well.

BUT....  then Cartographer SLAM no longer works with the negative scan time increment.

so the next question is can you make a change to Cartographer ~ "cartographer_turtlebot" ?   to solve this clockwise spin of lidar issue...

a) does Cartographer have a parameter to set this ? in LUA ? ... I cannot find it.
b) or can the code that takes the scan in Cartographer be changed to accept either a negative scan time increment, or explicitly reverse the scan order some other way ?

Thanks

Michael Hallett

ROBOTIS-Will, 2020-09-11:
Hi Michael,
I'm afraid that I can make further changes on our code for a specific case, but if the rotational direction is the problem, you can try installing the LDS-01 upside down (though you'll have to sacrifice some angles where supports are installed).
This will give you the correct rotation for your application.

MikeHallettUK, 2020-09-11:
I agree the issue of "TurtleBot3 LDS Lidar laser scan msg.time_increment is 20 x too small" has been solved.

But the Clockwise spin has not ... e.g. see at https://github.com/ROBOTIS-GIT/turtlebot3/issues/633 

**How to tf scans to PointCloud2 while TurtleBot3 LDS Lidar rotates clockwise ? #633**

closed 2020-09-11 by MikeHallettUK