ROBOTIS-GIT/turtlebot3 #635: Error related to rviz

kdonghoo, 2020-09-03:
ISSUE TEMPLATE ver. 0.3.0

1. Which TurtleBot3 you have?

    - [v] Burger
    - [ ] Waffle
    - [ ] Waffle Pi

2. Which ROS is working with TurtleBot3?

    - [v] ROS 1 Kinetic Kame
    - [ ] ROS 1 Melodic Morenia
    - [ ] ROS 2 Dashing Diademata
    - [ ] ROS 2 Eloquent Elusor
    - [ ] etc (PLEASE, WRITE DOWN YOUR ROS VERSION HERE)

3. Which SBC(Single Board Computer) is working on TurtleBot3?

    - [v] Raspberry Pi 3
    - [ ] Intel Joule 570x
    - [ ] etc (PLEASE, WRITE DOWN YOUR SBC HERE)

4. Which OS you installed in SBC?

    - [v] Raspbian
    - [ ] Ubuntu MATE 16.04 or later
    - [ ] Ubuntu Server 18.04 or later
    - [ ] etc (PLEASE, WRITE DOWN YOUR OS)

5. Which OS you installed in Remote PC?

    - [v] Ubuntu 16.04 LTS (Xenial Xerus)
    - [ ] Ubuntu 18.04 LTS (Bionic Beaver)
    - [ ] Linux Mint 18.x
    - [ ] Linux Mint 19.x
    - [ ] etc (PLEASE, WRITE DOWN YOUR OS)

6. Write down software version and firmware version

    - Software version: [1.2.3]
    - Firmware version: [4.19.66]

7. Write down the commands you used in order

    - rosrun rviz rviz -d `rospack find turtlebot3_description`/rviz/model.rviz

8. Copy and Paste your error message on terminal

    - [rospack] Error: package 'rviz' not found

9. Please, describe detailedly what difficulty you are in

    - I have a problem with the rviz file not existing, so I can't move on to the next chapter.

ROBOTIS-Will, 2020-09-04:
Check if you have installed rviz properly on your Remote PC.
```bash
$ source /opt/ros/kinetic/setup.bash
$ rospack find rviz
```
Thank you.

ROBOTIS-Will, 2020-12-09:
This issue will be closed as there isn't any recent activity. Please feel free to reopen when necessary.

closed 2020-12-09 by ROBOTIS-Will