ROBOTIS-GIT/turtlebot3 #636: ROS 2 Dashing TB3 Not Communicating with Remote PC

jlaframboise, 2020-09-05:
ISSUE TEMPLATE ver. 0.3.0

1. Which TurtleBot3 you have?

    - [ ] Burger
    - [ ] Waffle
    - [x] Waffle Pi

2. Which ROS is working with TurtleBot3?

    - [ ] ROS 1 Kinetic Kame
    - [ ] ROS 1 Melodic Morenia
    - [x] ROS 2 Dashing Diademata
    - [ ] ROS 2 Eloquent Elusor
    - [ ] etc (PLEASE, WRITE DOWN YOUR ROS VERSION HERE)

3. Which SBC(Single Board Computer) is working on TurtleBot3?

    - [ ] Raspberry Pi 3
    - [ ] Intel Joule 570x
    - [x] Raspberry Pi 4

4. Which OS you installed in SBC?

    - [ ] Raspbian
    - [ ] Ubuntu MATE 16.04 or later
    - [x] Ubuntu Server 18.04 or later
    - [ ] etc (PLEASE, WRITE DOWN YOUR OS)

5. Which OS you installed in Remote PC?

    - [ ] Ubuntu 16.04 LTS (Xenial Xerus)
    - [x] Ubuntu 18.04 LTS (Bionic Beaver)
    - [ ] Linux Mint 18.x
    - [ ] Linux Mint 19.x
    - [ ] etc (PLEASE, WRITE DOWN YOUR OS)

7. Write down the commands you used in order

    I have installed ROS 2 Dashing on a Turtlebot3, and on a remote pc. I ran through the setup in the turtlebot e-manual, and then tried to go through the [bringup section](https://emanual.robotis.com/docs/en/platform/turtlebot3/ros2_bringup/#ros-2-bringup). 
    The TB3 and remote pc are connected on a static ip on wifi over ssh. 

9. Please, describe detailedly what difficulty you are in

    - The Turtlebot3 Successfully runs bringup, and I can use all turtlebot functionality on the TB3, however I cannot use that functionality from the remote pc. When I run run "ros2 topic list" on the TB3 I see all the topics, but when I run it on the remote pc I only see the two defaults.  
Specifically, when I run the bringup on the remote pc, I do not see the sensor data in Rviz2. 
    - It appears that ROS 2 is running well on both machines, but they are unable to communicate messages and topics. 
    - The ROS_DOMAIN_ID is set to 30 on both machines. 

10. Other context:
    - This issue seems to have been [encountered before here](https://answers.ros.org/question/332048/turtlebot3-ros2-dds-implementation-error/).

ROBOTIS-Will, 2020-09-07:
@jlaframboise 
Hi, thank you for reporting the issue. We'll try to reproduce this issue with our setup.

@ashekim 
Have you ever experienced this issue recently on your Dashing setup?

ashekim, 2020-09-07:
@ROBOTIS-Will 
Sorry, but I don't have experience testing ros2 dashing on Raspberry Pi 4. 
So, I think I need to check this issue.

@jlaframboise 
Hi @jlaframboise . Thanks for your interest in TB3
I will check this issue and I'll get back to you.

Thank you.

jlaframboise, 2020-09-07:
Hi @ROBOTIS-Will @ashekim ,

Thanks for your response, but I actually found the issue late last night, and I don't think it's an issue with ROS 2 or TB3 directly.

It turns out it was an issue with the firewall on the machines, as when I tried disabling the firewall the messages started coming through.

So I think we are okay to close this issue. The only possible improvement I see from this might be a little note on the E-Manual for ROS 2 setup /bring up that mentions checking for firewall issues.

ashekim, 2020-09-08:
@jlaframboise 

Hello @jlaframboise . 
I would greatly appreciate it.
I wish you had a good time with TB3.

Thank you.

closed 2020-09-08 by ashekim