ROBOTIS-GIT/turtlebot3 #637: Using TurtleBot3 Offline with Chrony Setup

ozangungortuhh, 2020-09-07:
ISSUE TEMPLATE ver. 0.3.0

1. Which TurtleBot3 you have?

    - [x] Burger
    - [ ] Waffle
    - [ ] Waffle Pi

2. Which ROS is working with TurtleBot3?

    - [x] ROS 1 Kinetic Kame
    - [ ] ROS 1 Melodic Morenia
    - [ ] ROS 2 Dashing Diademata
    - [ ] ROS 2 Eloquent Elusor
    - [ ] etc (PLEASE, WRITE DOWN YOUR ROS VERSION HERE)

3. Which SBC(Single Board Computer) is working on TurtleBot3?

    - [x] Raspberry Pi 3
    - [ ] Intel Joule 570x
    - [ ] etc (PLEASE, WRITE DOWN YOUR SBC HERE)

4. Which OS you installed in SBC?

    - [x] Raspbian
    - [ ] Ubuntu MATE 16.04 or later
    - [ ] Ubuntu Server 18.04 or later
    - [ ] etc (PLEASE, WRITE DOWN YOUR OS)

5. Which OS you installed in Remote PC?

    - [ ] Ubuntu 16.04 LTS (Xenial Xerus)
    - [x] Ubuntu 18.04 LTS (Bionic Beaver)
    - [ ] Linux Mint 18.x
    - [ ] Linux Mint 19.x
    - [ ] etc (PLEASE, WRITE DOWN YOUR OS)

6. Write down software version and firmware version

    - Software version: [x.x.x]
    - Firmware version: [x.x.x]

7. Write down the commands you used in order

    - roscore
    - roslaunch turtlebot3_bringup turtlebot3_robot.launch
    - roslaunch turtlebot3_navigation turtlebot3_navigation.launch

8. Copy and Paste your error message on terminal

    -  [ WARN] [1503985578.700748532]: Costmap2DROS transform timeout. Current time: 1503985578.7007, global_pose stamp: 1503985533.6728, tolerance: 0.2000

9. Please, describe detailedly what difficulty you are in

    - I have found out from reading old issues and ROS answers that in order to sync the time between TurtleBot and Master PC, I need to use chrony. But unfortunately I could not manage it so far. 
    - Could anyone explain me how exactly I could set chrony up and use it? I followed the isolated networks that is referred in the comment below but I don't see any change in raspberry-pi time. Is there any other step after copying the lines to /etc/chrony.conf file?

(https://answers.ros.org/question/296657/ros-time-synchronization-in-isolated-local-network/)
(https://github.com/ROBOTIS-GIT/turtlebot3/issues/78)

Have you tried using chrony to synchronize?  I've been running my burger without an internet connection using chrony.  I had to do some fiddling with the config files to get it working. Isolated networks at the bottom of this page was helpful https://chrony.tuxfamily.org/doc/3.3/chrony.conf.html

_Originally posted by @jrjennings in https://github.com/ROBOTIS-GIT/turtlebot3/issues/78#issuecomment-391832742_

ashekim, 2020-09-08:
@ozangungortuhh 

Hi. @ozangungortuhh . Thank you for your questions.
The content you gave inquiries seems to be a problem caused by the timesync of remotePC and turtlebot do not match

Please refer to the [https://emanual.robotis.com/docs/en/platform/turtlebot3/faq/#timesync-between-turtlebot3-and-remotepc](url) shown in the eManual.

Thank you.

ROBOTIS-Will, 2020-09-08:
Hi, if you are working under an isolated network environment, chrony looks like the right direction.
However, unfortunately we haven't tested it to sync the time with TurtleBot3.
Thank you.

ozangungortuhh, 2020-10-03:
I have solved the problem using TB3 offline with chrony sync.

closed 2020-10-03 by ozangungortuhh