ROBOTIS-GIT/turtlebot3 #638: question on building TurtleBot3 Friend Conveyor?

stevensu1838, 2020-09-12:
Hi team,

I am a user of turtlebot3 and I am trying to build a TurtleBot3 Friends: Conveyor. However, the instruction on how to assembly it is not enough for me. Like I don't exactly know **how to connect 8 motors to one OpenCR board**. Could you please send me some details instructions on building TurtleBot3 Friends Conveyor? Thanks a lot.

Looking forward to hearing from you soon. Cheers.

ashekim, 2020-09-14:
@stevensu1838 
Hi @stevensu1838 . Thank you for your questions.

Probably, I know that my colleague sent you an answer. 
If you need additional answers, please comment again.

Thank you.

stevensu1838, 2020-09-14:
Hi Team,
Just checked the email and I got the following answer:

To connect wheels to OpenCR Board, please see the following image. You can adopt a daisy chain to connect between DYNAMIXEL and OpenCR 1.0. 
[image]

I just want to double check that do I really connect all the eight XL430-W250-T motor in a chain like this? Any chance you can share a pic of your Conveyor robot?

OpenCR_Monster.png

My last question is the two motors come with turtlebot kit are already set up and assigned ID 1 and ID 2. May I please ask what shall I do with the extra six XL430-W250-T motors I bought from you and make them work with OpenCR for TurtleBot3 Friend Conveyor?

ashekim, 2020-09-15:
@stevensu1838 
Hi, again.

1. Please refer to this image

-front
[image]

-back
[image]

2. Please refer to the link for Dynamixel ID setting.
[2.4.1.1 Dynamixel2Arduino Class](https://emanual.robotis.com/docs/en/popup/arduino_api/setID/)
[5.2 OpenCR and OpenCM Tutorials](https://emanual.robotis.com/docs/en/software/dynamixel/dynamixel_workbench/#opencr-and-opencm-tutorials)

Thank you.

ROBOTIS-Will, 2020-09-25:
@stevensu1838 You can also distribute the DYNAMIXEL as evenly as possible for available OpenCR DYNAMIXEL ports.

ROBOTIS-Will, 2020-12-09:
This issue will be closed as there isn't any recent activity. Please feel free to reopen when necessary.

closed 2020-12-09 by ROBOTIS-Will