ROBOTIS-GIT/turtlebot3 #639: AMCL can't run in noetic version!

GuancongLuo, 2020-09-18:
ISSUE TEMPLATE ver. 0.3.0

1. Which TurtleBot3 you have?

    - [ y] Burger
    - [ ] Waffle
    - [ ] Waffle Pi

2. Which ROS is working with TurtleBot3?

    - [ ] ROS 1 Kinetic Kame
    - [ ] ROS 1 Melodic Morenia
    - [ ] ROS 2 Dashing Diademata
    - [ ] ROS 2 Eloquent Elusor
    - [ y] etc (PLEASE, WRITE DOWN YOUR ROS VERSION HERE)
**ROS1 Noetic**

3. Which SBC(Single Board Computer) is working on TurtleBot3?

    - [ ] Raspberry Pi 3
    - [ ] Intel Joule 570x
    - [ y] etc (PLEASE, WRITE DOWN YOUR SBC HERE)
turtlebot3_simulation gazebo

4. Which OS you installed in SBC?

    - [ ] Raspbian
    - [ ] Ubuntu MATE 16.04 or later
    - [ ] Ubuntu Server 18.04 or later
    - [ y] etc (PLEASE, WRITE DOWN YOUR OS)
ubuntu 20.04
5. Which OS you installed in Remote PC?

    - [ ] Ubuntu 16.04 LTS (Xenial Xerus)
    - [ ] Ubuntu 18.04 LTS (Bionic Beaver)
    - [ ] Linux Mint 18.x
    - [ ] Linux Mint 19.x
    - [ ] etc (PLEASE, WRITE DOWN YOUR OS)
ubuntu 20.04
6. Write down software version and firmware version

    - Software version: [1.2.2]
    - Firmware version: [simulation]

7. Write down the commands you used in order

    - HERE
roslaunch amcl amcl.launch

8. Copy and Paste your error message on terminal

    - HERE
roslaunch amcl amcl_turtlebot3.launch
... logging to /home/robot/.ros/log/562291f8-f9ad-11ea-a46b-0fee0bba33be/roslaunch-robot-desktop-525205.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://robot-desktop:42235/

SUMMARY
========

PARAMETERS
 * /amcl/base_frame_id: base_footprint
 * /amcl/gui_publish_rate: 50.0
 * /amcl/initial_pose_a: 0.0
 * /amcl/initial_pose_x: 0.0
 * /amcl/initial_pose_y: 0.0
 * /amcl/kld_err: 0.02
 * /amcl/laser_lambda_short: 0.1
 * /amcl/laser_likelihood_max_dist: 2.0
 * /amcl/laser_max_beams: 180
 * /amcl/laser_max_range: 3.5
 * /amcl/laser_model_type: likelihood_field
 * /amcl/laser_sigma_hit: 0.2
 * /amcl/laser_z_hit: 0.5
 * /amcl/laser_z_max: 0.05
 * /amcl/laser_z_rand: 0.5
 * /amcl/laser_z_short: 0.05
 * /amcl/max_particles: 3000
 * /amcl/min_particles: 500
 * /amcl/odom_alpha1: 0.1
 * /amcl/odom_alpha2: 0.1
 * /amcl/odom_alpha3: 0.1
 * /amcl/odom_alpha4: 0.1
 * /amcl/odom_frame_id: odom
 * /amcl/odom_model_type: diff
 * /amcl/recovery_alpha_fast: 0.0
 * /amcl/recovery_alpha_slow: 0.0
 * /amcl/resample_interval: 1
 * /amcl/transform_tolerance: 0.5
 * /amcl/update_min_a: 0.2
 * /amcl/update_min_d: 0.2
 * /rosdistro: noetic
 * /rosversion: 1.15.8

NODES
  /
    amcl (amcl/amcl)

ROS_MASTER_URI=http://localhost:11311

process[amcl-1]: started with pid [525227]
[ WARN] [1600433823.102471927, 467.028000000]: No laser scan received (and thus no pose updates have been published) for 467.028000 seconds.  Verify that data is being published on the /scan topic.
[ WARN] [1600433823.102598125, 467.028000000]: No laser scan received (and thus no pose updates have been published) for 467.028000 seconds.  Verify that data is being published on the /scan topic.
*** buffer overflow detected ***: terminated
[amcl-1] process has died [pid 525227, exit code -6, cmd /home/robot/Clarence/navigation_ws/devel/lib/amcl/amcl scan:=scan __name:=amcl __log:=/home/robot/.ros/log/562291f8-f9ad-11ea-a46b-0fee0bba33be/amcl-1.log].
log file: /home/robot/.ros/log/562291f8-f9ad-11ea-a46b-0fee0bba33be/amcl-1*.log
all processes on machine have died, roslaunch will exit
shutting down processing monitor...
... shutting down processing monitor complete
done

9. Please, describe detailedly what difficulty you are in

    - HERE
I want to do navigation task with turtlebot3 and turtlbot3_simulation in gazeobo 11.1/ubuntu20.04

But I can't run success!

Thank you for you help!!

ROBOTIS-Will, 2020-09-21:
@GuancongLuo 
The current OpenCR firmware has not been updated to the Noetic and therefore failling in TF.
Please test with below temporary firmware.
Instructions should be running on the Raspberry Pi.
```bash
$ rm -rf ./opencr_update.tar.bz2
$ wget https://github.com/ROBOTIS-GIT/OpenCR-Binaries/raw/master/turtlebot3/ROS1/latest/opencr_update.tar.bz2
$ tar -xvf opencr_update.tar.bz2
$ cd ./opencr_update
$ wget https://github.com/ROBOTIS-Will/OpenCR/raw/master/arduino/opencr_develop/opencr_ld_shell/opencr_update/burger_turtlebot3_core_noetic.ino.bin
$ sudo chmod 775 burger_turtlebot3_core_noetic.ino.bin
$ ./opencr_ld_shell_arm make burger_turtlebot3_core_noetic.ino.bin burger_noetic 1.2.6
$ ./opencr_ld_shell_arm /dev/ttyACM0 115200 burger_noetic.opencr 1
```
Thank you.

GuancongLuo, 2020-09-23:
It is working!

Thank you so much.

closed 2020-09-25 by ROBOTIS-Will