ROBOTIS-GIT/turtlebot3 #640: Could not open file turtlebot3_burguer.urdf for parsing

andrestoga, 2020-09-19:
ISSUE TEMPLATE ver. 0.3.0

1. Which TurtleBot3 you have?

    - [x] Burger
    - [ ] Waffle
    - [ ] Waffle Pi

2. Which ROS is working with TurtleBot3?

    - [ ] ROS 1 Kinetic Kame
    - [ ] ROS 1 Melodic Morenia
    - [x] ROS 2 Dashing Diademata
    - [ ] ROS 2 Eloquent Elusor
    - [ ] etc (PLEASE, WRITE DOWN YOUR ROS VERSION HERE)

3. Which SBC(Single Board Computer) is working on TurtleBot3?

    - [x] Raspberry Pi 3
    - [ ] Intel Joule 570x
    - [ ] etc (PLEASE, WRITE DOWN YOUR SBC HERE)

4. Which OS you installed in SBC?

    - [ ] Raspbian
    - [x] Ubuntu MATE 16.04 or later
    - [ ] Ubuntu Server 18.04 or later
    - [ ] etc (PLEASE, WRITE DOWN YOUR OS)

5. Which OS you installed in Remote PC?

    - [ ] Ubuntu 16.04 LTS (Xenial Xerus)
    - [x] Ubuntu 18.04 LTS (Bionic Beaver)
    - [ ] Linux Mint 18.x
    - [ ] Linux Mint 19.x
    - [ ] etc (PLEASE, WRITE DOWN YOUR OS)

6. Write down software version and firmware version

    - Software version: [ros2 branch commit fb6b9ac]
    - Firmware version: [0.0.3] for ROS2

7. Write down the commands you used in order

ros2 launch turtlebot3_bringup robot.launch.py

8. Copy and Paste your error message on terminal

[robot_state_publisher-1] Initialize urdf model from file: /home/andrestoga/turtlebot3_ws/install/turtlebot3_description/share/turtlebot3_description/urdf/turtlebot3_burguer.urdf
[robot_state_publisher-1] Could not open file [/home/andrestoga/turtlebot3_ws/install/turtlebot3_description/share/turtlebot3_description/urdf/turtlebot3_burguer.urdf] for parsing.
[robot_state_publisher-1] Unable to initialize urdf::model from /home/andrestoga/turtlebot3_ws/install/turtlebot3_description/share/turtlebot3_description/urdf/turtlebot3_burguer.urdf
[ERROR] [robot_state_publisher-1]: process has died [pid 8260, exit code 255, cmd '/opt/ros/dashing/lib/robot_state_publisher/robot_state_publisher /home/andrestoga/turtlebot3_ws/install/turtlebot3_description/share/turtlebot3_description/urdf/turtlebot3_burguer.urdf __node:=robot_state_publisher __params:=/tmp/launch_params_b7hpkvkx'].
[turtlebot3_ros-3] terminate called after throwing an instance of 'std::runtime_error'
[turtlebot3_ros-3]   what():  Failed to parse parameters from file '/home/andrestoga/turtlebot3_ws/install/turtlebot3_bringup/share/turtlebot3_bringup/param/burguer.yaml': Error opening YAML file, at /tmp/binarydeb/ros-dashing-rcl-yaml-param-parser-0.7.9/src/parser.c:1387
[ERROR] [turtlebot3_ros-3]: process has died [pid 8262, exit code -6, cmd '/home/andrestoga/turtlebot3_ws/install/turtlebot3_node/lib/turtlebot3_node/turtlebot3_ros -i /dev/ttyACM0 __params:=/home/andrestoga/turtlebot3_ws/install/turtlebot3_bringup/share/turtlebot3_bringup/param/burguer.yaml'].
^C[WARNING] [launch]: user interrupted with ctrl-c (SIGINT)
[hlds_laser_publisher-2] [INFO] [hlds_laser_publisher]: Init hlds_laser_publisher Node Main
[hlds_laser_publisher-2] [INFO] [hlds_laser_publisher]: port : /dev/ttyUSB0 frame_id : base_scan
[hlds_laser_publisher-2] [INFO] [rclcpp]: signal_handler(signal_value=2)
[ERROR] [hlds_laser_publisher-2]: process has died [pid 8261, exit code 255, cmd '/home/andrestoga/turtlebot3_ws/install/hls_lfcd_lds_driver/lib/hls_lfcd_lds_driver/hlds_laser_publisher __node:=hlds_laser_publisher __params:=/tmp/launch_params_jj5l7d_0'].

9. Please, describe detailedly what difficulty you are in

I can't bring up turtlebot3 from the e-manual step 2:
https://emanual.robotis.com/docs/en/platform/turtlebot3/ros2_bringup/

ROBOTIS-Will, 2020-09-21:
Hi, I'm sorry for the confusion. We're currently undergoing some changes in the manual.
Could you try using `dashing-devel` branch instead of `ros2`?
Also, for Dashing, it is recommended to use Ubuntu 18.04, but please be aware that there is an ongoing issue (#593) with Ubuntu 18.04 and 20.04.
Thank you.

ROBOTIS-Will, 2020-12-14:
This issue will be closed as there isn't recent activity. Please feel free to reopen when necessary.

closed 2020-12-14 by ROBOTIS-Will