ROBOTIS-GIT/turtlebot3 #641: Dynamixel X and OpenCR button are not recognized in ROS Noetic

pieterwolfert, 2020-09-24:
ISSUE TEMPLATE ver. 0.3.0

1. Which TurtleBot3 you have?

    - [x] Burger
    - [ ] Waffle
    - [ ] Waffle Pi

2. Which ROS is working with TurtleBot3?

    - [ ] ROS 1 Kinetic Kame
    - [ ] ROS 1 Melodic Morenia
    - [ ] ROS 2 Dashing Diademata
    - [ ] ROS 2 Eloquent Elusor
    - [x] etc (PLEASE, WRITE DOWN YOUR ROS VERSION HERE)
          ROS 1 Noetic

3. Which SBC(Single Board Computer) is working on TurtleBot3?

    - [x] Raspberry Pi 3
    - [ ] Intel Joule 570x
    - [ ] etc (PLEASE, WRITE DOWN YOUR SBC HERE)

4. Which OS you installed in SBC?

    - [ ] Raspbian
    - [ ] Ubuntu MATE 16.04 or later
    - [ ] Ubuntu Server 18.04 or later
    - [x] etc (PLEASE, WRITE DOWN YOUR OS)
           Ubuntu Server 20.04

5. Which OS you installed in Remote PC?

    - [ ] Ubuntu 16.04 LTS (Xenial Xerus)
    - [ ] Ubuntu 18.04 LTS (Bionic Beaver)
    - [ ] Linux Mint 18.x
    - [ ] Linux Mint 19.x
    - [x] etc (PLEASE, WRITE DOWN YOUR OS)
          Ubuntu 20.04

6. Write down software version and firmware version

    - Software version: [x.x.x]
    - Firmware version: [1.2.6]

7. Write down the commands you used in order
    roscore on remote pc, roslaunch bringup command for the raspberry
    

8. Copy and Paste your error message on terminal

    - No error message, all launches

9. Please, describe detailedly what difficulty you are in

    - When running the remote teleop, the motors are not recognized and running. When I open the diagnostics topic using rqt, only the IMU sensor and the Lidar Sensor seem to be recognized. The other items that should be there according to the manual are not recognized. 

I have installed ROS Noetic on the Pi, followed the tutorial on setting the usb, updated the firmware of the OpenCR mentioned in another thread. Seems that the OpenCR does not recognize the Dynamixel X and OpenCR button.

ashekim, 2020-09-24:
@pieterwolfert 

Hi. Thank you for your questions.

Please refer to the [Release for ROS noetic #624](https://github.com/ROBOTIS-GIT/turtlebot3/issues/624#issuecomment-697145610)
Thank you :)

pieterwolfert, 2020-09-24:
Yes, works, thank you :+1:

closed 2020-09-24 by pieterwolfert