ROBOTIS-GIT/turtlebot3 #643: cannot install ros  package

inwk, 2020-09-26:
ISSUE TEMPLATE ver. 0.3.0

1. Which TurtleBot3 you have?

    - [o] Burger
    - [ ] Waffle
    - [ ] Waffle Pi

2. Which ROS is working with TurtleBot3?

    - [o] ROS 1 Kinetic Kame
    - [ ] ROS 1 Melodic Morenia
    - [ ] ROS 2 Dashing Diademata
    - [ ] ROS 2 Eloquent Elusor
    - [ ] etc (PLEASE, WRITE DOWN YOUR ROS VERSION HERE)

3. Which SBC(Single Board Computer) is working on TurtleBot3?

    - [o] Raspberry Pi 3
    - [ ] Intel Joule 570x
    - [ ] etc (PLEASE, WRITE DOWN YOUR SBC HERE)

4. Which OS you installed in SBC?

    - [o] Raspbian
    - [ ] Ubuntu MATE 16.04 or later
    - [ ] Ubuntu Server 18.04 or later
    - [ ] etc (PLEASE, WRITE DOWN YOUR OS)

5. Which OS you installed in Remote PC?

    - [o] Ubuntu 16.04 LTS (Xenial Xerus)
    - [ ] Ubuntu 18.04 LTS (Bionic Beaver)
    - [ ] Linux Mint 18.x
    - [ ] Linux Mint 19.x
    - [ ] etc (PLEASE, WRITE DOWN YOUR OS)

6. Write down software version and firmware version

    - Software version: [x.x.x]
    - Firmware version: [x.x.x]

7. Write down the commands you used in order

    - sudo apt-get install ros-kinetic-image-transport ros-kinetic-cv-bridge ros-kinetic-vision-opencv python-opencv libopencv-dev ros-kinetic-image-proc

8. Copy and Paste your error message on terminal

    - HERE
[image]

9. Please, describe detailedly what difficulty you are in

    - we've followed robotis emanual turtlebot3 including raspbian and ros image file. At emanual 13.2 Install Autorace packages, we have problems with installing additional dependent packages only in sbc(rpi 3b+). We have changed mirror servers and used new sd cards, but nothing changed.

ROBOTIS-Will, 2020-09-29:
@inwk 
Hi,
I'm sorry about the confusion. If you have installed Raspbian image on your RPi3B+, you don't need to install them as all necessary packages are preinstalled already. Please ignore the emanual instruction for that section. We'll revise the manual soon.
Thank you.

inwk, 2020-10-02:
thanks. problem solved

closed 2020-10-05 by ashekim