ROBOTIS-GIT/turtlebot3 #654: Raspberry Pi Camera Not work

houhaol, 2020-10-16:
ISSUE TEMPLATE ver. 0.4.0

1. Which TurtleBot3 platform do you use?
Waffle Pi
    - [ ] Burger
    - [ ] Waffle
    - [ ] Waffle Pi

2. Which ROS is working with TurtleBot3?
ROS 1 Melodic
    - [ ] ROS 1 Kinetic Kame
    - [ ] ROS 1 Melodic Morenia
    - [ ] ROS 1 Noetic Ninjemys
    - [ ] ROS 2 Dashing Diademata
    - [ ] ROS 2 Eloquent Elusor
    - [ ] ROS 2 Foxy Fitzroy
    - [ ] etc (Please specify your ROS Version here)

3. Which SBC(Single Board Computer) is working on TurtleBot3?
Raspberry Pi 3B
    - [ ] Intel Joule 570x
    - [ ] Raspberry Pi 3B+
    - [ ] Raspberry Pi 4
    - [ ] etc (Please specify your SBC here)

4. Which OS you installed in SBC?
Ubuntu MATE 18.04
    - [ ] Raspbian distributed by ROBOTIS
    - [ ] Ubuntu MATE (16.04/18.04/20.04)
    - [ ] Ubuntu preinstalled server (18.04/20.04)
    - [ ] etc (Please specify your OS here)

5. Which OS you installed in Remote PC?
Ubuntu 18.04 LTS (WSL)
    - [ ] Ubuntu 16.04 LTS (Xenial Xerus)
    - [ ] Ubuntu 18.04 LTS (Bionic Beaver)
    - [ ] Ubuntu 20.04 LTS (Focal Fossa)
    - [ ] etc (Please specify your OS here)

6. Specify the software and firmware version(Can be found from Bringup messages)

    - Software version: [1.14.9]
    - Firmware version: [x.x.x]

7. Specify the commands or instructions to replicate the issue.
[REMOTE PC]
`roscore`
[TURTLEBOT1]
`roslaunch turtlebot3_bringup turtlebot3_rpicamera.launch`
or
`roslaunch raspicam_node camerav2_1280x960.launch`
[REMOTE PC]
`roslaunch turtlebot3_bringup turtlebot3_remote.launch`
[REMOTE PC]
`rqt_image_view`
Doesnt show any camera input (shows some gradient)

8. Copy and Paste the error messages on terminal.
Turtlebot3
`SUMMARY
========

PARAMETERS
 * /raspicam_node/camera_frame_id: raspicam
 * /raspicam_node/camera_id: 0
 * /raspicam_node/camera_info_url: package://raspica...
 * /raspicam_node/camera_name: camerav2_1280x960
 * /raspicam_node/enable_imv: False
 * /raspicam_node/enable_raw: False
 * /raspicam_node/framerate: 30
 * /raspicam_node/height: 960
 * /raspicam_node/private_topics: True
 * /raspicam_node/width: 1280
 * /rosdistro: melodic
 * /rosversion: 1.14.9

NODES
  /
    raspicam_node (raspicam_node/raspicam_node)

ROS_MASTER_URI=http://172.19.33.244:11311

process[raspicam_node-1]: started with pid [2240]
[ INFO] [1602841023.993522993]: splitter component done

[ INFO] [1602841024.137982342]: camera calibration URL: package://raspicam_node/camera_info/camerav2_1280x960.yaml
[ INFO] [1602841024.193425430]: Camera successfully calibrated from default file
[ INFO] [1602841024.193682407]: using default calibration URL
[ INFO] [1602841024.193869905]: camera calibration URL: file:///home/houhao/.ros/camera_info/camerav2_1280x960.yaml
[ INFO] [1602841024.194164954]: Unable to open camera calibration file [/home/houhao/.ros/camera_info/camerav2_1280x960.yaml]
[ WARN] [1602841024.194277817]: Camera calibration file /home/houhao/.ros/camera_info/camerav2_1280x960.yaml not found.
[ INFO] [1602841024.194404586]: No device specifc calibration found
[ INFO] [1602841036.676383260]: Starting video capture (1280, 960, 80, 30)

[ INFO] [1602841036.678593446]: Video capture started`

9. Please describe the issue in detail.
I have a Turtlebot3 waffle_pi and followed your camera installation instructions on this 
[Turtlebot3 e-manual](https://emanual.robotis.com/docs/en/platform/turtlebot3/appendix_raspi_cam/)

I can successfully run `raspistill' which would capture the scene.

But when i run the raspicam node, no image showed up and only give following figure when i run `rqt`
[image]

I found someone has same issue but no answer provided #457 

I also tried SLAM node but no image shows up.

Can you help me about that? Thanks a lot.

ashekim, 2020-10-20:
@houhaol 

Hi. I'm sorry for the late reply.
I'll check the issue and get back to you.

Thank you.

houhaol, 2020-10-20:
> @houhaol
> 
> Hi. I'm sorry for the late reply.
> I'll check the issue and get back to you.
> 
> Thank you.

Thanks. I made this camera work on my Ubuntu PC. But still failed on this WSL.

ROBOTIS-Will, 2020-11-02:
Hi @houhaol ,
ROS1 is focused on Linux and there could be some unexpected issues when running on virtual machine.
Thank you.

houhaol, 2020-11-02:
@ROBOTIS-Will 
I see. No problem. I'll use Linux instead. Thx

closed 2020-11-02 by houhaol