ROBOTIS-GIT/turtlebot3 #655: Turtlebot3 IMU and ODOM yaw drifts and is non-zero on Startup

MikeHallettUK, 2020-10-18:
ISSUE TEMPLATE ver. 0.4.0

1. Which TurtleBot3 platform do you use?

    - [x ] Burger
    - [ ] Waffle
    - [ ] Waffle Pi

2. Which ROS is working with TurtleBot3?

    - [ ] ROS 1 Kinetic Kame
    - [x ] ROS 1 Melodic Morenia
    - [ ] ROS 1 Noetic Ninjemys
    - [ ] ROS 2 Dashing Diademata
    - [ ] ROS 2 Eloquent Elusor
    - [ ] ROS 2 Foxy Fitzroy
    - [ ] etc (Please specify your ROS Version here)

3. Which SBC(Single Board Computer) is working on TurtleBot3?

    - [ ] Intel Joule 570x
    - [ ] Raspberry Pi 3B+
    - [x ] Raspberry Pi 4
    - [ ] etc (Please specify your SBC here)

4. Which OS you installed in SBC?

    - [ ] Raspbian distributed by ROBOTIS
    - [ ] Ubuntu MATE (16.04/18.04/20.04)
    - [ ] Ubuntu preinstalled server (18.04/20.04)
    - [x ] Raspbian Buster (Please specify your OS here)

5. Which OS you installed in Remote PC?

    - [ ] Ubuntu 16.04 LTS (Xenial Xerus)
    - [ ] Ubuntu 18.04 LTS (Bionic Beaver)
    - [ ] Ubuntu 20.04 LTS (Focal Fossa)
    - [ x] Windows 10 (Please specify your OS here)

6. Specify the software and firmware version(Can be found from Bringup messages)

    - Software version: [x.x.x]
    - Firmware version: [x.x.x]

7. Specify the commands or instructions to replicate the issue.

`subO = rospy.Subscriber('/odom', Odometry, get_odom)`

8. Copy and Paste the error messages on terminal.

With my Turtlebot3 I soon noticed that the IMU and ODOM yaw drifts over time while the robot is stationary, and that the yaw is non-zero on Startup and after a Reset.  Typical startup values of odom yaw are 5 to 9 degrees, drifting by 0.3 to 0.9 degrees per minute.

9. Please describe the issue in detail.

For example a typical plot of odom yaw over time while stationary looks like the attached graph.

[image]

- Is this "normal", or is my OpenCR board IMU defective ?

Clearly if the yaw is not zero aligned with the X axis on startup, then any steering goals based on the odom x,y frame coordinates will be biased.  Also, when returning to a stationary robot after some minutes, if the yaw has shifted significantly for no apparent reason, it can be disconcerting. 

In the OpenCR Arduino "firmware" provided by Robotis, the odom yaw is taken from the IMU: the wheel-yaw calculation is commented out, and there is no EKF blending of yaw sensor inputs.

To fix the zeroing issue, I added this after line 531 where "theta" is set (in Melodic version):

`  if (odom_pose[0]==0.0 and odom_pose[1]==0.0 and odom_pose[2]==0.0) {  // V1 change: init odom sets all these to 0.0
    last_theta = theta;  
  }`

For the drift, I add a variable in this calcOdometry function:

`  static double driftrate = 0.0;     // V1 change: to correct for IMU yaw drift`

... and some code after delta_theta is set:

`    if (wheel_r == 0.0 and wheel_l == 0.0) {                     // V1 change: e.g not moving....                          
      if (delta_theta > -0.0001 and delta_theta < 0.0001) {      // only if less than max. expected (radians per 30Hz) 
      driftrate = 0.01*delta_theta + 0.99*driftrate; }       // exponential moving average of driftrate 
  }  
  delta_theta = delta_theta - driftrate;                // end of V1 change: correct for IMU yaw drift`

The driftrate samples delta_theta when the robot is stationary, and smooths it out with an exponential moving average.  This gives a much more stable, near zero yaw.

Is this just my problem ?, or is this worth fixing in your code ?

Thanks

ashekim, 2020-10-20:
@MikeHallettUK 

Hi. I'm sorry for the late reply.
I'll check the issue and get back to you.

Thank you.

MikeHallettUK, 2020-12-12:
fyi: I have just upgraded to NOETIC and OpenCR 1.4.16 Arduino code... and this is still a problem.

My code does fix this: so I am closing this issue.

closed 2020-12-12 by MikeHallettUK

ROBOTIS-Will, 2020-12-14:
Hi @MikeHallettUK 
I'm sorry about the delayed response on this issue.
It looks like your board has some significant drift on yaw axis, which I didn't experience.
Could you check which IMU is implemented on you OpenCR? The IMU has been replaced from MPU9250(EOL) to ICM20648.
You might need a magnifying glasses or Lupe to read the part number on IMU.
Below(ICM20648) is what I took with my iPhone.
[image]

MikeHallettUK, 2020-12-14:
On my OpenCR board, the IMU is marked "MP92".  The date marked is "2018-09-04". I bought it June 2020.

Michael Hallett

lslabon, 2022-01-21:
Hi!

My Yaw is also drifting.
To smooth out the values is not the perfect solution for me.

I have noticed, that when i upload an example sketch from turtlebot (Read Roll Pitch Yaw) the values are not drifting :(

I tried to read the serial debug, but did not manage to get the readings ...

What is the best way to debug the OPENCR Board?

Thx

MikeHallettUK, 2022-01-23:
OPENCR Board is programmed in Arduino... so you can add your own "debug / print-out" code to show variable values... to serial, and e.g. monitor this serial-USB plugged in to your PC.

... or rely on the values that come through ROS and just monitor those like I did.