ROBOTIS-GIT/turtlebot3 #657: Bot is not moving

sayyidabeegam, 2020-10-23:
ISSUE TEMPLATE ver. 0.4.0

1. Which TurtleBot3 platform do you use?

    - [ ] Burger

2. Which ROS is working with TurtleBot3?

    - [ ] ROS 1 Kinetic Kame
 
3. Which SBC(Single Board Computer) is working on TurtleBot3?

    - [ ] Raspberry Pi 3B+

4. Which OS you installed in SBC?

    - [ ] Raspbian distributed by ROBOTIS

5. Which OS you installed in Remote PC?

    - [ ] Ubuntu 16.04 LTS (Xenial Xerus)

9. Please describe the issue in detail.

 We have followed all the steps and done everything given in emanual for turtlebot3.
After we have run roscore, launched the robot.launch as well as remote launch as per the instruction.
In terminal there is no error is showing.
But when we try to test some examples given in basic examples the bot is not moving with the command.
we have tried Teleoperation, Move using Interative Marker, Move to goal position .... and all, everything is launching properly and there is no error in terminal. we have checked the network connectivity by pinging that is also working. 
Also we have checked the hardware by the provided push button, there is no problem with that also.
Please help to solve this

ROBOTIS-Will, 2020-10-26:
Hi @sayyidabeegam ,
Could you please copy and paste each terminal output so that I can check if all process is running correctly?

ROBOTIS-Will, 2020-12-14:
This issue will be closed as there isn't recent activity. Please feel free to reopen when necessary.

closed 2020-12-14 by ROBOTIS-Will