ROBOTIS-GIT/turtlebot3 #658: After training process, how to run dqn_stages with the save_model?

is0437if, 2020-10-27:
For Turtlebot **[ROS 1]** Machine Learning: https://emanual.robotis.com/docs/en/platform/turtlebot3/machine_learning/

After training your stages, a model is saved in the **save_model** directory. For example: If I trained dqn_stage_4 for around 3000 episodes, it saves the trained model in "save_model" folder in several formats such as JSON and h5. Whenever trying to run the dqn_stage_4, it starts from the beginning. If I wanna test my trained model, how can I load the JSON or h5 saved model into code so that I can avoid training from the beginning every time running the code?
@ROBOTIS-David @ROBOTIS-Will 
[image]

ROBOTIS-David, 2020-10-27:
@is0437if 

Hi is0437if, thank you for getting my attention to you. 

I would like to transfer this issue to our TurtleBot3 issue section as this is a inquiry as to TurtleBot3. 

@ROBOTIS-Will  or @ashekim  is going to help your issue, but you might as well fill the template so that a member in the team can give you a better solution. 

Thank you :)

is0437if, 2020-10-27:
Thanks @ROBOTIS-David for forwarding my issue here! I am eagerly waiting for having a reply from @ashekim @ROBOTIS-Will

ashekim, 2020-10-29:
@is0437if 
HI. @is0437if . I'm sorry for the late reply.

I will check and get back to you. 
Thank you for your patience