ROBOTIS-GIT/turtlebot3 #660: E: Unable to locate package ros-noetic-turtlebot3

toebgen, 2020-10-28:
ISSUE TEMPLATE ver. 0.4.0

1. Which TurtleBot3 platform do you use?

    - [x] Burger
    - [ ] Waffle
    - [ ] Waffle Pi

2. Which ROS is working with TurtleBot3?

    - [ ] ROS 1 Kinetic Kame
    - [ ] ROS 1 Melodic Morenia
    - [x] ROS 1 Noetic Ninjemys
    - [ ] ROS 2 Dashing Diademata
    - [ ] ROS 2 Eloquent Elusor
    - [ ] ROS 2 Foxy Fitzroy
    - [ ] etc (Please specify your ROS Version here)

5. Which OS you installed in Remote PC?

    - [ ] Ubuntu 16.04 LTS (Xenial Xerus)
    - [ ] Ubuntu 18.04 LTS (Bionic Beaver)
    - [x] Ubuntu 20.04 LTS (Focal Fossa)
    - [ ] etc (Please specify your OS here)

7. Specify the commands or instructions to replicate the issue.

    - `docker run -it ros:noetic-ros-core`
    - (in container: ) `apt-get update && apt-get upgrade -y`
    - (in container: ) `apt-get install ros-noetic-turtlebot3`

8. Copy and Paste the error messages on terminal.

    - `E: Unable to locate package ros-noetic-turtlebot3`

9. Please describe the issue in detail.

    - Hello! I am trying to install the turtlebot3 binaries under ROS Noetic via previously pasted commands. However, I get the pasted error message. When checking http://repositories.ros.org/status_page/ros_noetic_default.html?q=turtlebot I can see, that only turtlebot3_msgs is available for installation right now. Does this mean the other packages (such as `turtlebot3`, `turtlebot3_bringup`, `turtlebot3_description`, ...) have not yet been released for Noetic? Or are these packages temporarily not available?
    - Thank you!

ashekim, 2020-10-29:
@toebgen 
Hi. @toebgen.  Thank you for your questions.
Unfortunately, other turtlebot3 packages have been released yet.
So, You need to download it from GitHub and build it.

Thank you :)

toebgen, 2020-10-29:
Hi @ashekim,
thank you for your response!
Just for further clarification: The remaining packages are all available in the testing repository already, and staged to be synced into the main repositories, i.e. for release, right? So in a matter of days they are likely available?
Thank you!

ashekim, 2020-10-29:
@toebgen 
Hi again.
Please refer to the [link](https://github.com/ros/rosdistro/pull/26863 )
Turtlebot packages will be included in the next release~ :)

toebgen, 2020-10-29:
Thank you @ashekim for clarification! :-)

closed 2020-10-29 by toebgen