ROBOTIS-GIT/turtlebot3 #663: ROS2 Node Architecture

tweigel-dev, 2020-11-03:
ISSUE TEMPLATE ver. 0.4.0

1. Which TurtleBot3 platform do you use?

    - [ x] Burger
    - [ ] Waffle
    - [ ] Waffle Pi

2. Which ROS is working with TurtleBot3?

    - [ ] ROS 1 Kinetic Kame
    - [ ] ROS 1 Melodic Morenia
    - [ ] ROS 1 Noetic Ninjemys
    - [x] ROS 2 Dashing Diademata
    - [ ] ROS 2 Eloquent Elusor
    - [ ] ROS 2 Foxy Fitzroy
    - [ ] etc (Please specify your ROS Version here)

3. Which SBC(Single Board Computer) is working on TurtleBot3?

    - [ ] Intel Joule 570x
    - [ x] Raspberry Pi 3B+
    - [ ] Raspberry Pi 4
    - [ ] etc (Please specify your SBC here)

4. Which OS you installed in SBC?

    - [ ] Raspbian distributed by ROBOTIS
    - [ ] Ubuntu MATE (16.04/18.04/20.04)
    - [x ] Ubuntu preinstalled server (18.04/20.04)
    - [ ] etc (Please specify your OS here)

5. Which OS you installed in Remote PC?

    - [ ] Ubuntu 16.04 LTS (Xenial Xerus)
    - [x ] Ubuntu 18.04 LTS (Bionic Beaver)
    - [ ] Ubuntu 20.04 LTS (Focal Fossa)
    - [ ] etc (Please specify your OS here)

6. Specify the software and firmware version(Can be found from Bringup messages)

    - Software version: [x.x.x]
    - Firmware version: [x.x.x]

7. Specify the commands or instructions to replicate the issue.

    - HERE

8. Copy and Paste the error messages on terminal.

    - HERE

9. Please describe the issue in detail.

 Hay,
Currently i am going to describe an architecture with custom nodes for my project using the turtlebot. therefore i want to know the following:
Is there a documentation where is described, which nodes exists and which node is publishing which topic or service?

This information would be very helpful to understand how the turtlebot works in detail. 
I didn't found this information at the e-manual of the turtlebot3.
I just found a image at an forum-entry [[here]](https://discourse.ros.org/t/tb3-introducing-ros2-tutorials/5959), but it isn't complete, isn't it?

ashekim, 2020-11-05:
@tweigel-dev 

Hi. I'm sorry for the late reply.
Unfortunately, there are no other sources.
and It would be good to check the rqt for a description of the nodes.

Thank you.

tweigel-dev, 2020-11-05:
@ashekim 
Thanks for the response!

what did u mean with rqt?

ashekim, 2020-11-06:
@tweigel-dev 
Hi Again :)
Please refer to the [18.1 Topic Monitor](https://emanual.robotis.com/docs/en/platform/turtlebot3/ros2_basic_operation/#topic_monitor) shown in the eManual.

Thank you

ROBOTIS-Will, 2020-12-14:
This issue will be closed as there isn't recent activity. Please feel free to reopen when necessary.

closed 2020-12-14 by ROBOTIS-Will