ROBOTIS-GIT/turtlebot3 #664: Strange Control Issues in ROS2 Foxy

smorad, 2020-11-03:
ISSUE TEMPLATE ver. 0.4.0

1. Which TurtleBot3 platform do you use?

    - [x] Burger
    - [ ] Waffle
    - [ ] Waffle Pi

2. Which ROS is working with TurtleBot3?

    - [ ] ROS 1 Kinetic Kame
    - [ ] ROS 1 Melodic Morenia
    - [ ] ROS 1 Noetic Ninjemys
    - [ ] ROS 2 Dashing Diademata
    - [ ] ROS 2 Eloquent Elusor
    - [x] ROS 2 Foxy Fitzroy
    - [ ] etc (Please specify your ROS Version here)

3. Which SBC(Single Board Computer) is working on TurtleBot3?

    - [ ] Intel Joule 570x
    - [x] Raspberry Pi 3B+
    - [ ] Raspberry Pi 4
    - [ ] etc (Please specify your SBC here)

4. Which OS you installed in SBC?

    - [ ] Raspbian distributed by ROBOTIS
    - [ ] Ubuntu MATE (16.04/18.04/20.04)
    - [ ] Ubuntu preinstalled server (18.04/20.04)
    - [ ] etc (Please specify your OS here)

5. Which OS you installed in Remote PC?

    - [ ] Ubuntu 16.04 LTS (Xenial Xerus)
    - [ ] Ubuntu 18.04 LTS (Bionic Beaver)
    - [x] Ubuntu 20.04 LTS (Focal Fossa)
    - [ ] etc (Please specify your OS here)

6. Specify the software and firmware version(Can be found from Bringup messages)

    - Software version: commit: 2e117c71859c50f7df56c4351c5b90678bb50a13 (2.1.0)
    - Firmware version: [x.x.x] (in the opencr directory: released_0.0.3.txt)

7. Specify the commands or instructions to replicate the issue.

   ` ros2 topic pub /turtlebot_12/cmd_vel geometry_msgs/msg/Twist '{linear: {x: 0.2}, angular: {z: -2.5}}' `

8.

```
ubuntu@ubuntu:~/opencr_update$ ros2 launch turtlebot3_bringup robot.launch.py                                                                                                                              
[INFO] [launch]: All log files can be found below /home/ubuntu/.ros/log/2020-10-29-17-45-19-782173-ubuntu-2478   
[INFO] [launch]: Default logging verbosity is set to INFO                                                                                                                                                  
/home/ubuntu/ros2/install/turtlebot3_bringup/share/turtlebot3_bringup/launch/robot.launch.py:77: UserWarning: The parameter 'node_executable' is deprecated, use 'executable' instead
  Node(   
/opt/ros/foxy/lib/python3.8/site-packages/launch_ros/actions/node.py:184: UserWarning: The parameter 'node_namespace' is deprecated, use 'namespace' instead                                       [0/1541]
  warnings.warn("The parameter 'node_namespace' is deprecated, use 'namespace' instead")             
urdf_file_name : turtlebot3_burger.urdf
/home/ubuntu/ros2/install/turtlebot3_bringup/share/turtlebot3_bringup/launch/turtlebot3_state_publisher.launch.py:47: UserWarning: The parameter 'node_executable' is deprecated, use 'executable' instead
  Node(
/home/ubuntu/ros2/install/turtlebot3_bringup/share/turtlebot3_bringup/launch/turtlebot3_state_publisher.launch.py:47: UserWarning: The parameter 'node_name' is deprecated, use 'name' instead
  Node(
/home/ubuntu/ros2/install/hls_lfcd_lds_driver/share/hls_lfcd_lds_driver/launch/hlds_laser.launch.py:46: UserWarning: The parameter 'node_executable' is deprecated, use 'executable' instead
  Node(
/home/ubuntu/ros2/install/hls_lfcd_lds_driver/share/hls_lfcd_lds_driver/launch/hlds_laser.launch.py:46: UserWarning: The parameter 'node_name' is deprecated, use 'name' instead
  Node(
[INFO] [robot_state_publisher-1]: process started with pid [2480]
[INFO] [hlds_laser_publisher-2]: process started with pid [2482]
[INFO] [turtlebot3_ros-3]: process started with pid [2484]
[hlds_laser_publisher-2] [INFO] [1603993520.199263474] [hlds_laser_publisher]: Init hlds_laser_publisher Node Main
[hlds_laser_publisher-2] [INFO] [1603993520.204887693] [hlds_laser_publisher]: port : /dev/ttyUSB0 frame_id : base_scan
[robot_state_publisher-1] [WARN] [1603993520.215789724] [robot_state_publisher]: No robot_description parameter, but command-line argument available.  Assuming argument is name of URDF file.  This backwa
rds compatibility fallback will be removed in the future.                                            
[robot_state_publisher-1] Parsing robot urdf xml string.
[robot_state_publisher-1] Link base_link had 5 children
[robot_state_publisher-1] Link caster_back_link had 0 children
[robot_state_publisher-1] Link imu_link had 0 children
[robot_state_publisher-1] Link base_scan had 0 children
[robot_state_publisher-1] Link wheel_left_link had 0 children
[robot_state_publisher-1] Link wheel_right_link had 0 children
[robot_state_publisher-1] [INFO] [1603993520.249741130] [robot_state_publisher]: got segment base_footprint
[robot_state_publisher-1] [INFO] [1603993520.249945453] [robot_state_publisher]: got segment base_link
[robot_state_publisher-1] [INFO] [1603993520.250008786] [robot_state_publisher]: got segment base_scan
[robot_state_publisher-1] [INFO] [1603993520.250059828] [robot_state_publisher]: got segment caster_back_link
[robot_state_publisher-1] [INFO] [1603993520.250100141] [robot_state_publisher]: got segment imu_link 
[robot_state_publisher-1] [INFO] [1603993520.250140141] [robot_state_publisher]: got segment wheel_left_link
[robot_state_publisher-1] [INFO] [1603993520.250179672] [robot_state_publisher]: got segment wheel_right_link
[turtlebot3_ros-3] [INFO] [1603993520.274323474] [turtlebot_12.turtlebot3_node]: Init TurtleBot3 Node Main
[turtlebot3_ros-3] [INFO] [1603993520.276907797] [turtlebot_12.turtlebot3_node]: Init DynamixelSDKWrapper
[turtlebot3_ros-3] [INFO] [1603993520.282156755] [DynamixelSDKWrapper]: Succeeded to open the port(/dev/ttyACM0)!
[turtlebot3_ros-3] [INFO] [1603993520.289000036] [DynamixelSDKWrapper]: Succeeded to change the baudrate!
[ERROR] [hlds_laser_publisher-2]: process has died [pid 2482, exit code 255, cmd '/home/ubuntu/ros2/install/hls_lfcd_lds_driver/lib/hls_lfcd_lds_driver/hlds_laser_publisher --ros-args -r __node:=hlds_las
er_publisher -r __node:=hlds_laser_publisher --params-file /tmp/launch_params_4xjhg7f_'].            
[turtlebot3_ros-3] [INFO] [1603993520.333793005] [turtlebot_12.turtlebot3_node]: Start Calibration of Gyro
[turtlebot3_ros-3] [INFO] [1603993525.334282118] [turtlebot_12.turtlebot3_node]: Calibration End
[turtlebot3_ros-3] [INFO] [1603993525.334526180] [turtlebot_12.turtlebot3_node]: Add Motors
[turtlebot3_ros-3] [INFO] [1603993525.335517170] [turtlebot_12.turtlebot3_node]: Add Wheels
[turtlebot3_ros-3] [INFO] [1603993525.336569618] [turtlebot_12.turtlebot3_node]: Add Sensors
[turtlebot3_ros-3] [INFO] [1603993525.346670399] [turtlebot_12.turtlebot3_node]: Succeeded to create battery state publisher
[turtlebot3_ros-3] [INFO] [1603993525.352027899] [turtlebot_12.turtlebot3_node]: Succeeded to create imu publisher
[turtlebot3_ros-3] [INFO] [1603993525.356190972] [turtlebot_12.turtlebot3_node]: Succeeded to create sensor state publisher
[turtlebot3_ros-3] [INFO] [1603993525.359063836] [turtlebot_12.turtlebot3_node]: Succeeded to create joint state publisher
[turtlebot3_ros-3] [INFO] [1603993525.359286701] [turtlebot_12.turtlebot3_node]: Add Devices
[turtlebot3_ros-3] [INFO] [1603993525.359368784] [turtlebot_12.turtlebot3_node]: Succeeded to create motor power server
[turtlebot3_ros-3] [INFO] [1603993525.363761805] [turtlebot_12.turtlebot3_node]: Succeeded to create reset server
[turtlebot3_ros-3] [INFO] [1603993525.368599878] [turtlebot_12.turtlebot3_node]: Succeeded to create sound server
[turtlebot3_ros-3] [INFO] [1603993525.372007847] [turtlebot_12.turtlebot3_node]: Run!
[turtlebot3_ros-3] [INFO] [1603993525.421226284] [turtlebot_12.diff_drive_controller]: Init Odometry
[turtlebot3_ros-3] [INFO] [1603993525.434765139] [turtlebot_12.diff_drive_controller]: Run!

9. Please describe the issue in detail.

Robot does not move correctly in ROS2. Running the command:
`ros2 topic pub /turtlebot_12/cmd_vel geometry_msgs/msg/Twist '{linear: {x: 0.2}, angular: {z: -2.5}}'` results in no motion. 

Reducing the magnitude of either linear or angular velocity makes it appear to move slowly:
`ros2 topic pub /turtlebot_12/cmd_vel geometry_msgs/msg/Twist '{linear: {x: 0.2}, angular: {z: -2.0}}'

Robot runs at full speed with only linear component:
`ros2 topic pub /turtlebot_12/cmd_vel geometry_msgs/msg/Twist '{linear: {x: 0.2}}'

But setting it to the maximum velocity results in no motion at all:
`ros2 topic pub /turtlebot_12/cmd_vel geometry_msgs/msg/Twist '{linear: {x: 0.22}}'

We did not see any of these issues in ROS1.

ashekim, 2020-11-05:
@smorad 
Hi. @smorad . Thank you for your questions.

We will check this issue and get back to you.
Thank you for your patience

smorad, 2020-11-19:
Hi @ashekim, I wanted to see if there has been any progress on this issue. We're blocked from collecting data on the turtlebots until we can run them at higher speeds. Ideally, we want them to move forward with maximum velocity, unless they are turning, in which case one wheel should still be rotating at the maximum rate.

smorad, 2020-12-02:
It's been almost a month, has there been any movement on this?

ROBOTIS-Will, 2020-12-15:
Hi @smorad 
When I tested with the latest turtlebot3 package for foxy, I didn't find the issue.
If you were running other teleop nodes while publishing the cmd_vel topic, this may hinder the expected operation as teleop will keep publishing the same topic with different value.

I didn't experience your issue when I ran your commands with a modification from `/turtlebot_12/cmd_vel` to `/cmd_vel`.

smorad, 2020-12-17:
@ROBOTIS-Will We were using our own control code. For verification of the error, we only ran `ros2 topic pub` to ensure there were no code bugs on our end. 

I will update to the latest commit and try again. If we still see the error, I'll upload a video of it.

ROBOTIS-Will, 2020-12-18:
@smorad Understood. Let me know whether the latest commit resolves the issue or not.
Happy holiday and happy new year!

smorad, 2020-12-18:
@ROBOTIS-Will updating to the most recent `origin/ros2` repos and rebuilding fixed the issue. Thanks for the assist, happy holidays!

closed 2020-12-18 by smorad