ROBOTIS-GIT/turtlebot3 #665: Set Turtlebot3 Burger to Not Collide

Quattrociocchi, 2020-11-05:
ISSUE TEMPLATE ver. 0.4.0

1. Which TurtleBot3 platform do you use?

    - [x] Burger
    - [ ] Waffle
    - [ ] Waffle Pi

2. Which ROS is working with TurtleBot3?

    - [ ] ROS 1 Kinetic Kame
    - [x] ROS 1 Melodic Morenia
    - [ ] ROS 1 Noetic Ninjemys
    - [ ] ROS 2 Dashing Diademata
    - [ ] ROS 2 Eloquent Elusor
    - [ ] ROS 2 Foxy Fitzroy
    - [ ] etc (Please specify your ROS Version here)

3. Which SBC(Single Board Computer) is working on TurtleBot3?

    - [ ] Intel Joule 570x
    - [x] Raspberry Pi 3B+
    - [ ] Raspberry Pi 4
    - [ ] etc (Please specify your SBC here)

4. Which OS you installed in SBC?

    - [x] Raspbian distributed by ROBOTIS
    - [ ] Ubuntu MATE (16.04/18.04/20.04)
    - [ ] Ubuntu preinstalled server (18.04/20.04)
    - [ ] etc (Please specify your OS here)

5. Which OS you installed in Remote PC?

    - [ ] Ubuntu 16.04 LTS (Xenial Xerus)
    - [x] Ubuntu 18.04 LTS (Bionic Beaver)
    - [ ] Ubuntu 20.04 LTS (Focal Fossa)
    - [ ] etc (Please specify your OS here)

6. Issue:

I would like to make the turtlebot an object that cannot collide with other objects. I have tried to add
`<surface>
        <contact>
          <collide_without_contact>true</collide_without_contact>
        </contact>
      </surface>`
inside the `<collision>` tag in the `turtlebot3_burger.urdf.xacro` file in the turtlebot3_description package (found in `opt/ros/melodic/share/`. However nothing changes when I launch a simulation. I have also tried changing something small like the color, but that also does not work. The turtlebot packages are installed on the system level, so maybe my changes are being ignored?

Thank you for any help.

ashekim, 2020-11-06:
@Quattrociocchi 

Hi. @Quattrociocchi . Thank you for your questions.
I'll check and get back to you.

Thank you.

ashekim, 2020-11-06:
@Quattrociocchi

Did you install it as a binary?
Download as a source file from github, and please build and run after modify the code.

After testing, please let me know the result.
Thank you

ROBOTIS-Will, 2020-12-14:
This issue will be closed as there isn't recent activity. Please feel free to reopen when necessary.

closed 2020-12-14 by ROBOTIS-Will

Quattrociocchi, 2021-02-23:
Apologies @ashekim for the severe delay. Life in the pandemic.

I tested it really quickly this morning and it works! It was one of those, "duh you have to recompile" moments.

Thank you very much!
Q