ROBOTIS-GIT/turtlebot3 #666: RViz not receiving LaserScan data

tirafesi, 2020-11-05:
I'm trying to run turtlebot3 inside a docker container based on the official ros:foxy image. I'm following the guide [turtlebot3-simulation-using-gazebo](https://emanual.robotis.com/docs/en/platform/turtlebot3/ros2_simulation/#turtlebot3-simulation-using-gazebo).

The problem I'm having is that RViz is not displaying the laser scan data.

1. `ros2 launch turtlebot3_gazebo empty_world.launch.py`
2. Place some random object to test laser scan.
3. `ros2 launch turtlebot3_bringup rviz2.launch.py`

As you can see in the images below, although RViz is correctly displaying the robot model, TF and odometry, it does not display the laser scan. In fact, the laser scan status reads *Showing [0] points from [0] messages*.

[image]
[image]

ashekim, 2020-11-06:
@tirafesi 
Hi. @tirafesi . Thank you for your questions.

In Rviz's Display window, change LaserScan -> Topic -> Reliability Policy from Reliable to Best Effort.

Please refer to the image below
[image]

Thank you

tirafesi, 2020-11-06:
Thanks @ashekim , that solved the problem!

ashekim, 2020-11-09:
@tirafesi 
I would greatly appreciate it. :)
If you have any question or suggest, please ask to me any time. Thanks :)

closed 2020-11-09 by ashekim

ZengJunTao111, 2024-03-27:
Hello,@siferati,may I ask if your laser model is modified by turtlebot3_burger.urdf.xacro? Or is there an open source model?I don't have the laser model in this file. I would appreciate it if you could tell me how to add.
[image]