ROBOTIS-GIT/turtlebot3 #667: ModuleNotFoundError in Machine Learning Tutorial using ROS2 Dashing

xiaofeng-fan, 2020-11-06:
Hi, I am following this [tutorial](https://emanual.robotis.com/docs/en/platform/turtlebot3/ros2_machine_learning/) and I have setup ROS2 Dashing on Ubuntu 18.04.  When I run the command `ros2 run turtlebot3_dqn dqn_environment` at [24.3.1 stage 1](https://emanual.robotis.com/docs/en/platform/turtlebot3/ros2_machine_learning/#stage-1-no-obstacle), it returns this error:

```
flint@flint-Alienware-m17-R3:~/turtlebot3_ws$ ros2 run turtlebot3_dqn dqn_environment
Traceback (most recent call last):
  File "/opt/ros/dashing/lib/python3.6/site-packages/rosidl_generator_py/import_type_support_impl.py", line 40, in import_type_support
    return importlib.import_module(module_name, package=pkg_name)
  File "/usr/lib/python3.6/importlib/__init__.py", line 126, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 994, in _gcd_import
  File "<frozen importlib._bootstrap>", line 971, in _find_and_load
  File "<frozen importlib._bootstrap>", line 953, in _find_and_load_unlocked
ModuleNotFoundError: No module named 'turtlebot3_msgs.turtlebot3_msgs_s__rosidl_typesupport_c'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/flint/turtlebot3_ws/install/turtlebot3_dqn/lib/turtlebot3_dqn/dqn_environment", line 11, in <module>
    load_entry_point('turtlebot3-dqn', 'console_scripts', 'dqn_environment')()
  File "/home/flint/turtlebot3_ws/build/turtlebot3_dqn/turtlebot3_dqn/dqn_environment/dqn_environment.py", line 250, in main
    dqn_environment = DQNEnvironment()
  File "/home/flint/turtlebot3_ws/build/turtlebot3_dqn/turtlebot3_dqn/dqn_environment/dqn_environment.py", line 92, in __init__
    self.dqn_com_server = self.create_service(Dqn, 'dqn_com', self.dqn_com_callback)
```

It looks like error from the packages of the tutorials. Can give any hints? Thanks.

ROBOTIS-Will, 2021-01-29:
Closing this issue as there isn't recent activity. Please feel free to reopen when necessary.

closed 2021-01-29 by ROBOTIS-Will

Ashwath12052004, 2024-06-07:
need help with the first error
was tryna simulate dqn using turtlebot3 in gazebo11 using os foxy encountered this error specifically.

ros2 run turtlebot3_drl gazebo_goals
running on stage: 4, dynamic goals enabled: False
Traceback (most recent call last):
  File "/opt/ros/foxy/lib/python3.8/site-packages/rosidl_generator_py/import_type_support_impl.py", line 46, in import_type_support
    return importlib.import_module(module_name, package=pkg_name)
  File "/usr/lib/python3.8/importlib/__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1014, in _gcd_import
  File "<frozen importlib._bootstrap>", line 991, in _find_and_load
  File "<frozen importlib._bootstrap>", line 973, in _find_and_load_unlocked
ModuleNotFoundError: No module named 'turtlebot3_msgs.turtlebot3_msgs_s__rosidl_typesupport_c'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/ashwath/turtlebot3_drlnav/install/turtlebot3_drl/lib/turtlebot3_drl/gazebo_goals", line 11, in <module>
    load_entry_point('turtlebot3-drl', 'console_scripts', 'gazebo_goals')()
  File "/home/ashwath/turtlebot3_drlnav/build/turtlebot3_drl/turtlebot3_drl/drl_gazebo/drl_gazebo.py", line 263, in main
    drl_gazebo = DRLGazebo()
  File "/home/ashwath/turtlebot3_drlnav/build/turtlebot3_drl/turtlebot3_drl/drl_gazebo/drl_gazebo.py", line 74, in __init__
    self.task_succeed_server    = self.create_service(RingGoal, 'task_succeed', self.task_succeed_callback)
  File "/opt/ros/foxy/lib/python3.8/site-packages/rclpy/node.py", line 1300, in create_service
    check_for_type_support(srv_type)
  File "/opt/ros/foxy/lib/python3.8/site-packages/rclpy/type_support.py", line 29, in check_for_type_support
    msg_type.__class__.__import_type_support__()
  File "/home/ashwath/turtlebot3_drlnav/install/turtlebot3_msgs/lib/python3.12/site-packages/turtlebot3_msgs/srv/_ring_goal.py", line 273, in __import_type_support__
    module = import_type_support('turtlebot3_msgs')
  File "/opt/ros/foxy/lib/python3.8/site-packages/rosidl_generator_py/import_type_support_impl.py", line 48, in import_type_support
    raise UnsupportedTypeSupport(pkg_name)
rosidl_generator_py.import_type_support_impl.UnsupportedTypeSupport: Could not import 'rosidl_typesupport_c' for package 'turtlebot3_msgs'