ROBOTIS-GIT/turtlebot3 #668: cannot launch node of type [hls_lfcd_lds_driver/hlds_laser_publisher]: hls_lfcd_lds_driver

minaaa77, 2020-11-06:
ISSUE TEMPLATE ver. 0.4.0

1. Which TurtleBot3 platform do you use?

 Burger
  

2. Which ROS is working with TurtleBot3?

    -  ROS 1 Kinetic Kame
 

3. Which SBC(Single Board Computer) is working on TurtleBot3?

  
    - Raspberry Pi 3B

   

4. Which OS you installed in SBC?

   
    - [ ] Ubuntu MATE (16.04/18.04/20.04)
   
 
5. Which OS you installed in Remote PC?

    - [ ] Ubuntu 16.04 LTS (Xenial Xerus)
  
6. Specify the software and firmware version(Can be found from Bringup messages)

    ubuntu 16.04.7 LTS
    - Firmware version: 1.157.23

7. Specify the commands or instructions to replicate the issue.

   $ roslaunch turtlebot3_bringup turtlebot3_robot.launch

8. Copy and Paste the error messages on terminal.
(remote pc error)
[image]

Turtlebot3 error 
[image]

9. Please describe the issue in detail.

ashekim, 2020-11-09:
@minaaa77 

Hi. Thank you for your questions.

Could you enter the command below and run it again?
`$ sudo chmod a+rw /dev/ttyACM0`

Please let me know the test result :)

ROBOTIS-Will, 2020-12-14:
This issue will be closed as there isn't recent activity. Please feel free to reopen when necessary.

closed 2020-12-14 by ROBOTIS-Will