ROBOTIS-GIT/turtlebot3 #669: Wheels locked aftering blowing up the OpenCR and putting on a new one

stevensu1838, 2020-11-09:
ISSUE TEMPLATE ver. 0.4.0

1. Which TurtleBot3 platform do you use?
   - [ ] Waffle

2. Which ROS is working with TurtleBot3?
    - [ ] ROS 1 Kinetic Kame

3. Which SBC(Single Board Computer) is working on TurtleBot3?
    - [ ] Raspberry Pi 3B
4. Which OS you installed in SBC?
    - [ ] Ubuntu MATE (16.04/18.04/20.04)
5. Which OS you installed in Remote PC?
9. Please describe the issue in detail.
Hi all,

we burnt the OpenCR board on Waffle by plugging the wrong pins into Rasperry Pi and we saw smokes coming out of the OpenCR. Then we replaced the OpenCR with a new one. However, the wheels(motors) locked and stopped turning. Could you please help us out?

Btw, when we tested the motors with the new OpenCR as below, the wheels(motors) work well. However, apart from doing this, the wheels are totally locked.

[image]

ashekim, 2020-11-09:
@stevensu1838 

Hi. Thank you for your questions. :)

Please refer to the link below and set OpenCR again.
[6.3.1. OpenCR Firmware Upload for TB3](https://emanual.robotis.com/docs/en/platform/turtlebot3/opencr_setup/#opencr-firmware-upload-for-tb3)

Thank you.

stevensu1838, 2020-11-09:
Hi all,
I reconfigured the OpenCR as you instructed again. The wheels on my waffle still locked. I even set the same OpenCR to burger mode and tested this OpenCR on my burger, the wheels of burger locked as well. Could you please help me out?

ashekim, 2020-11-24:
@stevensu1838 
Hi. I'm sorry for the late reply.
Does being locked mean you have a torque but don't move?
Is it not moving when running the teleoperation node?

Would be grateful if you could give me more information
Thank you.

closed 2021-03-10 by ROBOTIS-Will