ROBOTIS-GIT/turtlebot3 #670: Unable to get lidar data in ROS2

sjev, 2020-11-10:
After reading the documentation and hours of googling through issues ([like this one](https://github.com/ros2/ros2_documentation/issues/569)), I'm still not able to read the lidar data published on `/scan` topic.

How to reproduce:
1. bring up tutrtlebot
2. start `rqt` on remote pc. -> there are no updates from the scan sensor (below is a screenshot from the emanual, showing *unknown* update frequency for the data. Nothing is coming in)
[image]
3. check topic with `ros2 topic echo /scan` -> **messages are being received**
4. subscribe from python `sub = node.create_subscription(
        String, 'scan', chatter_callback, qos_profile=qos_profile_sensor_data)` -> **no messages**

I'm at a loss here...

ashekim, 2020-11-17:
@sjev 
Hi. @sjev 
I'm sorry for the late reply.

I'will check this issue and get back to you.
Thank you for your patience

ROBOTIS-Will, 2020-12-01:
This is related to the ROS2 QoS setting.
In RViz2, setting Reliability Policy of LaserScan topic to "Best Effort" will draw LDS sensor data, but rqt seems not reflecting this change.
[image]

ROBOTIS-Will, 2021-01-29:
Closing this issue as there isn't recent activity. Please feel free to reopen when necessary.

closed 2021-01-29 by ROBOTIS-Will

sjev, 2021-01-31:
"One of our team member will address your issue as soons as possible."

Please deliver on your promise instead of waiting for more than 2 months and then closing this thread due to your own inactivity.

ROBOTIS-Will, 2021-02-01:
Hi @sjev 
Thank you for the reminder, and sorry about not informing the update on this.
The issue is mainly related to the QoS setting of rqt.
https://github.com/ros-visualization/rqt/issues/187

The updated rviz setting was available in the latest TurtleBot3 package.
https://github.com/ROBOTIS-GIT/turtlebot3/commit/f6078c11cf28f2324514f467198a2d11eb116ae9
Thank you.