ROBOTIS-GIT/turtlebot3 #671: where does the map_builder.lua and trajectory_builder.lua exists which are included in turtlebot3_lds_2d.lua file

JinXiangLai, 2020-11-11:
I am working with waffle_pi, but I can not find the map_builder.lua and trajectory_builder.lua file, they are include on the turtlebot3_lds_2d.lua file. Could anyone tell me where does the two files exists?

`include "map_builder.lua"    #I can not find it out
include "trajectory_builder.lua" #also I can not find this file out

options = {
  map_builder = MAP_BUILDER,
  trajectory_builder = TRAJECTORY_BUILDER,
  map_frame = "map",
  tracking_frame = "imu_link", -- imu_link, If you are using gazebo, use 'base_footprint' (libgazebo_ros_imu's bug)
  published_frame = "odom",
  odom_frame = "odom",
  provide_odom_frame = false,
  publish_frame_projected_to_2d = false,
  use_odometry = true,
  use_nav_sat = false,
  use_landmarks = false,
  num_laser_scans = 1,
  num_multi_echo_laser_scans = 0,
  num_subdivisions_per_laser_scan = 1,
  num_point_clouds = 0,
  lookup_transform_timeout_sec = 0.2,
  submap_publish_period_sec = 0.3,
  pose_publish_period_sec = 5e-3,
  trajectory_publish_period_sec = 30e-3,
  rangefinder_sampling_ratio = 1.,
  odometry_sampling_ratio = 0.1,
  fixed_frame_pose_sampling_ratio = 1.,
  imu_sampling_ratio = 0.1,
  landmarks_sampling_ratio = 1.,
}

MAP_BUILDER.use_trajectory_builder_2d = true

TRAJECTORY_BUILDER_2D.min_range = 0.1
TRAJECTORY_BUILDER_2D.max_range = 3.5
TRAJECTORY_BUILDER_2D.missing_data_ray_length = 3.
TRAJECTORY_BUILDER_2D.use_imu_data = true
TRAJECTORY_BUILDER_2D.use_online_correlative_scan_matching = true
TRAJECTORY_BUILDER_2D.motion_filter.max_angle_radians = math.rad(0.1)

POSE_GRAPH.constraint_builder.min_score = 0.65
POSE_GRAPH.constraint_builder.global_localization_min_score = 0.7

return options`

ashekim, 2020-11-16:
Hi. @JinXiangLai . 

If you installed it with ros2,
the map_builder.lua and trajectory_builder.lua files are in the [cartograper](https://github.com/ros2/cartographer/tree/ros2/configuration_files) package installed in this [section](https://emanual.robotis.com/docs/en/platform/turtlebot3/ros2_setup/#install-ros-2-dependent-packages).

If you installed with ros1 Please refer to this [ros wiki_cartographer](http://wiki.ros.org/cartographer)

Thank you.

MikeHallettUK, 2020-12-12:
at cmd line > `roscd cartographer`

then these files are in /configuration_files/ folder at that location.

closed 2021-01-29 by ROBOTIS-Will