ROBOTIS-GIT/turtlebot3 #674: Bringup a turtlebot3 not functioning

Mdbirley, 2020-11-13:
ISSUE TEMPLATE ver. 0.4.0

1. Which TurtleBot3 platform do you use?

    - [ ] Burger
    - [ ] Waffle
    - [ x] Waffle Pi

2. Which ROS is working with TurtleBot3?

    - [ x] ROS 1 Kinetic Kame
    - [ ] ROS 1 Melodic Morenia
    - [ ] ROS 1 Noetic Ninjemys
    - [ ] ROS 2 Dashing Diademata
    - [ ] ROS 2 Eloquent Elusor
    - [ ] ROS 2 Foxy Fitzroy
    - [ ] etc (Please specify your ROS Version here)

3. Which SBC(Single Board Computer) is working on TurtleBot3?

    - [ ] Intel Joule 570x
    - [ x] Raspberry Pi 3B+
    - [ ] Raspberry Pi 4
    - [ ] etc (Please specify your SBC here)

4. Which OS you installed in SBC?

    - [x ] Raspbian distributed by ROBOTIS
    - [ ] Ubuntu MATE (16.04/18.04/20.04)
    - [ ] Ubuntu preinstalled server (18.04/20.04)
    - [ ] etc (Please specify your OS here)

5. Which OS you installed in Remote PC?

    - [x ] Ubuntu 16.04 LTS (Xenial Xerus)
    - [ ] Ubuntu 18.04 LTS (Bionic Beaver)
    - [ ] Ubuntu 20.04 LTS (Focal Fossa)
    - [ ] etc (Please specify your OS here)

6. Specify the software and firmware version(Can be found from Bringup messages)

    - Software version: [x.x.x]
    - Firmware version: [x.x.x]
sorry Bringup not working

7. Specify the commands or instructions to replicate the issue.

    see below

8. Copy and Paste the error messages on terminal.

 while processing /home/pi/catkin_ws/src/turtlebot3/turtlebot3_bringup/launch/includes/description.launch.xml:
Invalid <arg> tag: turtlebot3_description
ROS path [0]=/opt/ros/kinetic/share/ros
ROS path [1]=/home/pi/catkin_ws/src
ROS path [2]=/opt/ros/kinetic/share. 

Arg xml is <arg default="$(find xacro)/xacro --inorder '$(find turtlebot3_description)/urdf/turtlebot3_$(arg model).urdf.xacro'" name="urdf_file"/>
The traceback for the exception was written to the log file

9. Please describe the issue in detail.
I run score on my ubuntu PC and this command on the Raspberry pi and get the error in section 8

   roslaunch turtlebot3_bringup turtlebot3_robot.launch

ashekim, 2020-11-16:
@Mdbirley 
Hi. @Mdbirley . Thank you for your questions.

Please refer to the link [Install Dependent Packages on TurtleBot PC](https://emanual.robotis.com/docs/en/platform/turtlebot3/raspberry_pi_3_setup/#3-install-dependent-packages-on-turtlebot-pc) 
and reinstall the turtlebot3 package.

Thank you.

Mdbirley, 2020-11-16:
hi
did as you asked
the 3 git clones were already installed
also it could not locate the 2 ros packages
I was applying the commands to the TurtleBot raspberry pi

Warmest Regards
 
Max Birley
 +1 830 998 9914
skype mdbirley
Please take a second to watch these videos
https://www.youtube.com/watch?v=5pK3G20UhYI&feature=youtu.be
https://youtu.be/b1RxYSp-Qf8
http://www.autonomousmowingrobots.com/

Mdbirley, 2020-11-16:
I have been trying to get this working all weekend and it is driving me nuts

the turtlebot moves when I use the game controller
Ros on my PC seems to be working ok. I can call turtle sim and make it drive around

I know that I need to call rescore in a laptop remote pc window

however then I get confused. 

when I run roslaunch…………robot.launch  from the remote PC I get error when I ssh into the raspberry pi it executes

however then subsequent commands throw errors.

let me know what info you require   

Warmest Regards
 
Max Birley
 +1 830 998 9914
skype mdbirley
Please take a second to watch these videos
https://www.youtube.com/watch?v=5pK3G20UhYI&feature=youtu.be
https://youtu.be/b1RxYSp-Qf8
http://www.autonomousmowingrobots.com/

ashekim, 2020-11-17:
Hi. again. :)

It seems like an error that occurs when running roslaunch turtlebot3_bringup turtlebot3_remote.launch on the PC.

Please refer to the link below
[turtlebot3_remote.launch](https://github.com/ROBOTIS-GIT/turtlebot3/blob/ba10294ccb02891cd6af456bd4662e77813e3245/turtlebot3_bringup/launch/turtlebot3_remote.launch#L5)

Have you ever run roslaunch turtlebot3_bringup turtlebot3_remote.launch?

If so, you are supposed to type TB3_MODEL.
`$ export TURTLEBOT3_MODEL=waffle_pi`

Please Give it a try and let me know the result.
Thank you.

Mdbirley, 2020-11-18:
Hi
I now have the instructions working well up to the end of bring it up

I am now trying to run Move with Interactive Makers and I get this message

[ WARN] [1605664176.416393575]: The STL file 'package://turtlebot3_description/meshes/bases/waffle_pi_base.stl' is malformed. It starts with the word 'solid', indicating that it's an ASCII STL file, but it does not contain the word 'endsolid' so it is either a malformed ASCII STL file or it is actually a binary STL file. Trying to interpret it as a binary STL file instead.
[ WARN] [1605664176.657963960]: The STL file 'package://turtlebot3_description/meshes/sensors/lds.stl' is malformed. It starts with the word 'solid', indicating that it's an ASCII STL file, but it does not contain the word 'endsolid' so it is either a malformed ASCII STL file or it is actually a binary STL file. Trying to interpret it as a binary STL file instead.

I seems to get alog of errors or warnings in Rviz

also when I get into RViz I loose teleop so I can not drive the robot and see the model move

or make the image move and see the physical robot wheels move

I am running the following

max@HP430:~$roscore

from the TB frame
max@HP430:~$ roslaunch turtlebot3_bringup turtlebot3_remote.launch

max@HP430:~$ export TURTLEBOT3_MODEL=waffle_pi

max@HP430:~$ roslaunch turtlebot3_example interactive_markers.launch

max@HP430:~$ rosrun rviz rviz -d `rospack find turtlebot3_example`/rviz/turtlebot3_interactive.rviz

can I also use this to get some movement?

max@HP430:~$ roslaunch turtlebot3_teleop turtlebot3_teleop_key.launch
 regards

On 11/16/20 8:37 PM, ROBOTIS-Ashe wrote:

Hi. again. :)

It seems like an error that occurs when running roslaunch turtlebot3_bringup turtlebot3_remote.launch on the PC.

Please refer to the link below
turtlebot3_remote.launch<https://github.com/ROBOTIS-GIT/turtlebot3/blob/ba10294ccb02891cd6af456bd4662e77813e3245/turtlebot3_bringup/launch/turtlebot3_remote.launch#L5>

Have you ever run roslaunch turtlebot3_bringup turtlebot3_remote.launch?

If so, you are supposed to type TB3_MODEL.
$ export TURTLEBOT3_MODEL=waffle_pi

Please Give it a try and let me know the result.
Thank you.

Mdbirley, 2020-11-18:
Warmest Regards
 
Max Birley
 +1 830 998 9914
skype mdbirley
Please take a second to watch these videos
https://www.youtube.com/watch?v=5pK3G20UhYI&feature=youtu.be
https://youtu.be/b1RxYSp-Qf8
http://www.autonomousmowingrobots.com/

Mdbirley, 2020-11-18:
Eventhough I have the interactive markers working and I am at last controlling the bot from Ross I still get the error message

any thoughts

Mdbirley, 2020-11-18:
I am trying to put together my own complete instructions for each of your examples.
Can you help me with the basic examples for the Point operation and Patrol as I think I need a RVIZ command
Please remember at first us beginners need every instruction otherwise it fails

Also is there a log file stored in one place that I can send so you can trouble shoot the problem?

By the way the you tube videos are too short, they need a voice over, the commands are entered too fast, you can not read the commands easily

regards

Bringup Example

roscore
$ roslaunch turtlebot3_bringup turtlebot3_robot.launch 

export TURTLEBOT3_MODEL=waffle_pi
$ roslaunch turtlebot3_bringup turtlebot3_remote.launch$ 

rosrun rviz rviz -d `rospack find turtlebot3_description`/rviz/model.rviz

BASIC EXAMPLES
A Interactive markers Example

1 max@HP430:~$roscore
2 from the TB frame SSH into raspberry pi
3 Raspberry:~$ roslaunch turtlebot3_bringup turtlebot3_robot.launch
4 max@HP430:~$ export TURTLEBOT3_MODEL=waffle_pi
 max@HP430:~$ roslaunch turtlebot3_bringup turtlebot3_remote.launch
5 max@HP430:~$ roslaunch turtlebot3_example interactive_markers.launch
6 max@HP430:~$ rosrun rviz rviz -d `rospack find turtlebot3_example`/rviz/turtlebot3_interactive.rviz

B Point Operation
 max@HP430:~$roscore
2 from the TB frame SSH into raspberry pi
3 Raspberry:~$ roslaunch turtlebot3_bringup turtlebot3_robot.launch
4 max@HP430:~$ export TURTLEBOT3_MODEL=waffle_pi
 max@HP430:~$ roslaunch turtlebot3_bringup turtlebot3_remote.launch
$ roslaunch turtlebot3_example turtlebot3_pointop_key.launch

I need an RVIS command here

Mdbirley, 2020-11-20:
Hi
I do not have the standard lidar that comes with the turtlebot kit I have the rplidar A1.

Can you help me with the hollowing instruction as they are not working

If I use the standard bringup commands then the lidar does not spin

If I uses the commands below without the bring up then the command is not recognised

getting this working will add a lot of functionality to the turtlebot

regards

How to build rplidar ros package

1) Clone this project to your catkin's workspace src folder 2) Running catkin_make to build rplidarNode and rplidarNodeClient

How to run rplidar ros package

There're two ways to run rplidar ros package

I. Run rplidar node and view in the rviz

roslaunch rplidar_ros view_rplidar.launch (for RPLIDAR A1/A2) , roslaunch rplidar_ros view_rplidar_a3.launch (for RPLIDAR A3) or roslaunch rplidar_ros view_rplidar_s1.launch (for RPLIDAR S1)

You should see rplidar's scan result in the rviz.

II. Run rplidar node and view using test application

roslaunch rplidar_ros rplidar.launch (for RPLIDAR A1/A2) , roslaunch rplidar_ros rplidar_a3.launch (for RPLIDAR A3) or roslaunch rplidar_ros rplidar_s1.launch (for RPLIDAR S1)

rosrun rplidar_ros rplidarNodeClient

You should see rplidar's scan result in the console

Notice: the different is serial_baudrate between A1/A2 and A3/S1

RPLidar frame

RPLidar frame must be broadcasted according to picture shown in rplidar-frame.png

ROBOTIS-Will, 2020-12-01:
Hi @Mdbirley ,
Unfortunately, we haven't tested RPLIDAR A1.
Have you successfully launched and get the lidar data from given instruction from [rplidar ros wiki page](http://wiki.ros.org/rplidar)?
If you are getting the data from Raspberry Pi, you don't need to launch turtlebot3_lidar.launch in the turtlebot3_robot.launch, but you should replace the launch file below to the one that launches RPLIDAR A1.
https://github.com/ROBOTIS-GIT/turtlebot3/blob/master/turtlebot3_bringup/launch/turtlebot3_robot.launch#L8
You might get more useful information from [ROS Answers](https://answers.ros.org/questions/) for customizing TurtleBot3.
Thank you.

ROBOTIS-Will, 2021-01-29:
Closing this issue as there isn't recent activity. Please feel free to reopen when necessary.

closed 2021-01-29 by ROBOTIS-Will

Mdbirley, 2021-11-22:
Hi 

My turtlebot3 waffle is working well and I appreciate the work you guys have done to make this happen.

Now I want to make another larger robot and  I have a few questions for you. This robot will have large wheels say 6 inch and be able to drive all around my house without having problems with the carpet like the waffle does

Can you recommend a motor controller and motor that can take 6 inch wheels that have good encoders built in that can be used.

The laptop I use for the TurtleBot3 has ubuntu 16.04 loaded on it with ros kinetic 
The raspberry pi 3B+ is set up with ubuntu 16.04 and kinetic as well

Regarding the releases of ubuntu and ros as time has moved on and I am making a larger robot should I use later releases of ubuntu and ros or stay with the 16.04 and kinetic versions as I know they work?

Also do the versions on the remote laptop and raspberry pi 3B+ (SBC) of ubuntu and ros need to be the same  or can they be different ?
this would be like, I keep ubuntu 16.04 on the laptop and update the raspberry pi to sat 18.04 or 21.1. Or must they be the same on both

regards
Max

> Hi. again. :)
>
> It seems like an error that occurs when running roslaunch turtlebot3_bringup turtlebot3_remote.launch on the PC.
>
> Please refer to the link below
> turtlebot3_remote.launch [...]

Warmest Regards
 
Max Birley

Check this out
https://youtu.be/oGikNKcbI0

Please respond to all emails from ***@***.*** and reply to  ***@***.*** as ***@***.*** will not be operable by late November

Cell +1 830 998 9914

Our Web Site is you no mow
http://www.younomow.com/

ROBOTIS-Will, 2021-11-25:
Hi @Mdbirley 

Since your last inquiry is not relevant to the topic of this thread, I've referred it in a new thread that I created #802 
Please continue the discussion in that thread.
Thank you.