ROBOTIS-GIT/turtlebot3 #676: SLAM Interference in /tf Topic with Multiple TurtleBot3

rpmccarter, 2020-11-17:
ISSUE TEMPLATE ver. 0.4.0

1. Which TurtleBot3 platform do you use?

    - [x] Burger
    - [ ] Waffle
    - [ ] Waffle Pi

2. Which ROS is working with TurtleBot3?

    - [x] ROS 1 Kinetic Kame
    - [ ] ROS 1 Melodic Morenia
    - [ ] ROS 1 Noetic Ninjemys
    - [ ] ROS 2 Dashing Diademata
    - [ ] ROS 2 Eloquent Elusor
    - [ ] ROS 2 Foxy Fitzroy
    - [ ] etc (Please specify your ROS Version here)

3. Which SBC(Single Board Computer) is working on TurtleBot3?

    - [ ] Intel Joule 570x
    - [x] Raspberry Pi 3B+
    - [ ] Raspberry Pi 4
    - [ ] etc (Please specify your SBC here)

4. Which OS you installed in SBC?

    - [x] Raspbian distributed by ROBOTIS
    - [ ] Ubuntu MATE (16.04/18.04/20.04)
    - [ ] Ubuntu preinstalled server (18.04/20.04)
    - [ ] etc (Please specify your OS here)

5. Which OS you installed in Remote PC?

    - [x] Ubuntu 16.04 LTS (Xenial Xerus)
    - [ ] Ubuntu 18.04 LTS (Bionic Beaver)
    - [ ] Ubuntu 20.04 LTS (Focal Fossa)
    - [ ] etc (Please specify your OS here)

6. Specify the software and firmware version(Can be found from Bringup messages)

    - Software version (assuming this is "rosverison"): [1.12.14]
    - Firmware version (assuming this is "core"): [1.2.3]

7. Specify the commands or instructions to replicate the issue.

*ON LAPTOP*
    $ roscore
*ON PI*
    $ roslaunch turtlebot3_bringup turtlebot3_robot.launch multi_robot_name:='turtle1'
*ON LAPTOP*
    $ roslaunch 
    $ roslaunch turtlebot3_bringup turtlebot3_remote.launch multi_robot_name:='turtle1'
    $ roslaunch turtlebot3_slam turtlebot3_slam.launch

8. Copy and Paste the error messages on terminal.

    [ WARN] [1605571674.341106139]: MessageFilter [target=odom ]: Dropped 100.00% of messages so far. Please turn the [ros.gmapping.message_notifier] rosconsole logger to DEBUG for more information.

9. Please describe the issue in detail.

    - We have been attempting to work with two turtlebots. Separating most topics is simple enough. We accomplished this by placing 'export ROS_NAMESPACE=turtle1' in turtle1's .bashrc, and repeating this for turtle2. That way we were able to send messages to, for example, '/turtle1/cmd_vel'.
    - When running SLAM on both of them simultaneously, there is interference in the '/tf' topic, which is used by the SLAM program and both turtles.
    - One walkthrough suggested using the multi_robot_name argument, which differentiates messages sent to '/tf' from turtle1 from those sent from turtle2. However, turtlebot3_slam.launch does not take any multi_robot_name argument. (I could not find any use of '$(arg multi_robot_name)' in turtlebot3_slam.launch, and the behavior is the same whether or not the argument is added.)
    - It would seem that SLAM is still looking for messages in '/tf' marked '/odom', even though they are now marked '/turtle1/odom'.
    - The result is that we are unable to run SLAM for both turtlebots independently, which is necessary for our project.

ashekim, 2020-11-17:
@rpmccarter 

Hi. @rpmccarter . 
Please refer to the link below for information on Multiple TurtleBot3 SLAM.
[11.2.1.6. Virtual SLAM by Multiple TurtleBot3s](https://emanual.robotis.com/docs/en/platform/turtlebot3/simulation/#virtual-slam-by-multiple-turtlebot3s)

Thank you. :)

rpmccarter, 2020-11-17:
@ashekim 

Our project requires that the turtlebots' maps remain separate, which the linked walkthrough does not allow. However, we were able to find a way to rework the example in the walkthrough to work with multiple distinct maps. I'll lay out the process below. Thanks for your help!!

*LAPTOP*
    $ roscore

*PI 1*
    $ ROS_NAMESPACE=turtle1 roslaunch turtlebot3_bringup turtlebot3_robot.launch multi_robot_name:=‘turtle1’ set_lidar_frame_id:=‘turtle1/base_scan’

*PI 2*
    $ ROS_NAMESPACE=turtle2 roslaunch turtlebot3_bringup turtlebot3_robot.launch multi_robot_name:=‘turtle1’ set_lidar_frame_id:=‘turtle1/base_scan’

*LAPTOP*
    $ ROS_NAMESPACE=turtle1 roslaunch turtlebot3_bringup turtlebot3_remote.launch multi_robot_name:=turtle1
    $ ROS_NAMESPACE=turtle1 roslaunch turtlebot3_slam turtlebot3_gmapping.launch set_base_frame:=turtle1/base_footprint set_odom_frame:=turtle1/odom set_map_frame:=turtle1

    $ ROS_NAMESPACE=turtle2 roslaunch turtlebot3_bringup turtlebot3_remote.launch multi_robot_name:=turtle2
    $ ROS_NAMESPACE=turtle2 roslaunch turtlebot3_slam turtlebot3_gmapping.launch set_base_frame:=turtle2/base_footprint set_odom_frame:=turtle2/odom set_map_frame:=turtle2

closed 2020-11-17 by rpmccarter