ROBOTIS-GIT/turtlebot3 #677: ROS2 Foxy turtlebot3 navigation run fail when set initial pose in rviz2

rafaelhuang-intel, 2020-11-18:
ISSUE TEMPLATE ver. 0.4.0

1. Which TurtleBot3 platform do you use?

    - [ ] Burger
    -  Waffle
    - [ ] Waffle Pi

2. Which ROS is working with TurtleBot3?

    - [ ] ROS 1 Kinetic Kame
    - [ ] ROS 1 Melodic Morenia
    - [ ] ROS 1 Noetic Ninjemys
    - [ ] ROS 2 Dashing Diademata
    - [ ] ROS 2 Eloquent Elusor
    - ROS 2 Foxy Fitzroy
    - [ ] etc (Please specify your ROS Version here)

3. Which SBC(Single Board Computer) is working on TurtleBot3?

    - [ ] Intel Joule 570x
    - [ ] Raspberry Pi 3B+
    - [ ] Raspberry Pi 4
    -  etc (  Intel Nuc6i7KYK )

4. Which OS you installed in SBC?

    - [ ] Raspbian distributed by ROBOTIS
    - [ ] Ubuntu MATE (16.04/18.04/20.04)
    -  Ubuntu preinstalled server (18.04/20.04)
    - [ ] etc (Please specify your OS here)

5. Which OS you installed in Remote PC?

    - [ ] Ubuntu 16.04 LTS (Xenial Xerus)
    - [ ] Ubuntu 18.04 LTS (Bionic Beaver)
    -  Ubuntu 20.04 LTS (Focal Fossa)
    - [ ] etc (Please specify your OS here)

6. Specify the software and firmware version(Can be found from Bringup messages)

    - Software version: [x.x.x]
    - Firmware version: [x.x.x]

7. Specify the commands or instructions to replicate the issue.

    terminal 1.  ros2 launch turtlebot3_bring robot.launch.py
    terminal 2.  ros2 launch nav2_bringup bringup_launch.py map:=$HOME/lab_map.yaml 
                          or - ros2 launch turtlebot3_navigation2 navigation2.launch.py map:=$HOME/lab_map.yaml
    terminal 3.  ros2 run rviz2 rviz2 -d $(ros2 pkg prefix nav2_bringup)/share/nav2_bringup/rviz/nav2_default_view.rviz

8. Copy and Paste the error messages on terminal.

    - error log: 
[tb3_navigation2_launch_fail_log_20201116.txt](https://github.com/ROBOTIS-GIT/turtlebot3/files/5558093/tb3_navigation2_launch_fail_log_20201116.txt)
     - used map create by turtlebot3 cartographer brfore run navigation2 test
        [lab_map3.zip](https://github.com/ROBOTIS-GIT/turtlebot3/files/5558136/lab_map3.zip)

9. Please describe the issue in detail.

    - I meet a problem is when I set initial pose in rviz, rviz crash and navigation terminal print more error info. Test map create by turtlebot3 cartographer before run navigation2 test.

ashekim, 2020-11-19:
@rafaelhuang-intel 
Hi. I'm sorry for the late reply.
I will check this issue and get back to you.
Thank you for your patience

rafaelhuang-intel, 2020-11-25:
@ashekim 
I've updated all ros-foxy-turtlebot3 relate package and tried again, this issue disappear, so I think this issue may fixed. I will close this one.

closed 2020-11-25 by rafaelhuang-intel