ROBOTIS-GIT/turtlebot3 #678: Lidar stops working after a while for no reason

royito55, 2020-11-18:
ISSUE TEMPLATE ver. 0.4.0

1. Which TurtleBot3 platform do you use?

    - [ x] Burger
    - [ ] Waffle
    - [ ] Waffle Pi

2. Which ROS is working with TurtleBot3?

    - [ ] ROS 1 Kinetic Kame
    - [x ] ROS 1 Melodic Morenia
    - [ ] ROS 1 Noetic Ninjemys
    - [ ] ROS 2 Dashing Diademata
    - [ ] ROS 2 Eloquent Elusor
    - [ ] ROS 2 Foxy Fitzroy
    - [ ] etc (Please specify your ROS Version here)

3. Which SBC(Single Board Computer) is working on TurtleBot3?

    - [ ] Intel Joule 570x
    - [x ] Raspberry Pi 3B+
    - [ ] Raspberry Pi 4
    - [ ] etc (Please specify your SBC here)

4. Which OS you installed in SBC?

    - [ ] Raspbian distributed by ROBOTIS
    - [x ] Ubuntu MATE (16.04/18.04/20.04)
    - [ ] Ubuntu preinstalled server (18.04/20.04)
    - [ ] etc (Please specify your OS here)

5. Which OS you installed in Remote PC?

    - [ ] Ubuntu 16.04 LTS (Xenial Xerus)
    - [ ] Ubuntu 18.04 LTS (Bionic Beaver)
    - [ ] Ubuntu 20.04 LTS (Focal Fossa)
    - [x ] etc (Please specify your OS here) - Linux Mint

6. Specify the software and firmware version(Can be found from Bringup messages)

    - Software version: [x.x.x]
    - Firmware version: [x.x.x]

7. Specify the commands or instructions to replicate the issue.

    - HERE

8. Copy and Paste the error messages on terminal.

    - HERE

9. Please describe the issue in detail.

Hi, I wanted to ask about an issue I'm having with the lidar of the TurtleBot3 burger. I've noticed that sometimes it doesn't start when I launch the controllers with turtlebot3_robot.launch, and it also turns off randomly after it works for a while. I also saw that I think the green light on the board controlling the lidar (between RPi3 and OpenCR) indicates that the lidar should be turning, but sometimes it's on and the lidar does not turn. When this happens, I can still see the /scan topic being offered but it's empty. I also saw that the issue goes away (although only for some time before it comes back) if I unplug and plug back the lidar from the RPi3's USB port.

Does this have a fix? How do I make sure it's not a hardware issue?

lslabon, 2020-11-18:
I am having the same issue. Let me please know if you fixed it!

ashekim, 2020-11-19:
@royito55  
Hi. Thank you for your questions.

Please test after replacing the USB cable.
Please tell me the result after testing

Thank you.

ROBOTIS-Will, 2020-11-23:
If there are other USB devices connected to Raspberry Pi usb port, lidar may not work properly as it requires more power than other USB devices to run the motor.
Especially, when starting the motor, more power is needed than usual.
If the USB cable is not firmly connected, it could also cause such issues.
Thank you.

royito55, 2020-12-01:
Hi, thank you for the suggestions. Switching the cable did not seem to work. Is there a special rating that this micro USB to USB cable needs to have? It's the cable from the RPi3 to the lidar's PCB board.

ROBOTIS-Will, 2020-12-08:
@royito55 
Please [test the Lidar on a PC](https://emanual.robotis.com/docs/en/platform/turtlebot3/appendix_lds_01/#user-guide-for-driver) and see if it runs well.
If symptom is not reproduced when connected to the PC, Raspberry Pi may not have sufficient power.
The cable included in the package should be able to supply enough power to LDS-01.
This may be caused by connecting other USB devices on Raspberry Pi or OpenCR cannot supply sufficient power to Raspberry Pi.
Below is an excerpt from OpenCR manual.
> 5V power source is supplied from regulated 12V output. Total power consumption on 12V and 5V ports should not exceed 55W.

closed 2021-01-29 by ROBOTIS-Will