ROBOTIS-GIT/turtlebot3 #679: Remote PC does not publish through ssh to turtlebot (Raspberry Pi)

lslabon, 2020-11-18:
ISSUE TEMPLATE ver. 0.4.0

1. Which TurtleBot3 platform do you use?

    - [ ] Burger
    - [ ] Waffle
    - [x] Waffle Pi

2. Which ROS is working with TurtleBot3?

    - [ ] ROS 1 Kinetic Kame
    - [x] ROS 1 Melodic Morenia
    - [ ] ROS 1 Noetic Ninjemys
    - [ ] ROS 2 Dashing Diademata
    - [ ] ROS 2 Eloquent Elusor
    - [ ] ROS 2 Foxy Fitzroy
    - [ ] etc (Please specify your ROS Version here)

3. Which SBC(Single Board Computer) is working on TurtleBot3?

    - [ ] Intel Joule 570x
    - [ ] Raspberry Pi 3B+
    - [x] Raspberry Pi 4
    - [ ] etc (Please specify your SBC here)

4. Which OS you installed in SBC?

    - [ ] Raspbian distributed by ROBOTIS
    - [ ] Ubuntu MATE (16.04/18.04/20.04)
    - [ ] Ubuntu preinstalled server (18.04/20.04)
    - [x] Ubuntu Bionic Beaver (18.04)

5. Which OS you installed in Remote PC?

    - [ ] Ubuntu 16.04 LTS (Xenial Xerus)
    - [x] Ubuntu 18.04 LTS (Bionic Beaver)
    - [ ] Ubuntu 20.04 LTS (Focal Fossa)
    - [ ] etc (Please specify your OS here)

6. Specify the software and firmware version(Can be found from Bringup messages)

    - Software version: [1.14.10]
    - Firmware version: [x.x.x]

7. Specify the commands or instructions to replicate the issue.

    - roscore (Turtlebot)
    - roslaunch turtlebot3_bringup turtlebot3_robot.launch (Turtlebot)
    - roslaunch turtlebot3_teleop turtlebot3_teleop_key.launch (remote PC)

Network settings ( .bashrc):
TurtlebotPC (master):
     - export ROS_MASTER_URI=http://IP_of_TurtlebotPC:11311
     - export ROS_IP=IP_of_TurtlebotPC
RemotePC:
     - export ROS_MASTER_URI=http://IP_of_TurtlebotPC:11311
     - export ROS_IP=IP_of_RemotePC

8. Copy and Paste the error messages on terminal.

    - no error messages

9. Please describe the issue in detail.

I am trying to control my turtlebot with the teleop node, but no messages are published through the ssh connection. 
When launching the teleop node from my turtlebot, everything launches fine. Messages are been published and turtlebot is moving. I can view the messages via rostopic echo /cmd_vel on both devices: Turtlebot (master) and remote PC.

But if I launch the teleop node from my remote PC the turtlebot is not moving. 
In detail: 
rostopic info cmd_vel tells me that remote pc is publisher and turtlebot_core subscriber. 
When I look for rostopic echo cmd_vel on my remote PC the messages are being published. 
But when I look for rostopic echo cmd_vel on the turtlebot it does not receive anything.

So turtlebot->remote works fine but remote->turlebot does not.

Both devices are connected to the same network. Turlebot via wifi and remote PC with LAN-cable (same router).

Thank you for your help!

ashekim, 2020-11-19:
@lslabon 
Hi. Thank you for your questions.
Please refer to the [6.1.4 Network Configuration](https://emanual.robotis.com/docs/en/platform/turtlebot3/pc_setup/#network-configuration) shown in the eManual. 

If you try to run it with remote -> turtlebot, yout Network Configuration change

Network settings ( .bashrc):
TurtlebotPC (master):
- export ROS_MASTER_URI=http://IP_of_RemotePC:11311
- export ROS_IP=IP_of_TurtlebotPC
RemotePC:
- export ROS_MASTER_URI=http://IP_of_RemotePC:11311
- export ROS_IP=IP_of_RemotePC

Thank you.

lslabon, 2020-11-19:
Hey, thanks for your reply!
Sadly this doesn't work for me either... Do you have any other idea?

lslabon, 2020-11-20:
Never mind.. 
I defined the Ip of my remotePC in etc/hosts but did not add it in etc/hosts of turtlebotPC :) 

Now everything is working fine!

closed 2020-11-24 by ashekim