ROBOTIS-GIT/turtlebot3 #682: Can not boot up the Raspberry Pi

ironman6888, 2020-12-05:
ISSUE TEMPLATE ver. 0.4.0

1. Which TurtleBot3 platform do you use?

    - [o ] Burger
    - [ ] Waffle
    - [ ] Waffle Pi

2. Which ROS is working with TurtleBot3?

    - [ ] ROS 1 Kinetic Kame
    - [o ] ROS 1 Melodic Morenia
    - [ ] ROS 1 Noetic Ninjemys
    - [ ] ROS 2 Dashing Diademata
    - [ ] ROS 2 Eloquent Elusor
    - [ ] ROS 2 Foxy Fitzroy
    - [ ] etc (Please specify your ROS Version here)

3. Which SBC(Single Board Computer) is working on TurtleBot3?

    - [ ] Intel Joule 570x
    - [o ] Raspberry Pi 3B+
    - [ o] Raspberry Pi 4
    - [ ] etc (Please specify your SBC here)

4. Which OS you installed in SBC?

    - [o ] Raspbian distributed by ROBOTIS
    - [ ] Ubuntu MATE (16.04/18.04/20.04)
    - [ ] Ubuntu preinstalled server (18.04/20.04)
    - [ ] etc (Please specify your OS here)

5. Which OS you installed in Remote PC?

    - [ ] Ubuntu 16.04 LTS (Xenial Xerus)
    - [o ] Ubuntu 18.04 LTS (Bionic Beaver)
    - [ ] Ubuntu 20.04 LTS (Focal Fossa)
    - [ ] etc (Please specify your OS here)

6. Specify the software and firmware version(Can be found from Bringup messages)

    - Software version: [x.x.x]
    - Firmware version: [x.x.x]

7. Specify the commands or instructions to replicate the issue.

    - 

8. Copy and Paste the error messages on terminal.

    - 

9. Please describe the issue in detail.

    - Hello, I cannot start the Raspberry Pi normally when I try to use Turtlebot3 burger.

I set it up in the ubuntu18 environment based on the following website introduction https://emanual.robotis.com/docs/en/platform/turtlebot3/sbc_setup/

When I downloaded the Raspberry Pi3B according  this picture
[image]
 and installed it with the etcher.io software, when I started the Raspberry Pi, an error as shown like this.
[image]

When I changed to Raspberry Pi4B and downloaded it according this picture
[image]
 and installed it with etcher.io software, when I started the Raspberry Pi, an error as shown like this.
[image]

How can I solve this problem?

ROBOTIS-Will, 2020-12-08:
Hi,
It looks like the restore image has not been burned properly.
Please try with other method.
If you have Ubuntu, try the Restore Disk Image feature of `Disks` utility.

ironman6888, 2020-12-08:
> Hi,
> It looks like the restore image has not been burned properly.
> Please try with other method.
> If you have Ubuntu, try the Restore Disk Image feature of `Disks` utility.

Thank you. It been successed.

goekce, 2020-12-09:
@ironman6888 you can close your issue if your problem is solved.

closed 2020-12-09 by ironman6888