ROBOTIS-GIT/turtlebot3 #683: turtlebot3_core does not bringup correctly

lslabon, 2020-12-09:
ISSUE TEMPLATE ver. 0.4.0

1. Which TurtleBot3 platform do you use?

    - [ ] Burger
    - [ ] Waffle
    - [x] Waffle Pi

2. Which ROS is working with TurtleBot3?

    - [ ] ROS 1 Kinetic Kame
    - [x] ROS 1 Melodic Morenia
    - [ ] ROS 1 Noetic Ninjemys
    - [ ] ROS 2 Dashing Diademata
    - [ ] ROS 2 Eloquent Elusor
    - [ ] ROS 2 Foxy Fitzroy
    - [ ] etc (Please specify your ROS Version here)

3. Which SBC(Single Board Computer) is working on TurtleBot3?

    - [ ] Intel Joule 570x
    - [ ] Raspberry Pi 3B+
    - [x] Raspberry Pi 4
    - [ ] etc (Please specify your SBC here)

4. Which OS you installed in SBC?

    - [ ] Raspbian distributed by ROBOTIS
    - [ ] Ubuntu MATE (16.04/18.04/20.04)
    - [x] Ubuntu preinstalled server (18.04/20.04)
    - [ ] etc (Please specify your OS here)

5. Which OS you installed in Remote PC?

    - [ ] Ubuntu 16.04 LTS (Xenial Xerus)
    - [x] Ubuntu 18.04 LTS (Bionic Beaver)
    - [ ] Ubuntu 20.04 LTS (Focal Fossa)
    - [ ] etc (Please specify your OS here)

6. Specify the software and firmware version(Can be found from Bringup messages)

    - Software version: [1.14.10]
    - Firmware version: [x.x.x]

7. Specify the commands or instructions to replicate the issue.

    - roscore

    - roslaunch turtlebot3_bringup turtlebot3_robot.launch

8. Copy and Paste the error messages on terminal.

    - [INFO] [1607523410.626074, 0.000000]: Connecting to /dev/ttyACM0 at 115200 baud

9. Please describe the issue in detail.

I am facing the same issue as here ( #69 ) and here [https://answers.ros.org/question/287099/turtlebot-bringup-not-connecting-to-port/](url) 

When I try to launch my robot I get stuck at "Connecting to /dev/ttyACM0 at 115200 baud"
I already had this issue last week and was not able to figure out what the problem was. It just worked again after several reboots and unplugging the OpenCR.. Now I am stuck at the same thing again.

I already tried different things that where mentioned in the other threads:
So I am summarizing shortly:

1. Battery is fully charged

2. Port ttyACM0 is available on Raspberry Pi 4 (command: ls /dev/ttyACM*)

3. Unplugging OpenCR and reflashing the sketch (succesfully) shows no difference (-> USB cable is working)

4. Rebooting Raspberry Pi shows no difference

5. I have other devices e.g. lidar sensor attached. Therefore I already tried connecting to OpenCR without multiple devices ( Unplugged everything except OpenCr and launched turtlebot3_core seperatly)

Thank you for your help!
Laura

ROBOTIS-Will, 2020-12-10:
Hi @lslabon 
Thank you for the detailed description. 
I don't see any particular issues on your settings.
I'll check with my RPi4 and see if I can replicate the issue.
When I test the Melodic, I used the same Ubuntu image(18.04.3) for Dashing linked in the eManual. Please let me know if you have tried with other version.
Thank you.

lslabon, 2020-12-10:
I installed an Ubuntu 18.04.5 image (GNU/Linux 5.4.0-1023-raspi aarch64)

lslabon, 2020-12-12:
Hi @ROBOTIS-Will 

I think I figured out what the problem is here:

When I started a simulation with gazebo and closed it, some topics remained being published (even with everything closed!!!). They were only killed when killing roscore either! 
When launching the turtlebot3_core without killing and restarting roscore the port ttyACM0 I got stuck at Connecting to OpenCR. 
This was kinda strange because I could do cat dev/ttyACM0 succesfully until launching turtlebot3_core. After that it was found (ls dev/ttyACM*) but the port was closed (cat dev/ttyACM0 not possible anymore)

I don't know whats the bug is here but killing every node/topic before launching turtlebot3_core worked for me with 100% success each time I got stuck at connecting to OpenCR.

Maybe you can reproduce it that way

Best regards
Laura

ROBOTIS-Will, 2020-12-14:
Hi @lslabon,

Aha! I only launch the gazebo with `roslaunch` command which automatically creates the roscore(if it is not created before) and terminates it when I close the Gazebo, and that's why I couldn't reproduce the issue.
Thank you for sharing the solution.
This information will be very useful for others who stuck at connecting OpenCR.
Happy holiday and stay safe!

Best Regards,
Will

yuanzhedong, 2020-12-20:
> Hi @ROBOTIS-Will
> 
> I think I figured out what the problem is here:
> 
> When I started a simulation with **gazebo** and **closed** it, **some topics remained being published** (even with everything closed!!!). They were only killed when killing roscore either!
> When launching the turtlebot3_core without killing and restarting roscore the port ttyACM0 I got stuck at Connecting to OpenCR.
> This was kinda strange because I could do cat dev/ttyACM0 succesfully until launching turtlebot3_core. After that it was found (ls dev/ttyACM*) but the port was closed (cat dev/ttyACM0 not possible anymore)
> 
> I don't know whats the bug is here but killing every node/topic **(killing roscore and restarting) before launching turtlebot3_core** worked for me with 100% success each time I got stuck at connecting to OpenCR.
> 
> Maybe you can reproduce it that way
> 
> Best regards
> Laura

I got the exact same issue and you save my life. I also used gazebo before booting turtlebot. I was thinking it was some hardware issue, turns out restarting the roscore fixed it.

closed 2021-03-10 by ROBOTIS-Will