ROBOTIS-GIT/turtlebot3 #684: `ros2-devel` repos point to `ros2` branches

goekce, 2020-12-11:
I want to use `colcon` to build all `ros2-devel` version of packages. I saw that packages defined in `ros2-devel` branch of `turtlebot3.repos (https://github.com/ROBOTIS-GIT/turtlebot3/blob/ros2-devel/turtlebot3.repos) point to `ros2` branches.

Is this a bug or feature?

ROBOTIS-Will, 2020-12-14:
Hi @goekce ,
The `ros2-devel` is just a preliminary branch before merging and releasing on `ros2`.
`ros2-devel` and `ros2` is used to test and update the latest ros2 distro(at this point, Foxy).
Since the latest release has identical code between `ros2` and `ros2-devel`, you may use any of these branch, but I'd recommend using ros2.
Thank you.

goekce, 2020-12-14:
Dear Will,  I want to test the `ros2-devel` code.

> Since the latest release has identical code between ros2 and ros2-devel

As far as I understand, `ros2` and `ros2-devel` branches cannot be identical, because you use `ros2-devel` for preliminary tests. Do I understand correctly?

Here is an example that they are not identical:

https://github.com/ROBOTIS-GIT/turtlebot3_simulations/compare/ros2...ros2-devel

I want to able to fetch all `ros2-devel` branches of `turtlebot3` repositories. Currently I have to fetch the `ros2-devel` branches one by one.  If you would change `turtlebot3.repos` in `ros2-devel` branch accordingly, then it would be possible to fetch the `ros2-devel` branch by running:

```
mkdir -p tbws/src && cd tbws
git clone https://github.com/ROBOTIS-GIT/turtlebot3 -b ros2-devel
vcs import src < turtlebot3/turtlebot3.repos
```

ROBOTIS-Will, 2020-12-15:
Hi @goekce 
Oh, sorry. Now I see your point.
Since other ROS2 distro branches are also simply using ros2 in its repos file, I'll modify this.
It may take a few weeks before releasing binary package since we might take end of year holiday break.
Thank you.

goekce, 2020-12-15:
Oh I am glad I could clarify my point!

I do not use the binary packages, it is not a problem for me at all. I also did not know that there are binary packages for `ros-devel` branches!

Enjoy your holidays, thank you Will.

ROBOTIS-Will, 2020-12-17:
Hi @goekce 
I'm sorry about the confusion. We do not release the binary package for ros2-devel branch. The binary is only released for `${ROS_DISTRO}-devel` branches.
The change will be applied within a couple week as we are getting closer to the end of year.
Happy holiday and stay safe!

goekce, 2020-12-17:
We'll leave the issue open until the fix then. gamsahabnida for the clarification Will!

ROBOTIS-Will, 2021-01-29:
Closing this as this has been applied to the latest release.
Thank you for your contribution :)

closed 2021-01-29 by ROBOTIS-Will