ROBOTIS-GIT/turtlebot3 #685: OpenCR firmware update fails - even in recovery mode

shspears, 2020-12-16:
ISSUE TEMPLATE ver. 0.4.0

1. Which TurtleBot3 platform do you use?

    - [ ] Burger
    - [X ] Waffle
    - [ ] Waffle Pi

2. Which ROS is working with TurtleBot3?

    - [ ] ROS 1 Kinetic Kame
    - [ ] ROS 1 Melodic Morenia
    - [ ] ROS 1 Noetic Ninjemys
    - [ ] ROS 2 Dashing Diademata
    - [ ] ROS 2 Eloquent Elusor
    - [ X] ROS 2 Foxy Fitzroy
    - [ ] etc (Please specify your ROS Version here)

3. Which SBC(Single Board Computer) is working on TurtleBot3?

    - [ ] Intel Joule 570x
    - [ X] Raspberry Pi 3B+
    - [ ] Raspberry Pi 4
    - [ ] etc (Please specify your SBC here)

4. Which OS you installed in SBC?

    - [ ] Raspbian distributed by ROBOTIS
    - [ ] Ubuntu MATE (16.04/18.04/20.04)
    - [X ] Ubuntu preinstalled server (18.04/**20.04**)
    - [ ] etc (Please specify your OS here)

5. Which OS you installed in Remote PC?

    - [ ] Ubuntu 16.04 LTS (Xenial Xerus)
    - [ ] Ubuntu 18.04 LTS (Bionic Beaver)
    - [ X] Ubuntu 20.04 LTS (Focal Fossa)
    - [ X ] etc (Also tried using Windows 10)

6. Specify the software and firmware version(Can be found from Bringup messages)

    - Software version: [x.x.x]
    - Firmware version: [x.x.x]

7. Specify the commands or instructions to replicate the issue.

    - Followed the instructions here: [OpenCR Setup](https://emanual.robotis.com/docs/en/platform/turtlebot3/opencr_setup/#opencr-setup)
    - 
8. Copy and Paste the error messages on terminal.

    - OpenCR Update Start..
opencr_ld_shell ver 1.0.0
opencr_ld_main
[  ] file name          : waffle.opencr
[  ] file size          : 127 KB
[  ] fw_name            : waffle
[  ] fw_ver             : V190829R3
[OK] Open port          : /dev/ttyACM0
[  ]
[NG] cmd_read_board_name fail : 0xF020

9. Please describe the issue in detail.

    - Cannot update firmware on OpenCR - even in firmware recovery mode
    - Followed the instructions above to no avail. Kept getting the 0xF020, even when in recovery mode.
    - Tried to upload using a windows 10 machine 
    - Followed the instructions to do it using the arduino IDE [here](https://emanual.robotis.com/docs/en/parts/controller/opencr10/#install-on-windows)
    - I tried to load the turtlebot_ros2 waffle example ino, but it doesn't compile
    - ERROR is : 
    - ------------------------------------------------------------------------------------------
    - WARNING: library turtlebot3_ros2 claims to run on opencr architecture(s) and may be incompatible with your current board which runs on OpenCR architecture(s).
C:\Users\spears\Documents\ArduinoData\packages\OpenCR\hardware\OpenCR\1.4.16\libraries\turtlebot3_ros2\src\turtlebot3\turtlebot3_motor_driver.cpp: In member function 'bool Turtlebot3MotorDriver::get_torque()':
C:\Users\spears\Documents\ArduinoData\packages\OpenCR\hardware\OpenCR\1.4.16\libraries\turtlebot3_ros2\src\turtlebot3\turtlebot3_motor_driver.cpp:104:31: error: 'TORQUE_ENABLE' was not declared in this scope
   if(dxl.readControlTableItem(TORQUE_ENABLE, left_wheel_id_) == true
                               ^
C:\Users\spears\Documents\ArduinoData\packages\OpenCR\hardware\OpenCR\1.4.16\libraries\turtlebot3_ros2\src\turtlebot3\turtlebot3_motor_driver.cpp:104:31: note: suggested alternative:
In file included from C:\Users\spears\Documents\Arduino\libraries\Dynamixel2Arduino\src/Dynamixel2Arduino.h:23:0,
                 from c:\users\spears\documents\arduinodata\packages\opencr\hardware\opencr\1.4.16\libraries\turtlebot3_ros2\include\turtlebot3\turtlebot3_motor_driver.h:20,
                 from C:\Users\spears\Documents\ArduinoData\packages\OpenCR\hardware\OpenCR\1.4.16\libraries\turtlebot3_ros2\src\turtlebot3\turtlebot3_motor_driver.cpp:17:
C:\Users\spears\Documents\Arduino\libraries\Dynamixel2Arduino\src/actuator.h:251:5: note:   'TORQUE_ENABLE'
     TORQUE_ENABLE,
     ^
exit status 1
Error compiling for board OpenCR Board.
-------------------------------------------------------------------------------------------------
- Dynamixel2Arduino version is 0.4.5
- OpenCR board manager is version 1.4.15

- Tried blinking the LED's using the arduino IDE (OpenCR basics blink LED example). 
- I get the following error - both normally, as well as in recovery mode on the OpenCR board
------------------------------------------------------------------------------------------
opencr_ld ver 1.0.2
opencr_ld_main 

>>

file name : C:\Users\spears\AppData\Local\Temp\arduino_build_52212/b_Blink_LED.ino.bin 

file size : 55 KB

An error occurred while uploading the sketch
Open port OK
Clear Buffer Start
Clear Buffer End
>>

cmd_read_board_name fail : 0xF020

----------------------------------------
- I tried new USB cables as well.

Any help would be greatly appreciated!!

ROBOTIS-Will, 2020-12-17:
Hi @shspears 
Could you try the below port settings on your Ubuntu? The error 0xF020 is a timeout error that occurs when the board does not respond on time.
```bash
$ wget https://raw.githubusercontent.com/ROBOTIS-GIT/OpenCR/master/99-opencr-cdc.rules
$ sudo cp ./99-opencr-cdc.rules /etc/udev/rules.d/
$ sudo udevadm control --reload-rules
$ sudo udevadm trigger
```
You can ignore the architecture incompatible warning which will be fixed in the next release(in a couple weeks).
Most of your issues should be resolved when installing DYNAMIXEL2Arduino, but if it doesn't, please add below temporary measurement in your sketch

```bash
#define TORQUE_ENABLE 1
```
Thank you.

ROBOTIS-Will, 2020-12-17:
The changes are being made at the OpenCR devel branch.
https://github.com/ROBOTIS-GIT/OpenCR/commit/7dcece77659965d01611dc94a976b755fe7e7ce9

shspears, 2020-12-17:
Hi @ROBOTIS-Will, 
thank you for getting back to me. Much appreciated.

I've made the fix on the RPi, which is connected to the opencr. Unfortunately I'm getting the same error.

--------
ubuntu@ubuntu:~/opencr_update$ ./update.sh $OPENCR_PORT $OPENCR_MODEL.opencr
aarch64
arm
OpenCR Update Start..
opencr_ld_shell ver 1.0.0
opencr_ld_main
[  ] file name          : waffle.opencr
[  ] file size          : 127 KB
[  ] fw_name            : waffle
[  ] fw_ver             : V190829R3
[OK] Open port          : /dev/ttyACM0
[  ]
[NG] cmd_read_board_name fail : 0xF020
[  ] file name          : waffle.opencr
[  ] file size          : 127 KB
[  ] fw_name            : waffle
[  ] fw_ver             : V190829R3
[OK] Open port          : /dev/ttyACM0
[  ]
[NG] cmd_read_board_name fail : 0xF020
[  ] file name          : waffle.opencr
[  ] file size          : 127 KB
[  ] fw_name            : waffle
[  ] fw_ver             : V190829R3
[OK] Open port          : /dev/ttyACM0
[  ]
[NG] cmd_read_board_name fail : 0xF020
------------

ROBOTIS-Will, 2020-12-18:
@shspears 
You might have a damaged board. Please try uploading the Bootloader and see if this fixes your issue.
Thanks.

closed 2021-03-10 by ROBOTIS-Will