ROBOTIS-GIT/turtlebot3 #686: Raspberry Pi 3B+ only provides /odom @ 23 hz and crashes (Ubuntu Noetic)

MikeHallettUK, 2020-12-17:
ISSUE TEMPLATE ver. 0.4.0

1. Which TurtleBot3 platform do you use?

    - [x ] Burger
    - [ ] Waffle
    - [ ] Waffle Pi

2. Which ROS is working with TurtleBot3?

    - [ ] ROS 1 Kinetic Kame
    - [ ] ROS 1 Melodic Morenia
    - [x ] ROS 1 Noetic Ninjemys
    - [ ] ROS 2 Dashing Diademata
    - [ ] ROS 2 Eloquent Elusor
    - [ ] ROS 2 Foxy Fitzroy
    - [ ] etc (Please specify your ROS Version here)

3. Which SBC(Single Board Computer) is working on TurtleBot3?

    - [ ] Intel Joule 570x
    - [x ] Raspberry Pi 3B+
    - [ ] Raspberry Pi 4
    - [ ] etc (Please specify your SBC here)

4. Which OS you installed in SBC?

    - [ ] Raspbian distributed by ROBOTIS
    - [ ] Ubuntu MATE (16.04/18.04/20.04)
    - [x ] Ubuntu preinstalled server (18.04/20.04)
    - [ ] etc (Please specify your OS here)

5. Which OS you installed in Remote PC?

    - [ ] Ubuntu 16.04 LTS (Xenial Xerus)
    - [ ] Ubuntu 18.04 LTS (Bionic Beaver)
    - [ ] Ubuntu 20.04 LTS (Focal Fossa)
    - [x ] WIndows10

6. Specify the software and firmware version(Can be found from Bringup messages)

    - Software version: [rosversion: 1.15.9]
    - Firmware version: [1.4.16]

7. Specify the commands or instructions to replicate the issue.

    - rostopic hz /odom

8. Copy and Paste the error messages on terminal.

subscribed to [/odom]
average rate: 23.679
        min: 0.000s max: 0.066s std dev: 0.02253s window: 70

9. Please describe the issue in detail.

- Raspberry Pi 3B+ with newly installed Ubuntu v20 + Noetic, only delivers /odom @ ~ 23hz.  This drops further if more topics run on PC, such as SLAM Cartographer.  SLAM Cartographer does not work well with many timeout errors.
- Raspberry Pi 3B+ then also crashes after ~ 5 to 10 minutes.
- in "top" output "python3" program shows as consuming ~ 66% to 70% of cpu

_Question:_ is Pi 3 not powerful enough (lacks performance) to do the job ?

This all works fine set up exactly the same on Pi 4: with odom hz average rate: 29.152 and cpu "python3" usage @ ~ 48% to 60%.

ROBOTIS-Will, 2020-12-18:
Hi @MikeHallettUK 
My RPi3B+ also consumes more than 70% of the CPU resource after the bringup and degrades the frequency of /odom pushlisher, but didn't experience the crash nor further degradation yet.
I'll try the Cartographer and see if it makes any differences.
Raspberry Pi 3B+ has limited resources, therefore, we're trying to run as little processes from it as possible.
Thank you.

MikeHallettUK, 2020-12-18:
OK thanks for reply, and confirmation that this is to be expected.  I have switched to a PI4 anyway.

closed 2020-12-18 by MikeHallettUK