ROBOTIS-GIT/turtlebot3 #687: /odom returns zero velocity

ozgudum, 2020-12-17:
ISSUE TEMPLATE ver. 0.4.0

1. Which TurtleBot3 platform do you use?

    Waffle

2. Which ROS is working with TurtleBot3?
    - [ ] ROS 2 Foxy Fitzroy

3. Which SBC(Single Board Computer) is working on TurtleBot3?
    - [ ] Raspberry Pi 3B+

When I'm running TB3, I am always getting near zero values for /odom/twist. Is this expected? I would expect to get robot velocity in m/s. No issue with /odom/pose

header:
  stamp:
    sec: 1608235551
    nanosec: 879066078
  frame_id: odom
child_frame_id: base_footprint
pose:
  pose:
    position:
      x: 0.14691394855059656
      y: 0.002690837393091869
      z: 0.0
    orientation:
      x: 0.0
      y: 0.0
      z: 0.00243727714663111
      w: 0.9999970298356443
  covariance:
...
twist:
  twist:
    linear:
      x: -2.1876057909291433e-12
      y: 0.0
      z: 0.0
    angular:
      x: 0.0
      y: 0.0
      z: 7.178701217325196e-13
  covariance:
...

ROBOTIS-Will, 2020-12-18:
Hi @ozgudum 
Thank you for reporting this issue.
I'll look into the code and see what causes this problem.
Thanks!

shanpenghui, 2021-06-09:
Hi @ozgudum 
Have you resolve the problem?  I have the same issue.

GyuH13, 2024-10-31:
This issue was raised so long ago that it is now challenging to address. 
Please refer to the updated manual, and if you have any further questions, feel free to open a new issue.
Thanks.

closed 2024-11-08 by sunghowoo