ROBOTIS-GIT/turtlebot3 #688: Creating SLAM in Rviz2

sfx-takei, 2020-12-22:
I'm sorry if it's a poor sentence.
We would appreciate it if you could read the question and answer it.

### Your enviroment
* Ubuntu MATE 18.04 
* ROS2 Dashing
* Gazebo9
* Rviz2

### Subject of the issue
I'm trying to validate SLAM with Rviz2.
However, the map cannot be displayed at the stage of forming the map with Cartographer.

### Steps to reproduce
You need to have the official ROS Turtlebot3 package installed / set up in advance.
First, start Gazebo of Tutrlebot3.
```
$ ros2 launch turtlebot3_gazebo turtlebot3_world.launch.py
```
Then start Cartographer.
```
$ ros2 launch turtlebot3_cartographer cartographer.launch.py use_sim_time:=True
```

### Expected behaviour
Within Rviz2, form the map acquired by Tutrlebot3.
[image]

### Actual behaviour
No map is formed and Map always returns "No map received".
[image]

### Log
```
[occupancy_grid_node-2] [WARN] [occupancy_grid_node]:  submap_slices and last_frame_id id empty
```

ROBOTIS-Will, 2020-12-24:
Hi @sfx-takei 
Thank you for reporting the issue.
I'll take a look at it and get back to you soon.

kantengri, 2021-01-10:
added these remappings to cartographer.launch.py from turtlebot3_cartographer:
    remappings=[
            ('/scan', '/turtlebot3_laserscan/out'),
            ('/imu', '/turtlebot3_imu/out'),
    ]
for each node: rviz, cartographer_node,
now it works

ROBOTIS-Will, 2021-01-12:
@sfx-takei 
Could you test this with the latest code in `dashing-devel` of turtlebot3 and turtlebot3_simulations package?
Updated pacakage will be released within a couple week.

sfx-takei, 2021-01-12:
@kantengri 
Thank you for your reply.
When I tried remapping as you said,
The cartographer has started successfully!
Much obliged. Thank you!

sfx-takei, 2021-01-12:
@ROBOTIS-Will 
Thank you for your reply.
I also tested in the same environment as you.
As a result it worked fine!

ROBOTIS-Will, 2021-01-13:
@sfx-takei 
Thank you for the update. Please don't forget to close the issue if it is resolved.
Happy New Year!

closed 2021-01-13 by sfx-takei