ROBOTIS-GIT/turtlebot3 #690: LDS Laser scan frequency skips to ~0.398 seconds every ~60 scans

MikeHallettUK, 2020-12-22:
ISSUE TEMPLATE ver. 0.4.0

1. Which TurtleBot3 platform do you use?

    - [ ] Burger
    - [x ] Waffle
    - [ ] Waffle Pi

2. Which ROS is working with TurtleBot3?

    - [ ] ROS 1 Kinetic Kame
    - [ ] ROS 1 Melodic Morenia
    - [x ] ROS 1 Noetic Ninjemys
    - [ ] ROS 2 Dashing Diademata
    - [ ] ROS 2 Eloquent Elusor
    - [ ] ROS 2 Foxy Fitzroy
    - [ ] etc (Please specify your ROS Version here)

3. Which SBC(Single Board Computer) is working on TurtleBot3?

    - [ ] Intel Joule 570x
    - [ ] Raspberry Pi 3B+
    - [x ] Raspberry Pi 4
    - [ ] etc (Please specify your SBC here)

4. Which OS you installed in SBC?

    - [ ] Raspbian distributed by ROBOTIS
    - [ ] Ubuntu MATE (16.04/18.04/20.04)
    - [x ] Ubuntu preinstalled server (18.04/20.04)
    - [ ] etc (Please specify your OS here)

5. Which OS you installed in Remote PC?

    - [ ] Ubuntu 16.04 LTS (Xenial Xerus)
    - [ ] Ubuntu 18.04 LTS (Bionic Beaver)
    - [ ] Ubuntu 20.04 LTS (Focal Fossa)
    - [x ] etc (Windows10)

6. Specify the software and firmware version(Can be found from Bringup messages)

    Software version: [rosversion: 1.15.9]
    Firmware version: [1.4.16]

7. Specify the commands or instructions to replicate the issue. = e.g. PYTHON code:

```
`#!/usr/bin/env python3
# laser scan frequency test

import rospy
from sensor_msgs.msg import PointCloud2, LaserScan
from laser_geometry.laser_geometry import LaserProjection
import numpy as np
import ros_numpy
import matplotlib.pyplot as plt

rospy.init_node("lasertest")
count = 0
stop = 600
lastscan = rospy.get_rostime()
lastseq = 0
lastskip = 0
lowrange = 0
numpoints = np.copy([])
skips = np.copy([])
freq = np.copy([])

def scan_cb(msg):
    global count, lastscan, lastseq, numpoints, skips, lastskip, lowrange, freq
    if count == 0:
        lastseq = msg.header.seq
        lastskip = msg.header.seq
    scantime = msg.header.stamp - lastscan
    scantime = scantime.to_sec()
    lastscan = msg.header.stamp
    gap = msg.header.seq - lastseq
    lastseq = msg.header.seq
    count = count + 1
    if scantime > 0.25 or gap > 1:              # => scan error ...
        interval = msg.header.seq - lastskip
        lastskip = msg.header.seq
        print("%d) Seq: %d , gap: %d, %d, Scantime %.4f " % (count, msg.header.seq, gap, interval, scantime))
        skips = np.append(skips, interval)
    else:
        if count != 1:
            freq = np.append(freq, scantime)
    points = np.count_nonzero(msg.ranges)                     # number of good range readings
    numpoints = np.append(numpoints, points)
    lowcount = np.average(numpoints) - 4.0 * np.std(numpoints)
    if points < lowcount:
        lowrange = lowrange + 1
        print("Seq: %d, low range count= %d, Average points %d, cutoff= %.1f, Scantime %.4f " % (
                msg.header.seq, points, np.average(numpoints), lowcount, scantime))

print("Starting to scan ~ %d times: should take ~ %.1f seconds" % (stop, stop/5.0))
rospy.Subscriber("/scan", LaserScan, scan_cb, queue_size=2)

while count < 600 and not rospy.is_shutdown():
    rospy.sleep(1.0)

rospy.signal_shutdown("Finished")
print("Finished testing after %d scans" % (count))
print("%d skips, Average interval %d, SDev= %.1f " % (len(skips), np.average(skips), np.std(skips)))
print("%d low range counted, Average points %d, SDev= %.1f " % (lowrange, np.average(numpoints), np.std(numpoints)))
av = np.average(freq)
sd = np.std(freq)
print("Average frequency %.3f, Average good interval %.4f, SDev= %.6f " % (1.0/av, av, sd))
print("Minimum good interval %.4f, Maximum good interval %.4f " % (np.min(freq), np.max(freq)))
top = av + 2.5 * sd
bot = av - 2.5 * sd
plt.hist(freq, bins=20, range=(bot,top))
plt.title("Good Scantimes Interval Distribution")
plt.show()`
```

8. Copy and Paste the error messages on terminal.

Starting to scan ~ 600 times: should take ~ 120.0 seconds
151) Seq: 26276 , gap: 1, 150, Scantime 0.3984
166) Seq: 26291 , gap: 1, 15, Scantime 0.3985
174) Seq: 26299 , gap: 1, 8, Scantime 0.3986
202) Seq: 26327 , gap: 1, 28, Scantime 0.3985
241) Seq: 26366 , gap: 1, 39, Scantime 0.3985
277) Seq: 26402 , gap: 1, 36, Scantime 0.3987
321) Seq: 26446 , gap: 1, 44, Scantime 0.3987
Seq: 26677, low range count= 258, Average points 265, cutoff= 259.1
553) Seq: 26678 , gap: 1, 232, Scantime 0.3985
Finished testing after 601 scans
8 skips, Average interval 69, SDev= 74.2
1 low range counted, Average points 265, SDev= 1.6
Average frequency 4.982, Average good interval 0.2007, SDev= 0.000403
Minimum interval 0.1974, Maximum interval 0.2044

9. Please describe the issue in detail.

As can be seen in the output above, the LDS Laser scan frequency skips to ~0.398 seconds every ~60 scans; very randomly.

Most of the time it is [solidly 0.2007 seconds](https://drive.google.com/file/d/1yM_OkTmekoVuZ4xmZDrJA3NhDcocx5k1/view?usp=sharing), with a low SDev= 0.000403.  Maybe of note, 0.398 seconds is just less than 2x 0.2007 so maybe something does not have enough time to get the next scan ?

I have tested this many times, with Melodic and Noetic, 2 different LDS scanners, Rasbian and Ubuntu, over wifi and directly on the SBC, under heavy load, and not: all show similar, very random, results.

Q1: Can you reproduce this on your systems ?

Q2: is there a possible code fix ? eg in hls_lfcd_lds_driver ?

Thanks
Michael Hallett

ROBOTIS-Will, 2020-12-29:
Hi @MikeHallettUK ,
Sorry about the delayed reply. 
I'll test this in our system and see if we can also reproduce this issue.
Thank you.

MikeHallettUK, 2021-01-10:
Also following this up ... I now ocassionally get "range_data_collator.cc Dropped x earlier points" error in Cartographer; indicate a problem with timestamps and scan->time_increment.

I have your HOTFIX for hlds_laser_publisher.cpp, so we now get a more correct scan->time_increment, (see issue @ [TurtleBot3 LDS Lidar laser scan msg.time_increment is 20 x too small ?](https://github.com/ROBOTIS-GIT/turtlebot3/issues/634)): but still some problems.

So I adjusted the above python code to add in this extra test, which shows the msg.scan_time is usually greater than the gap between timestamps:

```
lastseq = msg.header.seq
scantimeInc = 360.0 * msg.time_increment
if count == 1:
    print("%d) msg.scan_time= %.5f , scantimeInc= %.5f, StampdifScantime= %.5f " % (count, msg.scan_time, scantimeInc, scantime))
if scantime < msg.scan_time or scantime < scantimeInc:
    print("%d) msg.scan_time= %.5f , scantimeInc= %.5f, StampdifScantime= %.5f " % (count, msg.scan_time, scantimeInc, scantime))
count = count + 1
```

So I adjusted your HOTFIX code to include this line after you calculate the scan->time_increment:

`scan->time_increment = scan->time_increment * 358.6 / 360.0;    // adjust for cartographer drop points error`

This generally fixes it, with this sort of output:

```
Starting to scan ~ 600 times: should take ~ 120.0 seconds
0) msg.scan_time= 0.20056 , scantimeInc= 0.20056, StampdifScantime= 0.18412
1) msg.scan_time= 0.20056 , scantimeInc= 0.20056, StampdifScantime= 0.20070
Seq: 1002, low range count= 252, Average points 258, cutoff= 252.8, Scantime 0.20335
62) Seq: 1003 , gap: 1, 61, Scantime 0.39847
103) Seq: 1044 , gap: 1, 41, Scantime 0.39860
119) msg.scan_time= 0.20056 , scantimeInc= 0.20056, StampdifScantime= 0.20055
131) msg.scan_time= 0.20056 , scantimeInc= 0.20056, StampdifScantime= 0.20055
157) msg.scan_time= 0.20056 , scantimeInc= 0.20056, StampdifScantime= 0.20055
161) Seq: 1102 , gap: 1, 58, Scantime 0.39852
225) Seq: 1166 , gap: 1, 64, Scantime 0.39850
Finished testing after 600 scans
4 skips, Average interval 56, SDev= 8.9
1 low range counted, Average points 259, SDev= 1.7
Average frequency 4.983, Average good interval 0.20069, SDev= 0.000229
Minimum good interval 0.20047, Maximum good interval 0.20337
```

And … e.g. 

C:\ws>rostopic echo -n1 /scan
header:
  seq: 2466
  stamp:
    secs: 1610302235
    nsecs: 528392880
  frame_id: "base_scan"
angle_min: 0.0
angle_max: 6.26573181152
angle_increment: 0.0174532923847
time_increment: 0.000557109131478
scan_time: 0.200559288263
range_min: 0.119999997318
range_max: 3.5

The skip problem (occassional large delay in scan time) is also clear in a simple rostopic hz /scan output, looking at the max times:

C:\ws>rostopic hz /scan
subscribed to [/scan]
average rate: 4.992
        min: 0.200s max: 0.201s std dev: 0.00047s window: 4
average rate: 4.991
        min: 0.184s max: 0.301s std dev: 0.01903s window: 29
average rate: 4.990
        min: 0.116s max: 0.301s std dev: 0.02922s window: 34
average rate: 4.935
        min: 0.116s max: 0.401s std dev: 0.02678s window: 98

closed 2022-07-20 by MikeHallettUK