ROBOTIS-GIT/turtlebot3 #692: Basic Operation - Publisher Failure

LW-G38, 2021-01-04:
ISSUE TEMPLATE ver. 0.4.0

1. Which TurtleBot3 platform do you use?

    - [ ] Burger
    - [ ] Waffle
    - [*] Waffle Pi

2. Which ROS is working with TurtleBot3?

    - [ ] ROS 1 Kinetic Kame
    - [ ] ROS 1 Melodic Morenia
    - [*] ROS 1 Noetic Ninjemys
    - [ ] ROS 2 Dashing Diademata
    - [ ] ROS 2 Eloquent Elusor
    - [ ] ROS 2 Foxy Fitzroy
    - [ ] etc (Please specify your ROS Version here)

3. Which SBC(Single Board Computer) is working on TurtleBot3?

    - [ ] Intel Joule 570x
    - [*] Raspberry Pi 3B+
    - [ ] Raspberry Pi 4
    - [ ] etc (Please specify your SBC here)

4. Which OS you installed on SBC?

    - [ ] Raspbian distributed by ROBOTIS
    - [ ] Ubuntu MATE (16.04/18.04/20.04)
    - [*] Ubuntu preinstalled server (18.04/20.04)
    - [ ] etc (Please specify your OS here)

5. Which OS you installed on Remote PC?

    - [ ] Ubuntu 16.04 LTS (Xenial Xerus)
    - [ ] Ubuntu 18.04 LTS (Bionic Beaver)
    - [ ] Ubuntu 20.04 LTS (Focal Fossa)
    - [ ] Windows 10
    - [ ] MAC OS X (Specify version)
    - [*] Ubuntu preinstalled server 20.04 with desktop on a Raspeberry Pi 4B

6. Specify the software and firmware version(Can be found from Bringup messages)

    - Software version: [1.15.9]
    - Firmware version: [1.2.3]

7. Specify the commands or instructions to reproduce the issue.

    - roslaunch turtlebot3_teleop turtlebot3_teleop_key.launch

8. Copy and Paste the error messages on terminal.

    - [ERROR] [1609661002.887650]: Creation of publisher failed: Checksum does not match: 4ddae7f048e32fda22cac764685e3974,476f837fa6771f6e16e3bf4ef96f8770
[INFO] [1609661002.906556]: Setup publisher on magnetic_field [sensor_msgs/MagneticField]
[INFO] [1609661002.923992]: Setup publisher on /tf [tf/tfMessage]
[ERROR] [1609661002.957902]: Tried to publish before configured, topic id 131

9. Please describe the issue in detail.

    - When I'm trying out the basic operation (teleoperation), the error above occurs. I use macOS to run Arduino to upload certain programs to openCR but I also use a Raspberry PI 4 with Ubuntu 20.4 as the remote PC (communicating with Ubuntu 20.4 on the PI 3 B+ on the robot). 
    - I used macOS to run ssh to the remote PC (Pi 4) to run the terminal command above.
    - Both the remote PC and the SBC on the robot are installed Ubuntu 20.04 server and ROS 1 Melodic.
    - I understand that using a PI as the remote PC may not be recommended and wonder if the issue is from there.

ashekim, 2021-01-05:
@LW-G38 
Hi. Thank you for your questions.
Unfortunately, we haven't tested the RPi4 yet. 

- Did you run Bring up before performing Basic Operation?
- We will attach a link that you can refer to among the [turtlebot3 issue.](https://github.com/ROBOTIS-GIT/turtlebot3/issues/108#issue-290247929)

Thank you.
And Happy New Year! :)

LW-G38, 2021-01-05:
@ashekim 
Hi, I did see that post and have updated my software and firmware to the latest.
Now I used another laptop to do the bringup but it still gives me the same error in loops::

[ERROR] [1609833139.200197]: Creation of publisher failed: Checksum does not match: 4ddae7f048e32fda22cac764685e3974,476f837fa6771f6e16e3bf4ef96f8770
[INFO] [1609833139.235663]: Setup publisher on magnetic_field [sensor_msgs/MagneticField]
[INFO] [1609833141.597972]: Setup publisher on /tf [tf/tfMessage]
[INFO] [1609833141.651072]: Note: subscribe buffer size is 1024 bytes
[INFO] [1609833141.670310]: Setup subscriber on cmd_vel [geometry_msgs/Twist]
[INFO] [1609833141.721549]: Setup subscriber on sound [turtlebot3_msgs/Sound]
[INFO] [1609833141.769031]: Setup subscriber on motor_power [std_msgs/Bool]
[INFO] [1609833141.825065]: Setup subscriber on reset [std_msgs/Empty]
[INFO] [1609833141.866540]: Setup TF on Odometry [odom]
[INFO] [1609833141.885441]: Setup TF on IMU [imu_link]
[INFO] [1609833141.904680]: Setup TF on MagneticField [mag_link]
[INFO] [1609833141.926242]: Setup TF on JointState [base_link]
[ERROR] [1609833141.948017]: Tried to publish before configured, topic id 131
[INFO] [1609833141.965922]: Requesting topics...
[INFO] [1609833144.425350]: Calibration End
[INFO] [1609833144.539697]: Setup publisher on sensor_state [turtlebot3_msgs/SensorState]
[INFO] [1609833144.566786]: Setup publisher on firmware_version [turtlebot3_msgs/VersionInfo]
[INFO] [1609833144.587723]: Setup publisher on imu [sensor_msgs/Imu]
[INFO] [1609833144.611058]: Setup publisher on cmd_vel_rc100 [geometry_msgs/Twist]
[INFO] [1609833144.630899]: Setup publisher on odom [nav_msgs/Odometry]
[INFO] [1609833144.655117]: Setup publisher on joint_states [sensor_msgs/JointState]
[ERROR] [1609833144.675072]: Creation of publisher failed: Checksum does not match: 4ddae7f048e32fda22cac764685e3974,476f837fa6771f6e16e3bf4ef96f8770
[INFO] [1609833144.696934]: Setup publisher on magnetic_field [sensor_msgs/MagneticField]
[INFO] [1609833144.722701]: Setup publisher on /tf [tf/tfMessage]
[ERROR] [1609833144.761317]: Tried to publish before configured, topic id 131
[INFO] [1609833144.778485]: Requesting topics...
[ERROR] [1609833145.150760]: Tried to publish before configured, topic id 131
[INFO] [1609833145.178375]: Requesting topics...
[INFO] [1609833145.216615]: Setup publisher on joint_states [sensor_msgs/JointState]
[ERROR] [1609833145.243747]: Creation of publisher failed: Checksum does not match: 4ddae7f048e32fda22cac764685e3974,476f837fa6771f6e16e3bf4ef96f8770
[INFO] [1609833145.267522]: Setup publisher on magnetic_field [sensor_msgs/MagneticField]
[INFO] [1609833145.289139]: Setup publisher on /tf [tf/tfMessage]
[ERROR] [1609833145.365157]: Tried to publish before configured, topic id 131
[INFO] [1609833145.379529]: Requesting topics...
[INFO] [1609833145.443247]: Setup publisher on sensor_state [turtlebot3_msgs/SensorState]
[INFO] [1609833145.470302]: Setup publisher on firmware_version [turtlebot3_msgs/VersionInfo]
[INFO] [1609833145.494520]: Setup publisher on imu [sensor_msgs/Imu]
[INFO] [1609833145.520371]: Setup publisher on cmd_vel_rc100 [geometry_msgs/Twist]
[INFO] [1609833145.542923]: Setup publisher on odom [nav_msgs/Odometry]
[INFO] [1609833145.571301]: Setup publisher on joint_states [sensor_msgs/JointState]
[ERROR] [1609833145.591552]: Creation of publisher failed: Checksum does not match: 4ddae7f048e32fda22cac764685e3974,476f837fa6771f6e16e3bf4ef96f8770
[INFO] [1609833145.615149]: Setup publisher on magnetic_field [sensor_msgs/MagneticField]
[INFO] [1609833145.649438]: Setup publisher on /tf [tf/tfMessage]
[ERROR] [1609833145.748313]: Tried to publish before configured, topic id 131
[INFO] [1609833145.764523]: Requesting topics...
[ERROR] [1609833145.797689]: Mismatched protocol version in packet (b'\xef'): lost sync or rosserial_python is from different ros release than the rosserial client
[INFO] [1609833145.814205]: Protocol version of client is unrecognized, expected Rev 1 (rosserial 0.5+)
[INFO] [1609833145.851688]: Setup publisher on sensor_state [turtlebot3_msgs/SensorState]
[INFO] [1609833145.873260]: Setup publisher on firmware_version [turtlebot3_msgs/VersionInfo]
[INFO] [1609833145.892614]: Setup publisher on imu [sensor_msgs/Imu]
[INFO] [1609833145.912408]: Setup publisher on cmd_vel_rc100 [geometry_msgs/Twist]
[INFO] [1609833145.934848]: Setup publisher on odom [nav_msgs/Odometry]
[INFO] [1609833145.957789]: Setup publisher on joint_states [sensor_msgs/JointState]
[ERROR] [1609833145.976967]: Creation of publisher failed: Checksum does not match: 4ddae7f048e32fda22cac764685e3974,476f837fa6771f6e16e3bf4ef96f8770
[INFO] [1609833146.000128]: Setup publisher on magnetic_field [sensor_msgs/MagneticField]
[INFO] [1609833146.022831]: Setup publisher on /tf [tf/tfMessage]
[ERROR] [1609833146.111883]: Tried to publish before configured, topic id 131
[INFO] [1609833146.130072]: Requesting topics...
[INFO] [1609833146.177017]: Setup publisher on sensor_state [turtlebot3_msgs/SensorState]
[INFO] [1609833146.203036]: Setup publisher on firmware_version [turtlebot3_msgs/VersionInfo]
[INFO] [1609833146.227270]: Setup publisher on imu [sensor_msgs/Imu]
[INFO] [1609833146.250330]: Setup publisher on cmd_vel_rc100 [geometry_msgs/Twist]
[INFO] [1609833146.275131]: Setup publisher on odom [nav_msgs/Odometry]
[INFO] [1609833146.306628]: Setup publisher on joint_states [sensor_msgs/JointState]

LW-G38, 2021-01-05:
It looks like the error always happens on the line of battery_state [sensor_msgs/BatteryState] in the e-manual if this helps.

ROBOTIS-Will, 2021-01-05:
@LW-G38 
Hi, 
The checksum error occurs because there was a change in the message structure in Noetic.
Please update the OpenCR firmware via Arduino IDE as written in the eManual.
From the link below, select the `Noetic` button at the top to switch the contents for Noetic.
https://emanual.robotis.com/docs/en/platform/turtlebot3/opencr_setup/#opencr-setup

LW-G38, 2021-01-05:
@ROBOTIS-Will 
Thank you, Will. I think we are getting close.

After I revise turtlebot3_core_config.h and uncomment the NOETIC_SUPPORT defintion in the line 21 of the script, the following error shows up when uploading:

Arduino: 1.8.13 (Mac OS X), Board: "OpenCR Board, OpenCR Bootloader"

In file included from /Users/weiliang/Documents/Arduino/turtlebot3_core/turtlebot3_core.ino:19:0:
turtlebot3_core_config.h:168:14: error: 'BatteryState' in namespace 'sensor_msgs' does not name a type
 sensor_msgs::BatteryState battery_state_msg;
              ^
turtlebot3_core_config.h:169:52: error: 'battery_state_msg' was not declared in this scope
 ros::Publisher battery_state_pub("battery_state", &battery_state_msg);
                                                    ^
/Users/weiliang/Documents/Arduino/turtlebot3_core/turtlebot3_core.ino: In function 'void publishBatteryStateMsg()':
turtlebot3_core:305:3: error: 'battery_state_msg' was not declared in this scope
   battery_state_msg.header.stamp = rosNow();
   ^
exit status 1
'BatteryState' in namespace 'sensor_msgs' does not name a type

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

Is there a head file missing?

LW-G38, 2021-01-05:
@ROBOTIS-Will I think I got it, it seems like turtlebot3_core_config.h for the example of waffle is not fully updated.
I copy the code section of battery state in turtlebot3_core_config.h for the example of burger and it works.

closed 2021-01-05 by LW-G38

ROBOTIS-Will, 2021-01-05:
@LW-G38 Thank you for the feedback. I'll double check if I can reproduce the same problem that you had.
Also, the binary file for Noetic will be released soon.
Happy New Year!

tehnanmanna, 2021-04-16:
Hi,

I tried really hard to solve this issue but I don't understand what's the reason for this. 

 roslaunch turtlebot3_bringup turtlebot3_robot.launch
... logging to /home/ubuntu/.ros/log/3971a994-9e9f-11eb-907b-c83dd43d6ebb/roslaunch-ubuntu-3356.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://192.168.0.3:36065/

SUMMARY
========

PARAMETERS
 * /rosdistro: melodic
 * /rosversion: 1.14.10
 * /turtlebot3_core/baud: 115200
 * /turtlebot3_core/port: /dev/ttyACM0
 * /turtlebot3_core/tf_prefix: 
 * /turtlebot3_lds/frame_id: base_scan
 * /turtlebot3_lds/port: /dev/ttyUSB0

NODES
  /
    turtlebot3_core (rosserial_python/serial_node.py)
    turtlebot3_diagnostics (turtlebot3_bringup/turtlebot3_diagnostics)
    turtlebot3_lds (hls_lfcd_lds_driver/hlds_laser_publisher)

ROS_MASTER_URI=http://192.168.0.10:11311

process[turtlebot3_core-1]: started with pid [3365]
process[turtlebot3_lds-2]: started with pid [3366]
process[turtlebot3_diagnostics-3]: started with pid [3367]
[INFO] [1618569234.616125]: ROS Serial Python Node
[INFO] [1618569234.668727]: Connecting to /dev/ttyACM0 at 115200 baud
[INFO] [1618569236.793151]: Requesting topics...
[INFO] [1618569236.835786]: Note: publish buffer size is 1024 bytes
[INFO] [1618569236.852049]: Setup publisher on sensor_state [turtlebot3_msgs/SensorState]
[INFO] [1618569236.878350]: Setup publisher on firmware_version [turtlebot3_msgs/VersionInfo]
[INFO] [1618569237.290153]: Setup publisher on imu [sensor_msgs/Imu]
[INFO] [1618569237.310358]: Setup publisher on cmd_vel_rc100 [geometry_msgs/Twist]
[ERROR] [1618569237.397581]: Creation of publisher failed: nav_msgs
ROS path [0]=/opt/ros/melodic/share/ros
ROS path [1]=/home/ubuntu/catkin_ws/src
ROS path [2]=/opt/ros/melodic/share
[INFO] [1618569237.417449]: Setup publisher on joint_states [sensor_msgs/JointState]
[INFO] [1618569237.437296]: Setup publisher on battery_state [sensor_msgs/BatteryState]
[INFO] [1618569237.455953]: Setup publisher on magnetic_field [sensor_msgs/MagneticField]
[INFO] [1618569239.425774]: Setup publisher on /tf [tf/tfMessage]
[INFO] [1618569239.451975]: Note: subscribe buffer size is 1024 bytes
[INFO] [1618569239.461753]: Setup subscriber on cmd_vel [geometry_msgs/Twist]
[INFO] [1618569239.487198]: Setup subscriber on sound [turtlebot3_msgs/Sound]
[INFO] [1618569239.510308]: Setup subscriber on motor_power [std_msgs/Bool]
[INFO] [1618569239.533514]: Setup subscriber on reset [std_msgs/Empty]
[INFO] [1618569239.899785]: Setup TF on Odometry [odom]
[INFO] [1618569239.911091]: Setup TF on IMU [imu_link]
[INFO] [1618569239.922730]: Setup TF on MagneticField [mag_link]
[INFO] [1618569239.933266]: Setup TF on JointState [base_link]
[ERROR] [1618569239.951325]: Tried to publish before configured, topic id 129
[INFO] [1618569239.961800]: Requesting topics...
^C[turtlebot3_diagnostics-3] killing on exit
[turtlebot3_lds-2] killing on exit
[turtlebot3_core-1] killing on exit
[INFO] [1618569240.428040]: Send tx stop request
shutting down processing monitor...
... shutting down processing monitor complete
done

ROBOTIS-Will, 2021-04-19:
@tehnanmanna 
Please do NOT post the same issue on different threads.
https://github.com/ROBOTIS-GIT/turtlebot3/issues/261#issuecomment-821069333
It is strongly recommended to create a new thread when you don't know which thread your issue should be placed.

DeMischi, 2022-03-14:
Had the exact same problem as LW-G38, with the exact same errors. 

ROBOTIS-Will's answer is correct and solved the problem:

During setup, you have to change to noetic on this page: https://emanual.robotis.com/docs/en/platform/turtlebot3/opencr_setup/#opencr-setup

This will change the following: 

export OPENCR_MODEL=burger

to

export OPENCR_MODEL=burger_noetic

This is so subtle, that you will barely notice it, hence you probably have the setup wrong.

After I have done this, it installed the 1.2.6 firmeware of OpenCR and voilá!

No more error messages, problem is fixed!

Thanks!