ROBOTIS-GIT/turtlebot3 #694: The parameters file pointed to in `navigation2.launch.py` is not loaded

nielsvd, 2021-01-08:
ISSUE TEMPLATE ver. 0.4.0

2. Which ROS is working with TurtleBot3?

    - [ ] ROS 1 Kinetic Kame
    - [ ] ROS 1 Melodic Morenia
    - [ ] ROS 1 Noetic Ninjemys
    - [ ] ROS 2 Dashing Diademata
    - [ ] ROS 2 Eloquent Elusor
    - [x] ROS 2 Foxy Fitzroy
    - [ ] etc (Please specify your ROS Version here)

5. Which OS you installed on Remote PC?

    - [ ] Ubuntu 16.04 LTS (Xenial Xerus)
    - [ ] Ubuntu 18.04 LTS (Bionic Beaver)
    - [x] Ubuntu 20.04 LTS (Focal Fossa)
    - [ ] Windows 10
    - [ ] MAC OS X (Specify version)
    - [ ] etc (Please specify your OS here)

9. Please describe the issue in detail.

The parameters file defined in the `turtlebot3_navigation2 navigation2.launch.py` launch file is not properly loaded by the `nav2_bringup bringup_launch.py` launch file because the wrong argument keyword is used, see the following two links for the mismatch: 
    - https://github.com/ROBOTIS-GIT/turtlebot3/blob/300df090e2b8252c2b6e853c01ac34ddca220f98/turtlebot3_navigation2/launch/navigation2.launch.py#L75
    - https://github.com/ros-planning/navigation2/blob/77c1b742e959dda02acf6830c2a814f2278bb4a4/nav2_bringup/bringup/launch/bringup_launch.py#L71

In practice, the default parameters file as provided by `nav2_bringup` package is used.

ROBOTIS-Will, 2021-01-12:
@nielsvd 
Thank you for pointing this out. I'll review and update it for the next release.
Happy New Year!

ROBOTIS-Will, 2021-03-10:
Closing this issue as there isn't recent activity. Please feel free to reopen when necessary.

closed 2021-03-10 by ROBOTIS-Will

nielsvd, 2021-03-10:
The problem isn't actually fixed right? The argument key should still be renamed from `params` to `params_file`. I could be looking in the wrong place though.

ROBOTIS-Will, 2021-03-11:
Oh, my bad.
I forgot to commit the change to the branch.
Thanks for the kind ping :)