ROBOTIS-GIT/turtlebot3 #695: 'rosrun' not found

chuckl72, 2021-01-20:
ISSUE TEMPLATE ver. 0.4.0

1. Which TurtleBot3 platform do you use?

    - [ X] Burger
    - [ ] Waffle
    - [ ] Waffle Pi

2. Which ROS is working with TurtleBot3?

    - [ ] ROS 1 Kinetic Kame
    - [ ] ROS 1 Melodic Morenia
    - [X ] ROS 1 Noetic Ninjemys
    - [ ] ROS 2 Dashing Diademata
    - [ ] ROS 2 Eloquent Elusor
    - [ ] ROS 2 Foxy Fitzroy
    - [ ] etc (Please specify your ROS Version here)

3. Which SBC(Single Board Computer) is working on TurtleBot3?

    - [ ] Intel Joule 570x
    - [X ] Raspberry Pi 3B+
    - [ ] Raspberry Pi 4
    - [ ] etc (Please specify your SBC here)

4. Which OS you installed on SBC?

    - [ ] Raspbian distributed by ROBOTIS
    - [ ] Ubuntu MATE (16.04/18.04/20.04)
    - [X ] Ubuntu preinstalled server (18.04/20.04)
    - [ ] etc (Please specify your OS here)

5. Which OS you installed on Remote PC?

    - [ ] Ubuntu 16.04 LTS (Xenial Xerus)
    - [ ] Ubuntu 18.04 LTS (Bionic Beaver)
    - [X ] Ubuntu 20.04 LTS (Focal Fossa)
    - [ ] Windows 10
    - [ ] MAC OS X (Specify version)
    - [ ] etc (Please specify your OS here)

6. Specify the software and firmware version(Can be found from Bringup messages)

    - Software version: [x.x.x]
    - Firmware version: [x.x.x]

7. Specify the commands or instructions to reproduce the issue.

   rosrun turtlebot3_bringup create_udev_rules

8. Copy and Paste the error messages on terminal.

   Command 'rosrun' not found, but can be installed with:

sudo apt install rosbash

9. Please describe the issue in detail.
I'm trying to setup my Pi 3+ with Noetic following the instructions in the TurtleBot 3.2 SBC Setup.  Everything has loaded successfully up to the point of issuing the command "rosrun turtlebot3_bringup create_udev_rules".  Running that command fails.  The following are the messages.  Any help would be appreciated.
Chuck

ubuntu@ubuntu:~$ rosrun turtlebot3_bringup create_udev_rules

Command 'rosrun' not found, but can be installed with:

sudo apt install rosbash

ubuntu@ubuntu:~$ sudo apt install rosbash
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 rosbash : Depends: catkin but it is not going to be installed
E: Unable to correct problems, you have held broken packages.

ROBOTIS-Will, 2021-01-25:
Hi,

Looks like rosbash has not been installed correctly.
Could you enter the command below and try again?
```bash
$ sudo apt install ros-noetic-rosbash
```

chuckl72, 2021-01-26:
Will,
Thank you, the command worked.  However I have a new problem, the keyboard teleoperation doesn't work.  Roscore and Bringup both launched with out any errors.  However none of the keys, w, s, x, a, d work.  Topic monitor, rqt, brings up the default-rqt window which is blank.  Can you help me with this or should I start a new post?  I did a search and didn't find anything that helped.
Chuck

ROBOTIS-Will, 2021-01-26:
Have you correctly [configured the network](https://emanual.robotis.com/docs/en/platform/turtlebot3/quick-start/#network-configuration-2) for both remote PC and TurtleBot3 SBC?

If you haven't installed rqt plugins, you can install plugin with below command.
When running rqt for the first time with rqt command, it will be just an empty window, but you can select and open the plugin you want to see.
```bash
$ sudo apt-get install ros-noetic-rqt-*
```

chuckl72, 2021-01-26:
Will,

I pretty sure the network connections are correct, see the first two pictures below.  The first picture is from the Pi and the second is from the PC.  The topic list appears to show no connection with the Pi, see the third picture below.  After some tests, it appears that when roslaunch turtlebot3_bringup turtlebot3_robot.launch is run, the Pi is crashing, requiring it to be rebooted.  Any ideas?  

Chuck
[image]
[image] 
[image]

chuckl72, 2021-01-27:
Will,

I was able to get some runtime without it crashing.  One run was about 3 minutes, the next was 4.5 minutes, and the last one ran for quite a while.  This last run crashed when rqt was run.  There really doesn't seem to be a definite cause of it crashing.

Chuck

ROBOTIS-Will, 2021-01-27:
Chuck,

Thanks for sharing the captured images.
I'm pretty sure that you've followed the eManual instruction, but could you check if you have disabled sleep / suspend and automatic updates on your RPi?
```bash
$ sudo nano /etc/apt/apt.conf.d/20auto-upgrades
```
Edit the content 1 -> 0 as below
> APT::Periodic::Update-Package-Lists "0";
> APT::Periodic::Unattended-Upgrade "0";

Disable Suspend and Hibernation.
```bash
$ sudo systemctl mask sleep.target suspend.target hibernate.target hybrid-sleep.target
```

chuckl72, 2021-01-27:
Will,

I did both of those items during the original installation.  The picture shows what the upgrades were and I did rerun the Hibernation command.  Tried running the bringup and keyboard commands twice now, and it crashed right away.  Is it possible that one of the electric components is bad.  Is there a way to do any diagnostics on the Opencr board?  Could the Pi be an issue, I can swap it out with another one.  If you have any other suggestions, let me know.

Chuck

[image]

chuckl72, 2021-01-28:
Will,

Would you have a working image of Noetic for the Pi 3+ you could put in a drop box?  That might help to identify if I'm not having success in setting up the SD card.

Chuck

ROBOTIS-Will, 2021-01-29:
Hi Chuck,

Unfortunately, sharing a modified OS image is prohibited by Ubuntu policy.
I'll see if I can record my screen of Noetic installation process and share it with you.
This may take up to a couple weeks, but I'll do my best to bring this forward.
Thank you.

chuckl72, 2021-01-29:
Will

Ok on the image, no problem.  The BIG problem is the Turtlebot keeps crashing and I don't know what to do.  I just finished reloading my laptop as a clean install and reflashed the SD card, and it still crashes.  Just now, it crashed without touching the  keyboard w, s, etc keys.  Tomorrow I will try using my other Pi.  If you have any other ideas, let me know.

Chuck

chuckl72, 2021-01-29:
Will,

A little progress today.  Out of frustration, I reloaded my laptop with Ubuntu 16.04 and got the Turtlebot3 running via keyboard.  The only problem is that the wifi doesn't work with Ubuntu 16.04 on the laptop.  I will work on that for a while, but really want to get Noetic working with the laptop.  Wifi does work on the laptop with Ubuntu 20.04.  So, if I'm following the directions in the emanual for the Noetic installation, why is it crashing?? 

Chuck

ROBOTIS-Will, 2021-02-01:
Hi Chuck,
There could be many reasons of system crash, and micro SD card could be one of the reason.
Please refer to the Noetic on RPi 4 test installation video clip (there isn't much difference in setting up RPi3 and RPi4)
https://youtu.be/NeKiY4JB0Kk

chuckl72, 2021-02-01:
Will,

The video is hard for me to watch without any one talking.  I will try burning a new SD card tomorrow, up to this point I have been using BalenaEtcher on my Mac.  I will try burning another one using Ubuntu.  

I have been running the TurtleBot3 Kinetic successfully except for one issue.  The first time running it using the keyboard it worked fine.  After the first run, it will only run if it is booted using the battery and the USB cable disconnected from the Bot.

Chuck

closed 2021-02-22 by chuckl72

Andres01032002, 2023-09-10:
Is the forum still active?