ROBOTIS-GIT/turtlebot3 #697: bash: rosrun: command not found

Schellimaen, 2021-01-25:
ISSUE TEMPLATE ver. 0.4.0

1. Which TurtleBot3 platform do you use?

    - [ ] Burger
    - [ ] Waffle
    - [X] Waffle Pi

2. Which ROS is working with TurtleBot3?

    - [ ] ROS 1 Kinetic Kame
    - [ ] ROS 1 Melodic Morenia
    - [X] ROS 1 Noetic Ninjemys
    - [ ] ROS 2 Dashing Diademata
    - [ ] ROS 2 Eloquent Elusor
    - [ ] ROS 2 Foxy Fitzroy
    - [ ] etc (Please specify your ROS Version here)

3. Which SBC(Single Board Computer) is working on TurtleBot3?

    - [ ] Intel Joule 570x
    - [X] Raspberry Pi 3B+
    - [ ] Raspberry Pi 4
    - [ ] etc (Please specify your SBC here)

4. Which OS you installed on SBC?

    - [X] Raspbian distributed by ROBOTIS
    - [ ] Ubuntu MATE (16.04/18.04/20.04)
    - [ ] Ubuntu preinstalled server (18.04/20.04)
    - [ ] etc (Please specify your OS here)

5. Which OS you installed on Remote PC?

    - [ ] Ubuntu 16.04 LTS (Xenial Xerus)
    - [ ] Ubuntu 18.04 LTS (Bionic Beaver)
    - [X] Ubuntu 20.04 LTS (Focal Fossa)
    - [ ] Windows 10
    - [ ] MAC OS X (Specify version)
    - [ ] etc (Please specify your OS here)

6. Specify the software and firmware version(Can be found from Bringup messages)

    - Software version: Release 3.6 - January 2021
    - Firmware version: Linux raspberrypi 5.4.83-v7+

7. Specify the commands or instructions to reproduce the issue.

    - pi@raspberrypi:~/ros_catkin_ws $ rosrun turtlebot3_bringup create_udev_rules

8. Copy and Paste the error messages on terminal.

    - bash: rosrun: command not found

9. Please describe the issue in detail.

    I had problems installing ROS according to the instructions because the installation with the command '$ sudo apt install' did not work.
See instructions in the link below: 
https://emanual.robotis.com/docs/en/platform/turtlebot3/sbc_setup/#sbc-setup

That's why I installed ROS according to the instruction in the link below: 
http://wiki.ros.org/noetic/Installation/Source

I have already installed 'rosbash' without success. I still cannot run 'rosrun'.

I am new to the field and hope you can help me.

Best regards

ROBOTIS-Will, 2021-01-26:
Hi,
Have you successfully installed rosbash with the command below?
```bash
$ sudo apt install ros-noetic-rosbash
```

Please try loading the ROS setup file as below and see if this fixes the issue.
```bash
$ source /opt/ros/noetic/setup.bash
```

Thank you.

Schellimaen, 2021-01-26:
Hi,
When I run 'sudo apt install', I always get the error that the requested packages couldn't be found, as in the command below:

` $ sudo apt install ros-noetic-rosbash
Reading package lists... Done
Building dependency tree       
Reading state information... Done
E: Unable to locate package ros-noetic-rosbash`

If I run the second command I get the following failure:

`$ source /opt/ros/noetic/setup.bash
-bash: /opt/ros/noetic/setup.bash: No such file or directory`

Because 'sudo apt install' doesn't work generally with ROS-applications, I tried to install everything from source. As in the Link http://wiki.ros.org/noetic/Installation/Source.

Best regards

ROBOTIS-Will, 2021-01-26:
It looks like ROS has not been installed properly.
Have you connected your Raspberry Pi to the network successfully?
Does the command below returns the string "noetic"?
```bash
$ printenv ROS_DISTRO
```

Schellimaen, 2021-01-26:
Hi,

I know, I have to click on Noetic. But I didn't install Unbuntu on Raspberry. I installed Raspberry Pi OS with desktop and recommended software.

When I run your last command i get nothing
[image]

Best regards

ROBOTIS-Will, 2021-01-27:
Hi,

Raspbian has several limitations when installing ROS so it is not recommended for beginners.
If you are going to use Raspbian for your Raspberry Pi, you should not follow the eManual.
Thank you.

Schellimaen, 2021-01-27:
Hi,
Ok nice to know. 
Thank you.

closed 2021-01-27 by Schellimaen