ROBOTIS-GIT/turtlebot3 #698: Dynamixel Stopped Working after a Few Moments

LW-G38, 2021-02-08:
ISSUE TEMPLATE ver. 0.4.0

1. Which TurtleBot3 platform do you use?

    - [ ] Burger
    - [ ] Waffle
    - [* ] Waffle Pi

2. Which ROS is working with TurtleBot3?

    - [ ] ROS 1 Kinetic Kame
    - [ ] ROS 1 Melodic Morenia
    - [ *] ROS 1 Noetic Ninjemys
    - [ ] ROS 2 Dashing Diademata
    - [ ] ROS 2 Eloquent Elusor
    - [ ] ROS 2 Foxy Fitzroy
    - [ ] etc (Please specify your ROS Version here)

3. Which SBC(Single Board Computer) is working on TurtleBot3?

    - [ ] Intel Joule 570x
    - [ ] Raspberry Pi 3B+
    - [ ] Raspberry Pi 4
    - [ ] etc (Please specify your SBC here)

4. Which OS you installed on SBC?

    - [ ] Raspbian distributed by ROBOTIS
    - [ ] Ubuntu MATE (16.04/18.04/20.04)
    - [ ] Ubuntu preinstalled server (18.04/20.04)
    - [ ] etc (Please specify your OS here)

5. Which OS you installed on Remote PC?

    - [ ] Ubuntu 16.04 LTS (Xenial Xerus)
    - [ ] Ubuntu 18.04 LTS (Bionic Beaver)
    - [* ] Ubuntu 20.04 LTS (Focal Fossa)
    - [ ] Windows 10
    - [ ] MAC OS X (Specify version)
    - [ ] etc (Please specify your OS here)

6. Specify the software and firmware version(Can be found from Bringup messages)

    - Software version: [1.15.9]
    - Firmware version: [1.2.6]

7. Specify the commands or instructions to reproduce the issue.

    - roslaunch turtlebot3_slam turtlebot3_slam.launch
    - roslaunch turtlebot3_teleop turtlebot3_teleop_key.launch

8. Copy and Paste the error messages on terminal.

    - N/A

9. Please describe the issue in detail.

    - When entering SLAM mode using telepo (https://emanual.robotis.com/docs/en/platform/turtlebot3/slam/#run-slam-node), the robot moves okay at the beginning. But after a few moments, the robot will stop moving without having any obstacles nearby, and when I check the dynamixel motors, I find them locked and the wheel cannot spin anymore. There is no battery shortage. Not sure if there are any operational mistakes from me, or it is a bug.

ROBOTIS-Will, 2021-02-09:
Hi,
When the robot is stalled, do you see any updated topic values coming from the robot?
If the reason of stall can be identified such as the RPi stall or corrupted cmd_vel topic, it might be easier to find the resolution.

ROBOTIS-Will, 2021-03-10:
Closing this issue as there isn't recent activity. Please feel free to reopen when necessary.

closed 2021-03-10 by ROBOTIS-Will

LW-G38, 2021-04-12:
Hi Will,

There is no updated topic values coming from the robot. I found out that this situations happens not only under SLAM teleop mode, but also normal teleop mode.

When this happens in SLAM mode, I killed the SLAM process and relaunch it, and find the map is empty and also I cannot make the robot move unless I reboot the robot.