ROBOTIS-GIT/turtlebot3 #700: TurtleBot3 Wifi Crash

chuckl72, 2021-02-11:
ISSUE TEMPLATE ver. 0.4.0

1. Which TurtleBot3 platform do you use?

    - [x ] Burger
    - [ ] Waffle
    - [ ] Waffle Pi

2. Which ROS is working with TurtleBot3?

    - [ ] ROS 1 Kinetic Kame
    - [x ] ROS 1 Melodic Morenia
    - [ ] ROS 1 Noetic Ninjemys
    - [ ] ROS 2 Dashing Diademata
    - [ ] ROS 2 Eloquent Elusor
    - [ ] ROS 2 Foxy Fitzroy
    - [ ] etc (Please specify your ROS Version here)

3. Which SBC(Single Board Computer) is working on TurtleBot3?

    - [ ] Intel Joule 570x
    - [ x] Raspberry Pi 3B+
    - [ ] Raspberry Pi 4
    - [ ] etc (Please specify your SBC here)

4. Which OS you installed on SBC?

    - [ x] Raspbian distributed by ROBOTIS
    - [ ] Ubuntu MATE (16.04/18.04/20.04)
    - [ ] Ubuntu preinstalled server (18.04/20.04)
    - [ ] etc (Please specify your OS here)

5. Which OS you installed on Remote PC?

    - [ ] Ubuntu 16.04 LTS (Xenial Xerus)
    - [ x] Ubuntu 18.04 LTS (Bionic Beaver)
    - [ ] Ubuntu 20.04 LTS (Focal Fossa)
    - [ ] Windows 10
    - [ ] MAC OS X (Specify version)
    - [ ] etc (Please specify your OS here)

6. Specify the software and firmware version(Can be found from Bringup messages)

    - Software version: [x.x.x]
    - Firmware version: [x.x.x]

7. Specify the commands or instructions to reproduce the issue.

    rosrun turtlebot3_bringup create_udev_rules

8. Copy and Paste the error messages on terminal.

    - HERE

9. Please describe the issue in detail.

    I've been struggling for a month to get the Turtlebot3 running.  Noetic crashes within minutes of starting up.  After many attempts I gave up, and successfully got Kinetic to work, however the wifi on my HP laptop won't work with Ubuntu 16.04.  This limits the area to scan to a very small area.  Now, I was able to get Melodic running successfully with a ethernet cable connected.  However when I went to set up the wifi in the .yaml file on the TurtleBot3, it won't work.  I flashed a new SD card and ran all the commands using the wifi connection.  After running the "rosrun turtlebot3_bringup create_udev_rules" command and rebooting, the wifi doesn't work anymore.  

Does anyone now the trick to get either Melodic or Noetic to work properly?  I'm using the instructions in the emanual.

Chuck

ROBOTIS-Will, 2021-02-15:
@chuckl72 
Sorry about the inconvenience.
I've been contacting a relative company since last year to find a way to share the RPi3 SD card image and I'm waiting for the final confirmation.
As soon as I hear from them, I'll share the image with you.
In the meantime, could you try setting up the country code for your wifi setting?
https://www.raspberrypi.org/forums/viewtopic.php?t=208591

chuckl72, 2021-02-15:
Will,
I wasn't successful in editing the wpa_supplicant.conf in order to make the wifi work.  I'm sure I didn't something wrong.  The wpa_supplicant.conf was an empty file.  This is what I added to the file:

ctrl_interface=DIR=/var/run/wpa_supplicant GROUP=netdev
update_config=1
country=US

network={
        ssid="ARRIS-8DA9"
        psk="6e5MgwF3[|5gCR"
        key_mgmt=WPA-PSK
}

ROBOTIS-Will, 2021-02-16:
@chuckl72 
I'll test the SBC setup instruction for Noetic on my RPi 3B+ and get back to you.
I might be able to create a micro SD image after the test.

chuckl72, 2021-02-16:
Will,
Thank you, I await your results.
Chuck

ROBOTIS-Will, 2021-02-17:
@chuckl72 
Hi, I've created a [test recovery image for RPi3B+ & Noetic](https://drive.google.com/file/d/18q0epskE9SOokeDwrXTOLniZESu5G6R6/view?usp=sharing). This image file will be available for the next 3 days so please save it if you need for later.
The below procedure has been tested with my Burger.

1. Download the image file.
2. Extracting the file and burn the `img` file to microSD card. I used `disks` utility in Ubuntu.
3. Connect monitor & keyboard to RPI3B+.
4. Turn on RPi and login with ID : `ubuntu`, Password : `turtlebot`
5. Replace your `WIFI_SSID` and `WIFI_PASSWORD` to `50-cloud-init.yaml` file using below command.
```bash
$ sudo nano /etc/netplan/50-cloud-init.yaml
```
6. Save the change and reboot
```bash
$ sudo reboot
```

chuckl72, 2021-02-17:
Will,

It works!!!! Thank you!  I've been able to drive the Turtlebot3 with the keyboard without any issues.  However I'm having an issue with SLAM.  When running the command "roslaunch turtlebot3_slam turtlebot3_slam.launch", errors shown in the picture happen.  I thought I ran all the installation commands on the PC.

Chuck

[image]

ROBOTIS-Will, 2021-02-18:
@chuckl72 
Great to hear that the image works for you.
It looks like the RViz has not been installed.
Please use below command to install the RViz and related plugins.
```bash
$ sudo apt-get install ros-noetic-rviz*
```

chuckl72, 2021-02-18:
Will,

Thanks again, RViz is working now.

Chuck

chuckl72, 2021-02-19:
Will, 

I could use your help with two new issues when trying to run SLAM.  First of all there are no plugins when running rqt.  This screen shot is all that comes up, and under plugins only an empty folders for Container and Visualization appear.
[image]

The other issue is with the RViz.  The screen shot shows jumbled activity with the Turtlebot3 sitting in a fixed location for 15 minutes.
[image]

Chuck

ROBOTIS-Will, 2021-02-22:
Hi @chuckl72

When reporting a new issue, please create another ticket so that other developers can refer to the thread to resolve similar issues.

With regard to the rqt issue, you need to install plugins that you are going to use, and the below is the command that installs all available plugins.
```bash
$ sudo apt-get install ros-noetic-rqt*
```

I've run the SLAM for about 20 minutes but couldn't reproduce the issue you are experiencing.
I do not see the LaserScan(green dots on my image below) on your RViz window and the angle of the sensor data doesn't seem normal.
Have you checked whether you are getting the scan topic from the TurtleBot3?

[image]

chuckl72, 2021-02-22:
Will,

Thanks for the rqt help.  Unless you see a quick fix on the scanning, I will start a new issue.

Chuck

[image]
[image]

ROBOTIS-Will, 2021-02-22:
Looks like you are only receiving half of the scan topic as the frequency is 2.21Hz.
The frequency should be close to 5Hz (LDS sensor turns 5 times per second).
[image]

This could be due to the low bandwidth wifi router, or the dropping invalid data, or LDS-01 hardware issue in worst case.

Try time sync on both PC and RPi as instructed below.
https://emanual.robotis.com/docs/en/platform/turtlebot3/faq/#timesync-between-turtlebot3-and-remote-pc

Instead of bringup from TurtleBot3, try below command and see if you are getting higher frequency.
```bash
$ roslaunch hls_lfcd_lds_driver hlds_laser.launch
```
[image]

chuckl72, 2021-02-22:
Will,

I started a new issue on the scanner problem, so you can close this issue.
https://github.com/ROBOTIS-GIT/turtlebot3/issues/710

Chuck

closed 2021-02-22 by chuckl72