ROBOTIS-GIT/turtlebot3 #702: SBC setup difficulty

johncabibihan, 2021-02-13:
ISSUE TEMPLATE ver. 0.4.0

1. Which TurtleBot3 platform do you use?

    - [ ] Burger
    - [ ] Waffle
    - [x] Waffle Pi

2. Which ROS is working with TurtleBot3?

    - [ ] ROS 1 Kinetic Kame
    - [ ] ROS 1 Melodic Morenia
    - [x] ROS 1 Noetic Ninjemys
    - [ ] ROS 2 Dashing Diademata
    - [ ] ROS 2 Eloquent Elusor
    - [ ] ROS 2 Foxy Fitzroy
    - [ ] etc (Please specify your ROS Version here)

3. Which SBC(Single Board Computer) is working on TurtleBot3?

    - [ ] Intel Joule 570x
    - [x] Raspberry Pi 3B+
    - [ ] Raspberry Pi 4
    - [ ] etc (Please specify your SBC here)

4. Which OS you installed on SBC?

    - [ ] Raspbian distributed by ROBOTIS
    - [ ] Ubuntu MATE (16.04/18.04/20.04)
    - [ ] Ubuntu preinstalled server (18.04/20.04)
    - [x] etc (Please specify your OS here) Ubuntu 20.04.1 Preinstalled Server for Raspberry Pi3 as shown on the Noetic tab at this link on Sec. 3.2.1 [link](https://emanual.robotis.com/docs/en/platform/turtlebot3/sbc_setup/#sbc-setup)

5. Which OS you installed on Remote PC?

    - [ ] Ubuntu 16.04 LTS (Xenial Xerus)
    - [ ] Ubuntu 18.04 LTS (Bionic Beaver)
    - [x] Ubuntu 20.04 LTS (Focal Fossa)
    - [ ] Windows 10
    - [ ] MAC OS X (Specify version)
    - [ ] etc (Please specify your OS here)

6. Specify the software and firmware version(Can be found from Bringup messages)

    - Software version: [x.x.x]
    - Firmware version: [x.x.x]

7. Specify the commands or instructions to reproduce the issue.

    - I followed the 10 steps in "Configure the Raspberry Pi" at this [link](https://emanual.robotis.com/docs/en/platform/turtlebot3/sbc_setup/#sbc-setup). On step 10, I could not connect to the RPi.

8. Copy and Paste the error messages on terminal.

    - ssh ubuntu@xxx.xxx.x.xxx      
    - ssh: connect to host xxx.xxx.x.xxx port 22: No route to host

9. Please describe the issue in detail.

    - On the RPi, I tried ifconfig. The message is: Command 'ifconfig' not found.

Paulino1956, 2021-02-14:
Suggest you to consult : How to Enable SSH on Ubuntu 20.04 at https://linuxize.com/post/how-to-enable-ssh-on-ubuntu-20-04/.

ROBOTIS-Will, 2021-02-15:
@johncabibihan 
Make sure that you have acquired IP address after writing the `50-cloud-init.yaml` file and rebooting the Raspberry Pi.
The Noetic SBC instruction describes how to install and enable SSH as below after setting up the WIFI network.
```bash
$ sudo apt install ssh
$ sudo systemctl enable --now ssh
$ reboot
```

johncabibihan, 2021-02-15:
@ROBOTIS-Will 
Thanks for your kind reply. I have doubts if I properly appended the text at the end of the file in the Noetic instructions
$ sudo nano /etc/netplan/50-cloud-init.yaml

I just copied the text and added my WIFI_SSID and WIFI_PASSWORD in the location as shown in the instructions. Is everything correct in this file? Please advise. Thanks.

[image]

johncabibihan, 2021-02-15:
@Paulino1956 

Many thanks! I followed the instructions in the link. Here's what I got. I know I'm doing something wrong, but I don't know where. Any help will be very much appreciated.

[image]

ROBOTIS-Will, 2021-02-16:
@johncabibihan 
There's a typo in `access-point:` which should be revised to `access-points:`.
Raspberry Pi wifi could be very tricky.
Have you tried setting the wifi country code?

johncabibihan, 2021-02-18:
@ROBOTIS-Will 

Many thanks. I have changed the  access-points with an "s". Still no success.

I tried to change the wifi country code using various online suggestions. 

The year is goes back to 2020 - very strange. 

Any help would be very much appreciated. Thanks.

ROBOTIS-Will, 2021-02-19:
@johncabibihan 
Hi, could you try the Noetic image that I shared in the other thread #700?

johncabibihan, 2021-02-23:
@ROBOTIS-Will 

Success! It worked. Thanks! 

Which step do I follow next? 

3.2.5 Install ROS Packages?
https://emanual.robotis.com/docs/en/platform/turtlebot3/sbc_setup/#sbc-setup

or 

3.5 Bringup?
https://emanual.robotis.com/docs/en/platform/turtlebot3/bringup/#bringup

or something else?

ROBOTIS-Will, 2021-02-24:
@johncabibihan 
Once you've recovered your RPi with provided image, you don't have to install additional packages on RPi.
But don't forget to configure the network setting part in the `.bashrc` file below.
```bash
export ROS_MASTER_URI=http://{IP_ADDRESS_OF_REMOTE_PC}:11311
export ROS_HOSTNAME={IP_ADDRESS_OF_RASPBERRY_PI_3}
```

The next step will be uploading the Noetic firmware on OpenCR.
https://emanual.robotis.com/docs/en/platform/turtlebot3/opencr_setup/#opencr-setup
Then, you'll be ready for the Bringup and teleopration :D

johncabibihan, 2021-02-26:
@ROBOTIS-Will 

Thanks for your suggestion. 

1. I configured the network according to your instructions. 
2. I have performed the set-up for OpenCR and assembled the hardware. The remote control operation works. 
3. On the Bringup, I typed roscore on the PC. Below is the screenshot. 
4. After seeing that message, I pinged the IP addresses of the PC and the RPI, both are sending and receiving from one another. 

Your expert advise will be very much appreciated. 

[image]

ROBOTIS-Will, 2021-02-26:
@johncabibihan 
In your deleted picture, I see a typo when setting `export ROS_HOSTNAME=192.18.0.190`, but should be `192.168.0.190`.

johncabibihan, 2021-02-26:
@ROBOTIS-Will 

My apologies. I sent the wrong photo, that's why I deleted it immediately.

Here are the correct screenshots...

On the bashrc file...

[image]

After typing roscore on the PC. Then, pinging the RPi's IP address.

[image]

Also pinging the PC's IP address...

[image]

ROBOTIS-Will, 2021-03-08:
@johncabibihan 
I'm sorry about the delayed response.
Could you show me the configuration of the ROS_MASTER_URI and the ROS_HOSTNAME on your Remote PC?
I'm expecting to see something like 
```bash
export ROS_MASTER_URI=http://192.168.0.149:11311
export ROS_HOSTNAME=192.168.0.149
```
Also, did you run `$ roscore` from your Remote PC before bringing up the TurtleBot3 from Raspberry Pi?

johncabibihan, 2021-03-18:
@ROBOTIS-Will:
Thanks for your message. 

On my remote PC, this is what shows up. 

export ROS_MASTER_URI=http://192.168.0.149:11311
export ROS_HOSTNAME=192.168.0.190

Running roscore from remote PC before bringup, this is what shows up:

RLException: Unable to contact my own server at [http://192.168.0.190:38193/].
This usually means that the network is not configured properly.

A common cause is that the machine cannot connect to itself.  Please check
for errors by running:

	ping 192.168.0.190

Please advise.

ROBOTIS-Will, 2021-03-18:
@johncabibihan 
If the IP address of your remote PC is 192.168.0.149, you should have an identical IP address for ROS_MASTER_URI and ROS_HOSTNAME on your remote PC as below.

```
export ROS_MASTER_URI=http://192.168.0.149:11311
export ROS_HOSTNAME=192.168.0.149
```

Meanwhile on your SBC(let's say its IP address is 192.168.0.190), it should look like :
```
export ROS_MASTER_URI=http://192.168.0.149:11311
export ROS_HOSTNAME=192.168.0.190
```

johncabibihan, 2021-03-18:
@ROBOTIS-Will 

Thanks for that. I followed your instructions.  

On my PC, I ran roscore. Does it really take long for anything to change on the screen?

SUMMARY
========

PARAMETERS
 * /rosdistro: noetic
 * /rosversion: 1.15.9

NODES

auto-starting new master
process[master]: started with pid [4353]
ROS_MASTER_URI=http://192.168.0.149:11311/

setting /run_id to d4ba7016-87a7-11eb-a140-d158c09527d5
process[rosout-1]: started with pid [4363]
started core service [/rosout]

ROBOTIS-Will, 2021-03-18:
The pasted output looks good.
The `roscore` doesn't do anything but creating the ros master node for other nodes.
Once roscore is up and running, you can bringup the TurtleBot3 from RPi3
https://emanual.robotis.com/docs/en/platform/turtlebot3/bringup/#bringup-turtlebot3-1

Paulino1956, 2021-03-18:
Thanks. 

All key functions are working : bringup, teleop via keyboard, SLAM, RViZ, GMap and Navigation. I will get started now on the next steps. 
I am still having issues executing $rqt, but will move forward. 

Thanks again for all your support! 

Paulino

johncabibihan, 2021-03-19:
@ROBOTIS-Will 

Thanks for the instructions. I missed the "Open a new terminal from PC" part. 

With the terminal with the roscore running and copying below to the new terminal: $ roslaunch turtlebot3_bringup turtlebot3_robot.launch

I seem to get some of the expected output as copied below. However, the error messages seem to be in an infinite loop. I left it running for 8 hours, non-stop. 

Any advise?   

started roslaunch server http://192.168.0.190:40907/

SUMMARY
========

PARAMETERS
 * /rosdistro: noetic
 * /rosversion: 1.15.9
 * /turtlebot3_core/baud: 115200
 * /turtlebot3_core/port: /dev/ttyACM0
 * /turtlebot3_core/tf_prefix: 
 * /turtlebot3_lds/frame_id: base_scan
 * /turtlebot3_lds/port: /dev/ttyUSB0

NODES
  /
    turtlebot3_core (rosserial_python/serial_node.py)
    turtlebot3_diagnostics (turtlebot3_bringup/turtlebot3_diagnostics)
    turtlebot3_lds (hls_lfcd_lds_driver/hlds_laser_publisher)

ROS_MASTER_URI=http://192.168.0.149:11311

process[turtlebot3_core-1]: started with pid [37153]
process[turtlebot3_lds-2]: started with pid [37154]
process[turtlebot3_diagnostics-3]: started with pid [37155]
[ERROR] [1616094415.381769960]: An exception was thrown: open: No such file or directory
[turtlebot3_lds-2] process has died [pid 37154, exit code 255, cmd /opt/ros/noetic/lib/hls_lfcd_lds_driver/hlds_laser_publisher __name:=turtlebot3_lds __log:=/home/ubuntu/.ros/log/7e2b8ce8-87b2-11eb-8b5e-27feae547c66/turtlebot3_lds-2.log].
log file: /home/ubuntu/.ros/log/7e2b8ce8-87b2-11eb-8b5e-27feae547c66/turtlebot3_lds-2*.log
[INFO] [1616094416.532680]: ROS Serial Python Node
[INFO] [1616094416.571050]: Connecting to /dev/ttyACM0 at 115200 baud
[INFO] [1616094418.692333]: Requesting topics...
[INFO] [1616094418.724689]: Note: publish buffer size is 1024 bytes
[INFO] [1616094418.732629]: Setup publisher on sensor_state [turtlebot3_msgs/SensorState]
[INFO] [1616094418.752156]: Setup publisher on firmware_version [turtlebot3_msgs/VersionInfo]
[INFO] [1616094418.881152]: Setup publisher on imu [sensor_msgs/Imu]
[INFO] [1616094418.905100]: Setup publisher on cmd_vel_rc100 [geometry_msgs/Twist]
[INFO] [1616094418.954475]: Setup publisher on odom [nav_msgs/Odometry]
[INFO] [1616094418.972352]: Setup publisher on joint_states [sensor_msgs/JointState]
[ERROR] [1616094418.984951]: Creation of publisher failed: Checksum does not match: 4ddae7f048e32fda22cac764685e3974,476f837fa6771f6e16e3bf4ef96f8770
[INFO] [1616094419.006905]: Setup publisher on magnetic_field [sensor_msgs/MagneticField]
[INFO] [1616094419.711700]: Setup publisher on /tf [tf/tfMessage]
[INFO] [1616094419.737563]: Note: subscribe buffer size is 1024 bytes
[INFO] [1616094419.744959]: Setup subscriber on cmd_vel [geometry_msgs/Twist]
[INFO] [1616094419.769180]: Setup subscriber on sound [turtlebot3_msgs/Sound]
[INFO] [1616094419.792692]: Setup subscriber on motor_power [std_msgs/Bool]
[INFO] [1616094419.815269]: Setup subscriber on reset [std_msgs/Empty]
[INFO] [1616094421.838482]: Setup TF on Odometry [odom]
[INFO] [1616094421.851232]: Setup TF on IMU [imu_link]
[INFO] [1616094421.862070]: Setup TF on MagneticField [mag_link]
[INFO] [1616094421.875912]: Setup TF on JointState [base_link]
[ERROR] [1616094421.889750]: Tried to publish before configured, topic id 131
[INFO] [1616094421.898921]: Requesting topics...
[INFO] [1616094424.398676]: Calibration End
[INFO] [1616094424.420109]: Setup publisher on sensor_state [turtlebot3_msgs/SensorState]
[INFO] [1616094424.431175]: Setup publisher on firmware_version [turtlebot3_msgs/VersionInfo]
[INFO] [1616094424.441901]: Setup publisher on imu [sensor_msgs/Imu]
[INFO] [1616094424.454527]: Setup publisher on cmd_vel_rc100 [geometry_msgs/Twist]
[INFO] [1616094424.466410]: Setup publisher on odom [nav_msgs/Odometry]
[INFO] [1616094424.478574]: Setup publisher on joint_states [sensor_msgs/JointState]
[ERROR] [1616094424.489981]: Creation of publisher failed: Checksum does not match: 4ddae7f048e32fda22cac764685e3974,476f837fa6771f6e16e3bf4ef96f8770
[INFO] [1616094424.501803]: Setup publisher on magnetic_field [sensor_msgs/MagneticField]
[INFO] [1616094424.512158]: Setup publisher on /tf [tf/tfMessage]
[ERROR] [1616094424.536448]: Tried to publish before configured, topic id 131
[INFO] [1616094424.543837]: Requesting topics...
[ERROR] [1616094424.704352]: Tried to publish before configured, topic id 131
[INFO] [1616094424.712627]: Requesting topics...
[ERROR] [1616094424.747251]: Tried to publish before configured, topic id 131
[INFO] [1616094424.755708]: Requesting topics...
[ERROR] [1616094424.821037]: Tried to publish before configured, topic id 131
[INFO] [1616094424.828456]: Requesting topics...
[ERROR] [1616094424.864352]: Tried to publish before configured, topic id 131
[INFO] [1616094424.873162]: Requesting topics...
[ERROR] [1616094424.946249]: Tried to publish before configured, topic id 131

This can go on for 8 hours or more...

ROBOTIS-Will, 2021-03-19:
Hi @johncabibihan 
When running TurtleBot3 with Noetic, you should update the OpenCR firmware as there is an updated message structure that is not compatible with older ROS versions.
Please follow the OpenCR setup section of the eManual, and don't forget to select the correct ROS version you are trying to use at the top of the eManual page.
https://emanual.robotis.com/docs/en/platform/turtlebot3/opencr_setup/#opencr-setup
[image]

johncabibihan, 2021-03-19:
@ROBOTIS-Will 

Success! It worked. No error messages. What a joy!

On 3.5.3. Load TurtleBot3 on Rviz, I received the error below. 
https://emanual.robotis.com/docs/en/platform/turtlebot3/bringup/#bringup

Please advise. Thanks. 

$ rosrun rviz rviz -d 'rospack find turtlebot3_description'/rviz/model.rviz
[rospack] Error: package 'rviz' not found

ROBOTIS-Will, 2021-03-19:
@johncabibihan 

Yay!
Looks like you haven't installed RViz.
```bash
$ sudo apt install ros-noetic-rviz
```

johncabibihan, 2021-03-21:
@ROBOTIS-Will 

Thanks so much for all the help! 

My turtlebot3 waffle pi is now working. I will try the pi camera and the robot arm to get working next.

johncabibihan, 2021-03-22:
@ROBOTIS-Will 

Here we go again... raspi-config not found.

13.1.10.1 Installation
[link](https://emanual.robotis.com/docs/en/platform/turtlebot3/appendix_raspi_cam/)

$ sudo raspi-config
results into command not found

I tried a second alternative, is from the [raspberrypi.org](https://projects.raspberrypi.org/en/projects/getting-started-with-picamera/2)

The Raspberry Pi configuration tool does not show up when I connected the SBC to the monitor, mouse, and keyboard. 

Your kind help will be very much appreciated.

johncabibihan, 2021-03-23:
@ROBOTIS-Will 

Thanks for the clarification. I just followed the instructions in the manual. 

Any hints on how I can get the RPi camera module v2 running? Thanks.

ROBOTIS-Will, 2021-03-23:
@johncabibihan 
I'm sorry about the misleading information in the eManual which is written for Raspbian OS.
I'll request to update the content.
It seems like there's no proper Raspberry Pi cam driver for the latest Ubuntu 64 bit yet.
I'll keep looking for the solution and update the eManual if I find any.
Thanks!

Paulino1956, 2021-04-01:
Thanks for your support! Turtlebot is operational ! 

Now learning how run it ! 

Paulino

New2Turtle, 2021-07-13:
Hi everyone, i have tries #700 and update the Timezone and have update the SSID and password. 
However when i type ifconfig, it only show the lo: 127.0.0.1. 

Please advice.

Thank you.

ROBOTIS-Will, 2021-07-14:
@New2Turtle 
It looks like something went wrong with your RPi recovery with the image as neither `etho0` or `wlan0` is shown in the list.
What does `$ ip link show` return?
Are you using a default TurtleBot3 package?

New2Turtle, 2021-07-14:
@ROBOTIS-Will : thanks for your time. Are you referring to the Emanual turtlebot3 package? 
I am using the image which you shared at #700 , what i did is apt- update and upgrade using LAN cable. 

Should I flash the image again ? I remembered seeing the eth0 and wlan0 before I insert the ssid and password. As for the ip link show, I will response later as I have no access to it now. 

Thank you.

ROBOTIS-Will, 2021-07-14:
@New2Turtle 
The ROS1 Noetic image for Raspberry Pi 3B+ is updated very recently.
In some cases upgrading corrupts the Raspberry Pi network driver([similar issue thread](https://community.robotsource.org/t/sbc-wifi-problem-with-melodic/1995/6)) so please try using it without apt upgrade.
Thank you.

Tianpeng-Zhang, 2021-08-12:
Hi, I encountered the same issue in WiFi setup on ROS2 Foxy Rasberry Pi 3B+. It turns out the WiFi and ethernet interface are disabled by default. 

I was able to fix this issue following [https://askubuntu.com/questions/883305/connecting-to-new-wifi-network
]

Hope this is helpful.

soulking123, 2021-09-29:
i followed the instruction untill sbc setup( no error) , but when i connect sd card to raspberry and hdmi to monitor, the screen says "no signal". i've been check the monitor and hdmi cable, nothing wrong  .
Any advice ? is it maybe because my wifi ssid have some space or something ? my wifi ssid = my wifi

ROBOTIS-Will, 2021-09-30:
Closing this thread as there isn't recent activity related on this issue.

closed 2021-09-30 by ROBOTIS-Will