ROBOTIS-GIT/turtlebot3 #703: SBC not communicating with PC

Paulino1956, 2021-02-13:
ISSUE TEMPLATE ver. 0.4.0

1. Which TurtleBot3 platform do you use?

    - [ ] Burger
    - [ ] Waffle
    - [ x] Waffle Pi

2. Which ROS is working with TurtleBot3?

    - [ ] ROS 1 Kinetic Kame
    - [x ] ROS 1 Melodic Morenia
    - [ ] ROS 1 Noetic Ninjemys
    - [ ] ROS 2 Dashing Diademata
    - [ ] ROS 2 Eloquent Elusor
    - [ ] ROS 2 Foxy Fitzroy
    - [ ] etc (Please specify your ROS Version here)

3. Which SBC(Single Board Computer) is working on TurtleBot3?

    - [ ] Intel Joule 570x
    - [ x] Raspberry Pi 3B+
    - [ ] Raspberry Pi 4
    - [ ] etc (Please specify your SBC here)

4. Which OS you installed on SBC?

    - [ ] Raspbian distributed by ROBOTIS
    - [ x] Ubuntu MATE (16.04/18.04/20.04)
    - [ ] Ubuntu preinstalled server (18.04/20.04)
    - [ ] etc (Please specify your OS here) 

5. Which OS you installed on Remote PC?

    - [ ] Ubuntu 16.04 LTS (Xenial Xerus)
    - [x ] Ubuntu 18.04 LTS (Bionic Beaver)
    - [ ] Ubuntu 20.04 LTS (Focal Fossa)
    - [ ] Windows 10
    - [ ] MAC OS X (Specify version)
    - [ ] etc (Please specify your OS here)

6. Specify the software and firmware version(Can be found from Bringup messages)

    - Software version: [x.x.x]
    - Firmware version: [x.x.x]

7. Specify the commands or instructions to reproduce the issue.

    

8. Copy and Paste the error messages on terminal.

    

9. Please describe the issue in detail.

1-running (first) roscore on the PC terminal (Shell 2) and this is what I got :
paulino@paulino-desktop:~$ roscore
... logging to /home/paulino/.ros/log/39b280c6-6d63-11eb-979a-d8c0a6261e0f/roslaunch-paulino-desktop-13678.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://10.0.0.115:33115/
ros_comm version 1.14.10

SUMMARY
========

PARAMETERS
 * /rosdistro: melodic
 * /rosversion: 1.14.10

NODES

auto-starting new master
process[master]: started with pid [13689]
ROS_MASTER_URI=http://10.0.0.115:11311/

setting /run_id to 39b280c6-6d63-11eb-979a-d8c0a6261e0f
process[rosout-1]: started with pid [13700]
started core service [/rosout]

2- open shall 1 via ssh and run pi@raspberrypi:~ $ roslaunch turtlebot3_bringup turtlebot3_robot.launch
result:
... logging to /home/pi/.ros/log/39b280c6-6d63-11eb-979a-d8c0a6261e0f/roslaunch-raspberrypi-10475.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://10.0.0.27:37031/

SUMMARY
========

PARAMETERS
 * /rosdistro: melodic
 * /rosversion: 1.14.10
 * /turtlebot3_core/baud: 115200
 * /turtlebot3_core/port: /dev/ttyACM0
 * /turtlebot3_core/tf_prefix: 
 * /turtlebot3_lds/frame_id: base_scan
 * /turtlebot3_lds/port: /dev/ttyUSB0

NODES
  /
    turtlebot3_core (rosserial_python/serial_node.py)
    turtlebot3_diagnostics (turtlebot3_bringup/turtlebot3_diagnostics)
    turtlebot3_lds (hls_lfcd_lds_driver/hlds_laser_publisher)

ROS_MASTER_URI=http://10.0.0.115:11311

ERROR: cannot launch node of type [rosserial_python/serial_node.py]: rosserial_python
ROS path [0]=/opt/ros/melodic/share/ros
ROS path [1]=/home/pi/catkin_ws/src
ROS path [2]=/opt/ros/melodic/share
ERROR: cannot launch node of type [hls_lfcd_lds_driver/hlds_laser_publisher]: hls_lfcd_lds_driver
ROS path [0]=/opt/ros/melodic/share/ros
ROS path [1]=/home/pi/catkin_ws/src
ROS path [2]=/opt/ros/melodic/share
ERROR: cannot launch node of type [turtlebot3_bringup/turtlebot3_diagnostics]: Cannot locate node of type [turtlebot3_diagnostics] in package [turtlebot3_bringup]. Make sure file exists in package path and permission is set to executable (chmod +x)
No processes to monitor
shutting down processing monitor...
... shutting down processing monitor complete
pi@raspberrypi:~ $ 

at the pi3 I have installed : 

Installing ROS Melodic on the Raspberry Pi (ROS.ORG - http://wiki.ros.org/action/fullsearch/ROSberryPi/Installing%20ROS%20Melodic%20on%20the%20Raspberry%20Pi?action=fullsearch&context=180&value=linkto%3A%22ROSberryPi%2FInstalling+ROS+Melodic+on+the+Raspberry+Pi%22)

at the PC (NVIDIA XAVIER) I have installed : 
Ubuntu install of ROS Melodic from ROS.org also.

I have check the Network configurations and they are ok for  both PC and PI:

PC:
export ROS_MASTER_URI=http://10.0.0.115:11311
export ROS_HOSTNAME=10.0.0.115

Pi:
export ROS_MASTER_URI=http://10.0.0.115:11311
export ROS_HOSTNAME=10.0.0.27
    
Thanks

ROBOTIS-Will, 2021-02-15:
Hi @Paulino1956 
Please make sure that you have sourced the ros packages from Raspberry Pi as below.
```
$ source ~/catkin_ws/devel/setup.bash
```

Paulino1956, 2021-02-15:
Hi ! 

I did use your recommendation (as described above on your note), but unfortunately got the same result (as described above on my note).

Paulino1956, 2021-02-15:
Hi Will!

Thank for your support ! I used the code as you indicated (See below), but unfortunately obtained with the same type of resul as before t:

pi@raspberrypi:~ $  source ~/catkin_ws/devel/setup.bash
pi@raspberrypi:~ $ roslaunch turtlebot3_bringup turtlebot3_robot.launch
... logging to /home/pi/.ros/log/4c2beafa-6fcd-11eb-97fc-d8c0a6261e0f/roslaunch-raspberrypi-26579.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://10.0.0.27:35567/

SUMMARY
========

PARAMETERS
 * /rosdistro: melodic
 * /rosversion: 1.14.10
 * /turtlebot3_core/baud: 115200
 * /turtlebot3_core/port: /dev/ttyACM0
 * /turtlebot3_core/tf_prefix: 
 * /turtlebot3_lds/frame_id: base_scan
 * /turtlebot3_lds/port: /dev/ttyUSB0

NODES
  /
    turtlebot3_core (rosserial_python/serial_node.py)
    turtlebot3_diagnostics (turtlebot3_bringup/turtlebot3_diagnostics)
    turtlebot3_lds (hls_lfcd_lds_driver/hlds_laser_publisher)

ROS_MASTER_URI=http://10.0.0.115:11311

ERROR: cannot launch node of type [rosserial_python/serial_node.py]: rosserial_python
ROS path [0]=/opt/ros/melodic/share/ros
ROS path [1]=/home/pi/catkin_ws/src
ROS path [2]=/opt/ros/melodic/share
ERROR: cannot launch node of type [hls_lfcd_lds_driver/hlds_laser_publisher]: hls_lfcd_lds_driver
ROS path [0]=/opt/ros/melodic/share/ros
ROS path [1]=/home/pi/catkin_ws/src
ROS path [2]=/opt/ros/melodic/share
ERROR: cannot launch node of type [turtlebot3_bringup/turtlebot3_diagnostics]: Cannot locate node of type [turtlebot3_diagnostics] in package [turtlebot3_bringup]. Make sure file exists in package path and permission is set to executable (chmod +x)
No processes to monitor
shutting down processing monitor...
... shutting down processing monitor complete
pi@raspberrypi:~ $ 

pi@raspberrypi:~ $ ls
Bookshelf       Downloads                        ros_catkin_ws
build_isolated  install_ros_melodic_rp3.sh       src
catkin_ws       melodic-ros_comm-wet.rosinstall  Templates
Desktop         Music                            Videos
devel_isolated  Pictures
Documents       Public
pi@raspberrypi:~ $ 
pi@raspberrypi:~ $ ls
Bookshelf       Downloads                        ros_catkin_ws
build_isolated  install_ros_melodic_rp3.sh       src
catkin_ws       melodic-ros_comm-wet.rosinstall  Templates
Desktop         Music                            Videos
devel_isolated  Pictures
Documents       Public
pi@raspberrypi:~ $ 
pi@raspberrypi:~ $ ls
Bookshelf       Downloads                        ros_catkin_ws
build_isolated  install_ros_melodic_rp3.sh       src
catkin_ws       melodic-ros_comm-wet.rosinstall  Templates
Desktop         Music                            Videos
devel_isolated  Pictures
Documents       Public
pi@raspberrypi:~ $ 

on the NVIDIA side :

paulino@paulino-desktop:~$ roscore
... logging to /home/paulino/.ros/log/60e8942e-6fce-11eb-b82c-d8c0a6261e0f/roslaunch-paulino-desktop-13906.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://10.0.0.115:35523/
ros_comm version 1.14.10

SUMMARY
========

PARAMETERS
 * /rosdistro: melodic
 * /rosversion: 1.14.10

NODES

auto-starting new master
process[master]: started with pid [13916]
ROS_MASTER_URI=http://10.0.0.115:11311/

setting /run_id to 60e8942e-6fce-11eb-b82c-d8c0a6261e0f
process[rosout-1]: started with pid [13928]
started core service [/rosout]
^C[rosout-1] killing on exit
[master] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done
paulino@paulino-desktop:~$ ls
catkin_ws  Downloads               install_ros_melodic.sh.1  Port    Templates
Desktop    examples.desktop        Music                     Public  Videos
Documents  install_ros_melodic.sh  Pictures                  ROS
paulino@paulino-desktop:~$ 

Thanks again in advance for you support !

ROBOTIS-Will, 2021-02-16:
Hmm... Looks like you have followed a different installation than the eManual.
If you don't need a GUI environment on Raspberry Pi, I'd recommend to Install with [Ubuntu 18.04.3 preinstalled server ARM64 for Raspberry PI3](http://old-releases.ubuntu.com/releases/18.04.3/ubuntu-18.04.3-preinstalled-server-arm64+raspi3.img.xz)
Please select the ROS version at the top of the page before following the instructions.
https://emanual.robotis.com/docs/en/platform/turtlebot3/sbc_setup/

Paulino1956, 2021-03-02:
﻿Hello!

Thanks for the recommendation, it worked ! However, after successfully running the “bringup” procedure and operate via RC-100, I got into new issues : lost wireless connection with Pi (no wlan0 and IP address).

After several unsuccessful tentatives, I have also tried using another RBPi OS, and was able to get the wireless connection working via Pi (including wlan0 and IP address). 

Thanks for your help!

Paulino

Paulino1956, 2021-03-05:
Hello - I still cannot connect Pi3 to Wi-Fi. I have configured the network, as per the e-manual. 

Thanks in advance for your support.

Paulino

Sent from my iPad

> On Mar 3, 2021, at 6:03 PM, Paulino Rego Barros Jr. <pregobarros@gmail.com> wrote:
> 
> ﻿
>> 
>> Hello again !
> 
>> maybe I was not clear on my e-mail to you (see below),  and my question is still how can I operate with the OS that you recommend? How can I  fix it?  
>> 
>> Thanks for your help! 
>> 
>> Paulino
> 
> 
>> On Mar 2, 2021, at 12:50 PM, Paulino Rego Barros <pregobarros@gmail.com> wrote:
>> 
>> ﻿Hello!
>> 
>> Thanks for the recommendation, it worked ! However, after successfully running the “bringup” procedure and operate via RC-100, I got into new issues : lost wireless connection with Pi (no wlan0 and IP address).
>> 
>> <image.jpg>
>> 
>> 
>> After several unsuccessful tentatives, I have also tried using another RBPi OS, and was able to get the wireless connection working via Pi (including wlan0 and IP address). 
>> 
>> Thanks for your help!
>> 
>> Paulino
>> 
>> 
>>

ROBOTIS-Will, 2021-03-08:
@Paulino1956 
Hi, operating TurtleBot3 with RC-100 remote controller doesn't reuquire ROS running as it directly receives command from the controller and interprets in OpenCR.
Let me create a recovery image of Melodic for RPi3B+ and see if it works for you.
This may take a few days.
Thank you.

Paulino1956, 2021-03-08:
Fantastic ! Thank you! 

Paulino

ROBOTIS-Will, 2021-03-10:
@Paulino1956 
Hi, here's the Melodic test image for Raspberry Pi 3B+ that I built with my TurtleBot3 burger.
https://www.robotis.com/service/download.php?no=2011

The user name is `ubuntu` and the password is `turtlebot`

Before using it, you need to open and modify the network settings.
```bash
$ sudo nano /etc/netplan/50-cloud-init.yaml
```
Replace the `WIFI_SSID` and `WIFI_PASSWORD` for your wifi settings.
[image]

Then open the `~/.bashrc` file
```bash
$ nano ~/.bashrc
```
Modify where `ROS_MASTER_URI`, `ROS_HOSTNAME`, `TURTLEBOT3_MODEL` are defined.
```
# Replace IP_ADDRESS_OF_REMOTE_PC with the IP address of remote pc
# Both Remote PC and Raspberry Pi should be connected in the same local network
# e.g) export ROS_MASTER_URI=http://192.168.0.10:11311
export ROS_MASTER_URI=http://localhost:11311
# Replace IP_ADDRESS_OF_RASPBERRY_PI with the IP address of Raspberry Pi
# e.g) export ROS_HOSTNAME=192.168.0.11
export ROS_HOSTNAME=localhost

export TURTLEBOT3_MODEL=burger
```

Paulino1956, 2021-03-10:
Hello !

I executed the instructions as per your e-mail, and was able to make the Pi to access the Wi-Fi, and use ifconfig to get an IP address. As per the instructions, I made the modifications (on the nano ~/.bashrc ) using the new IP address.

After that, I was able to run ssh from the remote PC and connected to the Pi. I executed the procedure to the SBC set up and install the complete configuration including : ROS Melodic Meronia, the ROS Packages and Network Configuration. I started the installation of the OpenCR set up and had to stop, but  properly shut down the PI. When I returned to ssh via the remote terminal it was not able to connect to the Pi again. 

I run the PI on a terminal and use ifconfig to get the IP address, and got this message below:

The Pi is not connecting  to the network, which was the same issue I experienced before. It seems like something - removes the Pi from the network - when we install ROS.

Again, I really do appreciate your continuous support.

Thanks,

Paulino

ROBOTIS-Will, 2021-03-11:
@Paulino1956 
The image I uploaded has all the setup for SBC and you only need to configure the wifi and network settings as I described earlier.
Do NOT proceed additional SBC setup with the image.
Thank you.

Paulino1956, 2021-03-11:
Thanks ! Will do it again ! 

Paulino

Paulino1956, 2021-03-11:
Achieved good progress! Almost there, but I experiencing few more issues:

Problem 1- 
Got everything working via ssh and follow instruction and launched Bringup, then teleoperation. I got the expected screen results, follow the instructions toon, but was not able to make the turtlebot wheels to move/rotate via the screen commands (I.e. moving around : a,w,s,x and d). At the same time, the terminal was responding and registering linear and angular velocity like the movement was happening.

Problem 2-

Run rqt - it did not work, and got the following results :

I was able to run SLAM Node successfully,  and got RVIZ operating on the screen. Then, I run the Teleoperation Node procedure, but again was not able to move the Turtlebot from the Remote PC.

Problem 3-
I was note able to run the Navigation Node. When I run (on the Remote PC) the instruction $ roslaunch turtlebot3_navigation.launch map_file:=$HOME/map.yaml, the  RVIZ did not run properly on the screen and it shutdown after trying to locate the map. 

Thanks in advance for your support!

Paulino

ROBOTIS-Will, 2021-03-12:
@Paulino1956 
Hi, please check below options.

1. Make sure to correctly set up the network configuration for both Raspberry Pi and Remote PC.
[image]
2. Make sure that you have correctly [setup the OpenCR firmware](https://emanual.robotis.com/docs/en/platform/turtlebot3/opencr_setup/#opencr-setup).
3. Try [time sync](https://emanual.robotis.com/docs/en/platform/turtlebot3/faq/#timesync-between-turtlebot3-and-remote-pc) for both Raspberry Pi and Remote PC
4. I don't see the attached image. Replying directly via email doesn't include attached images.
If you haven't installed rqt, please install with `$ sudo apt-get install ros-melodic-rqt*`

Paulino1956, 2021-03-12:
Hello !

I have executed the instruction 1, 2, 3 and all worked fine. I have
executed instruction 4 ($ sudo apt-get install ros-melodic-rqt*), but I am
still not able to run rqt. And this is the screen message I get when I
execute $rqt:
$ rqt
[ERROR] [1615585777.000866575]: Skipped loading plugin with error: XML
Document '/opt/ros/melodic/share/rqt_virtual_joy/plugin.xml' has no Root
Element. This likely means the XML is malformed or missing..
RosPluginProvider._parse_plugin_xml() plugin file
"/opt/ros/melodic/share/rqt_virtual_joy/plugin.xml" in package
"rqt_virtual_joy" not found
RosPluginProvider._parse_plugin_xml() plugin file
"/opt/ros/melodic/share/rqt_virtual_joy/plugin.xml" in package
"rqt_virtual_joy" not found

Also, I am still not able to move the wheels from the PC. Message on screen:
$ roslaunch turtlebot3_teleop turtlebot3_teleop_key.launch
... logging to
/home/paulino/.ros/log/07e97012-837b-11eb-b010-d8c0a6261e0f/roslaunch-paulino-desktop-13402.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://10.0.0.115:40941/

SUMMARY
========

PARAMETERS
 * /model: waffle
 * /rosdistro: melodic
 * /rosversion: 1.14.10

NODES
  /
    turtlebot3_teleop_keyboard (turtlebot3_teleop/turtlebot3_teleop_key)

ROS_MASTER_URI=http://10.0.0.115:11311

process[turtlebot3_teleop_keyboard-1]: started with pid [13417]

Control Your TurtleBot3!
---------------------------
Moving around:
        w
   a    s    d
        x

w/x : increase/decrease linear velocity (Burger : ~ 0.22, Waffle and Waffle
Pi : ~ 0.26)
a/d : increase/decrease angular velocity (Burger : ~ 2.84, Waffle and
Waffle Pi : ~ 1.82)

space key, s : force stop

CTRL-C to quit

At the same time, the terminal was responding and registering linear and
angular velocity like the movement was happening:
Moving around:
        w
   a    s    d
        x

w/x : increase/decrease linear velocity (Burger : ~ 0.22, Waffle and Waffle
Pi : ~ 0.26)
a/d : increase/decrease angular velocity (Burger : ~ 2.84, Waffle and
Waffle Pi : ~ 1.82)

space key, s : force stop

CTRL-C to quit

At the same time, the terminal was responding and registering linear and
angular velocity like the movement was happening:

currently: linear vel 0.01 angular vel 0.0
currently: linear vel 0.02 angular vel 0.0
currently: linear vel 0.03 angular vel 0.0
currently: linear vel 0.04 angular vel 0.0
currently: linear vel 0.05 angular vel 0.0
currently: linear vel 0.06 angular vel 0.0
currently: linear vel 0.07 angular vel 0.0
currently: linear vel 0.08 angular vel 0.0
currently: linear vel 0.09 angular vel 0.0
currently: linear vel 0.1 angular vel 0.0
currently: linear vel 0.11 angular vel 0.0

NEXT - I have been able to execute on th Pi:
***@***.*** (pi) :~$ roslaunch turtlebot3_bringup
turtlebot3_robot.launch,
then executed on PC :
***@***.***:~$ roslaunch turtlebot3_slam
turtlebot3_slam.launch,and got the RVIz

But not able to move the wheels.

Again, Thanks in advance for your support ! Have a nice weekend !

Paulino

ROBOTIS-Will, 2021-03-15:
Hi, 
What did you get from running the rqt install command below? Were there any errors or warnings?
```bash
$ sudo apt-get install ros-melodic-rqt*
```
Could you try reading the ROS topics from the Raspberry Pi on TurtleBot3 with the command below?
```bash
$ rostopic list
```
I'm not sure why rqt is not working with your PC.
I'd recommend to reinstall the desktop full version of ROS and TurtleBot3 packages on your PC.

Paulino1956, 2021-03-15:
Thanks! Will do it! 

What about the other issue: 

I  am still not able to move the wheels from the PC. Message on screen:
$ roslaunch turtlebot3_teleop turtlebot3_teleop_key.launch
... logging to /home/paulino/.ros/log/07e97012-837b-11eb-b010-d8c0a6261e0f/roslaunch-paulino-desktop-13402.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://10.0.0.115:40941/

SUMMARY
========

PARAMETERS
 * /model: waffle
 * /rosdistro: melodic
 * /rosversion: 1.14.10

NODES
  /
    turtlebot3_teleop_keyboard (turtlebot3_teleop/turtlebot3_teleop_key)

ROS_MASTER_URI=http://10.0.0.115:11311

process[turtlebot3_teleop_keyboard-1]: started with pid [13417]

Control Your TurtleBot3!
---------------------------
Moving around:
        w
   a    s    d
        x

w/x : increase/decrease linear velocity (Burger : ~ 0.22, Waffle and Waffle Pi : ~ 0.26)
a/d : increase/decrease angular velocity (Burger : ~ 2.84, Waffle and Waffle Pi : ~ 1.82)

space key, s : force stop

CTRL-C to quit

At the same time, the terminal was responding and registering linear and angular velocity like the movement was happening:
Moving around:
        w
   a    s    d
        x

w/x : increase/decrease linear velocity (Burger : ~ 0.22, Waffle and Waffle Pi : ~ 0.26)
a/d : increase/decrease angular velocity (Burger : ~ 2.84, Waffle and Waffle Pi : ~ 1.82)

space key, s : force stop

CTRL-C to quit

Thanks again! 

Paulino

ROBOTIS-Will, 2021-03-15:
I need to see if Raspberry Pi and Remote PC is properly connected in the same network and sharing the ROS topics to each other.
In order to confirm this, `$ rostopic list` will list up the topics in the current network.
Also, when running the teleop keyboard from the remote PC, check if `cmd_vel` topic is correctly updated from the Raspberry Pi with entering below command.
```bash
$ rostopic echo /cmd_vel
```

Paulino1956, 2021-03-15:
👍! Will do !

And, thanks again !

Paulino

Paulino1956, 2021-03-15:
Hello !

On the PC-

This is what I get when I run - $ sudo apt-get install ros-melodic-rqt* :
***@***.***:~$ sudo apt-get install ros-melodic-rqt*
[sudo] password for paulino:
Reading package lists... Done
Building dependency tree
Reading state information... Done
Note, selecting 'ros-melodic-rqt-runtime-monitor' for glob
'ros-melodic-rqt*'
Note, selecting 'ros-melodic-rqt-multiplot' for glob 'ros-melodic-rqt*'
Note, selecting 'ros-melodic-rqt-py-trees' for glob 'ros-melodic-rqt*'
Note, selecting 'ros-melodic-rqt-virtual-joy' for glob 'ros-melodic-rqt*'
Note, selecting 'ros-melodic-rqt-rviz-dbgsym' for glob 'ros-melodic-rqt*'
Note, selecting 'ros-melodic-rqt-play-motion-builder' for glob
'ros-melodic-rqt*'
Note, selecting 'ros-melodic-rqt-controller-manager' for glob
'ros-melodic-rqt*'
Note, selecting 'ros-melodic-rqt-multiplot-dbgsym' for glob
'ros-melodic-rqt*'
Note, selecting 'ros-melodic-rqt-joint-trajectory-controller' for glob
'ros-melodic-rqt*'
Note, selecting 'ros-melodic-rqt-rosmon' for glob 'ros-melodic-rqt*'
Note, selecting 'ros-melodic-rqt-py-console' for glob 'ros-melodic-rqt*'
Note, selecting 'ros-melodic-rqt-image-view' for glob 'ros-melodic-rqt*'
Note, selecting 'ros-melodic-rqt-service-caller' for glob 'ros-melodic-rqt*'
Note, selecting 'ros-melodic-rqt-publisher' for glob 'ros-melodic-rqt*'
Note, selecting 'ros-melodic-rqt-topic' for glob 'ros-melodic-rqt*'
Note, selecting 'ros-melodic-rqt-gui-cpp' for glob 'ros-melodic-rqt*'
Note, selecting 'ros-melodic-rqt-gui-cpp-dbgsym' for glob 'ros-melodic-rqt*'
Note, selecting 'ros-melodic-rqt-bag' for glob 'ros-melodic-rqt*'
Note, selecting 'ros-melodic-rqt-robot-dashboard' for glob
'ros-melodic-rqt*'
Note, selecting 'ros-melodic-rqt-common-plugins' for glob 'ros-melodic-rqt*'
Note, selecting 'ros-melodic-rqt-launch' for glob 'ros-melodic-rqt*'
Note, selecting 'ros-melodic-rqt-plot' for glob 'ros-melodic-rqt*'
Note, selecting 'ros-melodic-rqt-robot-steering' for glob 'ros-melodic-rqt*'
Note, selecting 'ros-melodic-rqt-dep' for glob 'ros-melodic-rqt*'
Note, selecting 'ros-melodic-rqt-gui' for glob 'ros-melodic-rqt*'
Note, selecting 'ros-melodic-rqt-action' for glob 'ros-melodic-rqt*'
Note, selecting 'ros-melodic-rqt-ez-publisher' for glob 'ros-melodic-rqt*'
Note, selecting 'ros-melodic-rqt-gui-py' for glob 'ros-melodic-rqt*'
Note, selecting 'ros-melodic-rqt-moveit' for glob 'ros-melodic-rqt*'
Note, selecting 'ros-melodic-rqt-shell' for glob 'ros-melodic-rqt*'
Note, selecting 'ros-melodic-rqt-msg' for glob 'ros-melodic-rqt*'
Note, selecting 'ros-melodic-rqt-robot-monitor' for glob 'ros-melodic-rqt*'
Note, selecting 'ros-melodic-rqt-reconfigure' for glob 'ros-melodic-rqt*'
Note, selecting 'ros-melodic-rqt-tf-tree' for glob 'ros-melodic-rqt*'
Note, selecting 'ros-melodic-rqt-py-common' for glob 'ros-melodic-rqt*'
Note, selecting 'ros-melodic-rqt-console' for glob 'ros-melodic-rqt*'
Note, selecting 'ros-melodic-rqt-srv' for glob 'ros-melodic-rqt*'
Note, selecting 'ros-melodic-rqt-top' for glob 'ros-melodic-rqt*'
Note, selecting 'ros-melodic-rqt-pose-view' for glob 'ros-melodic-rqt*'
Note, selecting 'ros-melodic-rqt-play-motion-builder-dbgsym' for glob
'ros-melodic-rqt*'
Note, selecting 'ros-melodic-rqt-rosmon-dbgsym' for glob 'ros-melodic-rqt*'
Note, selecting 'ros-melodic-rqt-drone-teleop' for glob 'ros-melodic-rqt*'
Note, selecting 'ros-melodic-rqt-web' for glob 'ros-melodic-rqt*'
Note, selecting 'ros-melodic-rqt' for glob 'ros-melodic-rqt*'
Note, selecting 'ros-melodic-rqt-graph' for glob 'ros-melodic-rqt*'
Note, selecting 'ros-melodic-rqt-logger-level' for glob 'ros-melodic-rqt*'
Note, selecting 'ros-melodic-rqt-robot-plugins' for glob 'ros-melodic-rqt*'
Note, selecting 'ros-melodic-rqt-rotors' for glob 'ros-melodic-rqt*'
Note, selecting 'ros-melodic-rqt-nav-view' for glob 'ros-melodic-rqt*'
Note, selecting 'ros-melodic-rqt-joint-trajectory-plot' for glob
'ros-melodic-rqt*'
Note, selecting 'ros-melodic-rqt-ground-robot-teleop' for glob
'ros-melodic-rqt*'
[... 108 lines cut ...]
(0.4.8-0bionic.20201020.023339).
ros-melodic-rqt-topic is already the newest version
(0.4.11-1bionic.20201020.082857).
ros-melodic-rqt-virtual-joy is already the newest version
(0.1.2-1bionic.20210113.162105).
ros-melodic-rqt-web is already the newest version
(0.4.8-0bionic.20201020.023405).
0 upgraded, 0 newly installed, 0 to remove and 4 not upgraded.

On the Pi (via SSH from the PC)-

This is what I get when I run : $ rostopic list
***@***.***:~$ rostopic list
ERROR: Unable to communicate with master!
***@***.***:~$

also on  the Pi (via ssh from the PC) -
***@***.***:~$  rostopic echo /cmd_vel
ERROR: Unable to communicate with master!

I will reinstall ROS on the PC and get back to you shortly.

Thanks,

Paulino

Paulino1956, 2021-03-15:
Some more progress today !!

I was able to execute on PC - ***@***.***:~$ roslaunch
turtlebot3_teleop turtlebot3_teleop_key.launch and GOT TO  OPERATE
Turtlebot via Teleoperation/Keyboard.

Also, and as per your recommendation, I executed on Pi:

***@***.***:~$ rostopic list

/battery_state
/clicked_point
/cmd_vel
/cmd_vel_rc100
/diagnostics
/firmware_version
/imu
/initialpose
/joint_states
/magnetic_field
/motor_power
/move_base_simple/goal
/odom
/reset
/rosout
/rosout_agg
/rpms
/scan
/sensor_state
/sound
/tf
/tf_static
/version_info

and executed  the $ rostopic echo /cmd_vel (while  running the
Teleoperation)

***@***.***:~$ rostopic echo /cmd_vel
linear:
  x: 0.0
  y: 0.0
  z: 0.0
angular:
  x: 0.0
  y: 0.0
  z: 0.0

Now, I am still having problems when I execute

***@***.***:~$ rqt
[ERROR] [1615842639.190347851]: Skipped loading plugin with error: XML
Document '/opt/ros/melodic/share/rqt_virtual_joy/plugin.xml' has no Root
Element. This likely means the XML is malformed or missing..
RosPluginProvider._parse_plugin_xml() plugin file
"/opt/ros/melodic/share/rqt_virtual_joy/plugin.xml" in package
"rqt_virtual_joy" not found
RosPluginProvider._parse_plugin_xml() plugin file
"/opt/ros/melodic/share/rqt_virtual_joy/plugin.xml" in package
"rqt_virtual_joy" not found

***@***.***:~$ sudo apt-get install ros-melodic-rqt*

***@***.***:~$ sudo apt-get install ros-melodic-rqt*
[sudo] password for paulino:
Reading package lists... Done
Building dependency tree
Reading state information... Done
Note, selecting 'ros-melodic-rqt-runtime-monitor' for glob
'ros-melodic-rqt*'
Note, selecting 'ros-melodic-rqt-multiplot' for glob 'ros-melodic-rqt*'
[... 153 lines cut ...]
(0.6.0-0bionic.20210206.160356).
ros-melodic-rqt-rviz-dbgsym is already the newest version
(0.6.0-0bionic.20210206.160356).
ros-melodic-rqt-service-caller is already the newest version
(0.4.8-0bionic.20210114.100855).
ros-melodic-rqt-shell is already the newest version
(0.4.9-0bionic.20201020.023342).
ros-melodic-rqt-srv is already the newest version
(0.4.8-0bionic.20210114.193756).
ros-melodic-rqt-tf-tree is already the newest version
(0.6.0-0bionic.20210113.193336).
ros-melodic-rqt-top is already the newest version
(0.4.8-0bionic.20201020.023339).
ros-melodic-rqt-topic is already the newest version
(0.4.11-1bionic.20201020.082857).
ros-melodic-rqt-virtual-joy is already the newest version
(0.1.2-1bionic.20210113.162105).
ros-melodic-rqt-web is already the newest version
(0.4.8-0bionic.20201020.023405).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
***@***.***:~$

Thanks ! and Looking forward to get your feedback !

Paulino

ROBOTIS-Will, 2021-03-17:
Hi @Paulino1956 
Could you try below answer and see if it fixes the problem?
https://answers.ros.org/question/354039/rqt_virtual_joypluginxml-file-is-missing/

Paulino1956, 2021-03-17:
hello there!

Unfortunately it didn't work! I have executed te instructions

You can add plugin.xml in your directory
/opt/ros/melodic/share/rqt_virtual_joy.

   1.

   Go to rqt_virtual_joy directory.

   roscd rqt_vritual_joy

   2.

   Make file plugin.xml

   sudo touch plugin.xml

   3.

   Edit the plugin.xml file with gedit (or other)

   sudo gedit plugin.xml

   4.

   Copy and paste the contents of the plugin.xml file from the
rqt_virtual_joystick
   github <https://github.com/aquahika/rqt_virtual_joystick> to your
   plugin.xml.

edit the file plugin.xml
<library path="src">
<class name="My Plugin" type="rqt_virtual_joy.virtual_joy_module.MyPlugin"
base_class_type="rqt_gui_py::Plugin">
<description>
An example Python GUI plugin to create a great user interface.
</description>
<qtgui>
<group>
<label>Robot Tools</label>
</group>

<label>Virtual Joystick</label>
<icon type="file">resource/input-gaming.png</icon>
<statustip>Great user interface to provide real value.</statustip>
</qtgui>
</class>
</library> and executed the bringup proceure/teleoperation/topi monitor and
executed rqt . Please see  below:
***@***.***:~$ rqt
Traceback (most recent call last):
  File "/opt/ros/melodic/bin/rqt", line 13, in <module>
    sys.exit(main.main())
  File "/opt/ros/melodic/lib/python2.7/dist-packages/rqt_gui/main.py", line
65, in main
    hash(os.environ['ROS_PACKAGE_PATH'])))
  File "/opt/ros/melodic/lib/python2.7/dist-packages/qt_gui/main.py", line
600, in main
    plugin_manager.discover()
  File
"/opt/ros/melodic/lib/python2.7/dist-packages/qt_gui/plugin_manager.py",
line 125, in discover
    self._plugin_menu.add_plugin(plugin_descriptor)
  File
"/opt/ros/melodic/lib/python2.7/dist-packages/qt_gui/plugin_menu.py", line
82, in add_plugin
    self._enrich_action(action, action_attributes, base_path)
  File
"/opt/ros/melodic/lib/python2.7/dist-packages/qt_gui/plugin_menu.py", line
133, in _enrich_action
    action_attributes['icon'], action_attributes.get('icontype', None),
base_path)
  File
"/opt/ros/melodic/lib/python2.7/dist-packages/qt_gui/icon_loader.py", line
46, in get_icon
    raise UserWarning('icon "%s" not found' % str(path))
UserWarning: icon
"/opt/ros/melodic/share/rqt_virtual_joy/resource/input-gaming.png" not found
***@***.***:~$ rqt
Traceback (most recent call last):
  File "/opt/ros/melodic/bin/rqt", line 13, in <module>
    sys.exit(main.main())
  File "/opt/ros/melodic/lib/python2.7/dist-packages/rqt_gui/main.py", line
65, in main
    hash(os.environ['ROS_PACKAGE_PATH'])))
  File "/opt/ros/melodic/lib/python2.7/dist-packages/qt_gui/main.py", line
600, in main
    plugin_manager.discover()
  File
"/opt/ros/melodic/lib/python2.7/dist-packages/qt_gui/plugin_manager.py",
line 125, in discover
    self._plugin_menu.add_plugin(plugin_descriptor)
  File
"/opt/ros/melodic/lib/python2.7/dist-packages/qt_gui/plugin_menu.py", line
82, in add_plugin
    self._enrich_action(action, action_attributes, base_path)
  File
"/opt/ros/melodic/lib/python2.7/dist-packages/qt_gui/plugin_menu.py", line
133, in _enrich_action
    action_attributes['icon'], action_attributes.get('icontype', None),
base_path)
  File
"/opt/ros/melodic/lib/python2.7/dist-packages/qt_gui/icon_loader.py", line
46, in get_icon
    raise UserWarning('icon "%s" not found' % str(path))
UserWarning: icon
"/opt/ros/melodic/share/rqt_virtual_joy/resource/input-gaming.png" not found
***@***.***:~$

Thanks for your help !

Paulino

ROBOTIS-Will, 2021-03-18:
@Paulino1956 
I'm not familiar with the error you are getting.
Could you try uninstall and reinstall ROS?

Paulino1956, 2021-03-18:
Well,  I have already done that on both the PC and the Pi. I am using the following image on the Pi: https://www.robotis.com/service/download.php?no=2011 , which was sent by you guys.
The original e-manual image for the PI ( http://old-releases.ubuntu.com/releases/18.04.3/ubuntu-18.04.3-preinstalled-server-arm64+raspi3.img.xz) it was disconnecting the Pi from the WI-FI network. However, when I run the e-manual image on the Pi,  I was able to run the instruction $rqt.

Thanks fo your help !

Paulino

ROBOTIS-Will, 2021-03-18:
@Paulino1956 
The TurtleBot3 SBC image that I provided does not support GUI environment, and cannot run rqt on it.
Please run the GUI programs such as rqt, rviz, gazebo on the remote PC.

Paulino1956, 2021-06-08:
Dear Friends,

For some reason my RPi3 is not connecting to the internet. I had this issue before, but was able to make it connect using the image you sent me in March (please see below). Today, I used/burned the same image (you sent me in March) again in a new micro SD, followed the same instruction and I was not able to make it work. Same issue: the RPi3 does not connect to the internet ($ifconfig > no ip for the RPi3). Is this image still valid? Anyway I have also download and tried the image available on the e-manual and it did not work either. 

Thanks in advance for your support!

Paulino

ROBOTIS-Will, 2021-06-09:
Hi @Paulino1956 
Recently, the ROS GPG key has been expired which disturbs ROS package update and upgrade.
You can follow the instruction below to update the key, but this issue doesn't have to do with the network.
https://discourse.ros.org/t/ros-gpg-key-expiration-incident/20669

Paulino1956, 2021-06-09:
Thank you! Will try again tomorrow!

Paulino

closed 2024-11-08 by sunghowoo